SpinalHDL/VexRiscv #1: Failed to compile regression tests

roman3017, 2017-08-05:
Hello,

I have some problems to compile regression tests:

```bash
$ sbt "run-main vexriscv.demo.GenFull"
[warn] Executing in batch mode.
[warn]   For better performance, hit [ENTER] to switch to interactive mode, or
[warn]   consider launching sbt without any commands, or explicitly passing 'shell'
[info] Loading project definition from VexRiscv/project
[info] Set current project to VexRiscv (in build file:VexRiscv/)
[info] Running vexriscv.demo.GenFull 
[Runtime] SpinalHDL v0.10.15    git head : 7e6b26ca08f52c6959f931ef9e993590e23625e2
[Runtime] JVM max memory : 981.5MiB
[Runtime] Current date : 2017.08.05 09:21:22
[Progress] at 0.000 : Start elaboration
[Progress] at 0.841 : Start analysis and transform
[Progress] at 0.862 : Get names from reflection
[Progress] at 0.872 : Transform connections
[Progress] at 0.931 : Infer nodes's bit width
[Progress] at 1.040 : Check combinatorial loops
[Progress] at 1.075 : Check cross clock domains
[Progress] at 1.137 : Simplify graph's nodes
[Progress] at 1.166 : Check that there is no incomplete assignment
[Progress] at 1.180 : Collect signals not used in the graph
[Warning] 32 signals were pruned. You can call printPruned on the backend report to get more informations.
[Progress] at 1.191 : Finalise
[Info] Graph has 6170 nodes
[Progress] at 1.225 : Write Verilog
[Progress] at 1.229 :       emit UnsignedDivider
[Progress] at 1.252 :     emit InstructionCache
[Warning] memReadSync with dontCare is as readFirst into Verilog ways_0_tags_port1
[Warning] memReadSync with dontCare is as readFirst into Verilog ways_0_datas_port1
[Progress] at 1.265 :     emit DataCache
[Warning] memReadSync with dontCare is as readFirst into Verilog way_tags_port2
[Warning] memReadSync with dontCare is as readFirst into Verilog way_data_port1
[Warning] memReadSync with dontCare is as readFirst into Verilog victim_buffer_port2
[Progress] at 1.306 :     emit MixedDivider
[Progress] at 1.309 :   emit VexRiscv
[Warning] memReadSync with dontCare is as readFirst into Verilog MemoryTranslatorPlugin_shared_cache_port0
[Warning] memReadSync with dontCare is as readFirst into Verilog RegFilePlugin_regFile_port0
[Warning] memReadSync with dontCare is as readFirst into Verilog RegFilePlugin_regFile_port1
[Warning] memReadSync with dontCare is as readFirst into Verilog branchCache_port1
[Done] at 1.361
[success] Total time: 2 s, completed 5-Aug-2017 9:21:23 AM
```
```bash
$ verilator --version
Verilator 3.874 2015-06-06 rev verilator_3_872-20-g0d43051
```
```bash
$ g++ --version
g++ (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```bash
$ cd src/test/cpp/regression/
$ make clean run
rm -rf obj_dir
verilator -cc  ../../../../VexRiscv.v  -O3 -CFLAGS -std=c++11 -LDFLAGS -pthread  -CFLAGS -DIBUS_CACHED -CFLAGS -DDBUS_CACHED -CFLAGS -DREDO=10 -CFLAGS -pthread -CFLAGS -DTHREAD_COUNT=4 -CFLAGS -DDHRYSTONE -CFLAGS -DCSR -CFLAGS -DISA_TEST -CFLAGS -DMMU -CFLAGS -DMUL -CFLAGS -DDIV -CFLAGS -DDEBUG_PLUGIN -CFLAGS -DDEBUG_PLUGIN_STD -CFLAGS -DTRACE_START=0 --gdbbt  -Wno-WIDTH --x-assign unique --exe main.cpp
No stack.
make  -j  -C obj_dir/ -f VVexRiscv.mk VVexRiscv
make[1]: Entering directory 'obj_dir'
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=0 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DCSR -DDBUS_CACHED -DDEBUG_PLUGIN -DDEBUG_PLUGIN_STD -DDHRYSTONE -DDIV -DIBUS_CACHED -DISA_TEST -DMMU -DMUL -DREDO=10 -DTHREAD_COUNT=4 -DTRACE_START=0 -pthread -std=c++11   -c -o main.o ../main.cpp
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=0 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DCSR -DDBUS_CACHED -DDEBUG_PLUGIN -DDEBUG_PLUGIN_STD -DDHRYSTONE -DDIV -DIBUS_CACHED -DISA_TEST -DMMU -DMUL -DREDO=10 -DTHREAD_COUNT=4 -DTRACE_START=0 -pthread -std=c++11   -c -o verilated.o /usr/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=0 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DCSR -DDBUS_CACHED -DDEBUG_PLUGIN -DDEBUG_PLUGIN_STD -DDHRYSTONE -DDIV -DIBUS_CACHED -DISA_TEST -DMMU -DMUL -DREDO=10 -DTHREAD_COUNT=4 -DTRACE_START=0 -pthread -std=c++11   -c -o verilated_dpi.o /usr/share/verilator/include/verilated_dpi.cpp
/usr/bin/perl /usr/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VVexRiscv.cpp VVexRiscv_VexRiscv.cpp > VVexRiscv__ALLcls.cpp
/usr/bin/perl /usr/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VVexRiscv__Dpi.cpp VVexRiscv__Syms.cpp > VVexRiscv__ALLsup.cpp
[... 95 lines cut ...]
           ^
../main.cpp:1297:20: error: ‘class VVexRiscv’ has no member named ‘VexRiscv’
    iBusAccess(top->VexRiscv->writeBack_PC, &instruction, &error);
                    ^
../main.cpp:1299:26: error: ‘class VVexRiscv’ has no member named ‘VexRiscv’
     uint32_t code = top->VexRiscv->RegFilePlugin_regFile[28];
                          ^
      Archiving VVexRiscv__ALL.a ...
ar r VVexRiscv__ALL.a VVexRiscv__ALLcls.o VVexRiscv__ALLsup.o
ar: creating VVexRiscv__ALL.a
ranlib VVexRiscv__ALL.a
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-parentheses-equality’
VVexRiscv.mk:81: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory 'obj_dir'
makefile:95: recipe for target 'compile' failed
make: *** [compile] Error 2
```

Am I missing something?
Thanks,

Roman

Dolu1990, 2017-08-05:
Hi, you should take the verilator 3.9+ they changed something in their naming 
You can compile it from source : 

```
# Verilator (for sim only)
sudo apt-get install git make autoconf g++ flex bison
git clone http://git.veripool.org/git/verilator   # Only first time
unsetenv VERILATOR_ROOT  # For csh; ignore error if on bash
unset VERILATOR_ROOT  # For bash
cd verilator
git pull        # Make sure we're up-to-date
git tag         # See what versions exist
autoconf        # Create ./configure script
./configure
make
sudo make install
```

closed 2017-08-05 by Dolu1990

roman3017, 2017-08-05:
Thank you very much, it works with:
```
$ verilator --version
Verilator 3.906 2017-06-22 rev verilator_3_906
```

Dolu1990, 2017-08-05:
Good : D