SpinalHDL/VexRiscv #2: SIMD

fathomtreelimited, 2017-08-22:
Hi!

I've started to think about a MAC module, mainly for audio processing.

I'd like to multiply a 64-bit register location (8guard,32signal,24overflow) by a 32bit coefficient and store the product in a 88bit accumulator register 32,32,24.

Then ultimately 16-way SIMD. Does this feasable?

Dolu1990, 2017-08-22:
Hi,

Yes sure, then depending the maximal frequancy/performance that you want, you can fully integrate it into the CPU pipeline (see  the multiplication 
 instruction implementation : https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/MulPlugin.scala#L40 )
Or you can add new "coprocessor" stages dedicated to your math. (i don't have examples of that, but it's realy simple ^^)

Then about 16 ways SIMD, how do you want to feed the system with data ? 16 way is much ^^

fathomtreelimited, 2017-08-22:
Thanks for the source link!

Maybe 1 way MAC pointed to a large register file of 32 x 32-bit coefficient that the instruction can loop through with zero overhead.

Dolu1990, 2017-08-22:
Yes, would be  great ^^
So you would have to add something like 3 instruction ?
One to load values into the coefficent ram
One to reset the acumulator
One to push a data into the MAC ?

fathomtreelimited, 2017-08-23:
I'll have a go and submit a pull. ;-)

fathomtreelimited, 2017-08-25:
Firstly I will look to recreate an existing DSP like 48-bit data path with 76-bit accumulator optimised for digital audio processing such as IIR/FIR etc..

There will be separate registers or memory regions required for 
-- 28-bit coefficients (5.23)
-- 48-bit data (25.23)

The 48-bit audio data word could be aligned to two 32-bit words  There will of course be some unused bits in that case.

The 48-bit word structure is as follows:

MSB is the sign bit with lower 47bits are data bits
8 bits of overhead at the upper end of the word
32-bits of audio sample data data
	- bit 39 indicates audio samples are treated as signed
	- audio sample MSB next to overhead boundary
8-bit of precision or noise at the lower end of the word

There will be a 24-bit x 48-bit MAC & 76-bit adder unit.

The output will then be truncated back to a 48-bit audio word.

fathomtreelimited, 2017-08-26:
I spotted some code here that seems to extend the ALU Opcodes.
https://github.com/SpinalHDL/SpinalHDL/search?q=AluOpcode.MAC&type=Code&utf8=%E2%9C%93
Would this be a suitable route to follow?

Dolu1990, 2017-08-26:
Yes it is the implementation used in https://github.com/SpinalHDL/VexRiscv#add-a-custom-instruction-to-the-cpu-via-the-plugin-system

This example is a single cycle single stage one, but all other combination are possible.

Dolu1990, 2017-08-26:
Also there is an example of CPU using this extension : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/GenCustomSimdAdd.scala#L14

closed 2017-08-28 by Dolu1990