SpinalHDL/VexRiscv #3: FPGA Flow

riscveval, 2017-08-23:
HI, Can you add FPGA flow for all these cores.

Dolu1990, 2017-08-23:
Hi,
You mean the flow to synthetise and get numbers from each core vs each FPGA target ?

I use it : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/SynthesisBench.scala#L10

Note : I work only on windows, and you will have to update path to EDA tools (https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/SynthesisBench.scala#L78)

Let's me know ^^

riscveval, 2017-08-23:
Hi, I have some problems to Synthesis VexRiscv using command sbt "run-main vexriscv.demo.VexRiscvSynthesisBench"
[Done] at 7.385
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallestnoCSR_Artix7
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallestnoCSR_CycloneV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallestnoCSR_CycloneIV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallestnoCSR_CycloneII
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallest_Artix7
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallest_CycloneV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallest_CycloneIV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallest_CycloneII
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallandproductive_Artix7
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallandproductive_CycloneV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallandproductive_CycloneIV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvsmallandproductive_CycloneII
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfullnoMMUnocache_Artix7
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfullnoMMUnocache_CycloneV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfullnoMMUnocache_CycloneIV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfullnoMMUnocache_CycloneII
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfullnoMMU_Artix7
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfullnoMMU_CycloneV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfullnoMMU_CycloneIV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfullnoMMU_CycloneII
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfull_Artix7
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfull_CycloneV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfull_CycloneIV
rmdir /S /Q \home\krradhak\VexRiscv\FPGA_VexRiscvVexRiscvfull_CycloneII
VexRiscv smallest no CSR ->
[error] (run-main-0) java.io.IOException: Cannot run program "cmd": error=2, No such file or directory
java.io.IOException: Cannot run program "cmd": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1047)
	at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.scala:69)
	at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuilderImpl.scala:98)
	at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang(ProcessBuilderImpl.scala:112)
	at spinal.lib.eda.xilinx.VivadoFlow$.doCmd(VivadoFlow.scala:11)
	at spinal.lib.eda.xilinx.VivadoFlow$.apply(VivadoFlow.scala:27)
	at spinal.lib.eda.bench.XilinxStdTargets$$anon$4.synthesise(Targets.scala:75)
	at spinal.lib.eda.bench.Bench$$anonfun$1$$anonfun$apply$1$$anonfun$2.apply(Bench.scala:30)
	at spinal.lib.eda.bench.Bench$$anonfun$1$$anonfun$apply$1$$anonfun$2.apply(Bench.scala:30)
	at scala.concurrent.impl.Future$PromiseCompletingRunnable.liftedTree1$1(Future.scala:24)
	at scala.concurrent.impl.Future$PromiseCompletingRunnable.run(Future.scala:24)
	at java.util.concurrent.ForkJoinTask$AdaptedRunnable.exec(ForkJoinTask.java:1265)
	at java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:334)
	at java.util.concurrent.ForkJoinWorkerThread.execTask(ForkJoinWorkerThread.java:604)
	at java.util.concurrent.ForkJoinPool.scan(ForkJoinPool.java:784)
	at java.util.concurrent.ForkJoinPool.work(ForkJoinPool.java:646)
	at java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:398)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:187)
	at java.lang.ProcessImpl.start(ProcessImpl.java:130)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1028)
	at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.scala:69)
	at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuilderImpl.scala:98)
	at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang(ProcessBuilderImpl.scala:112)
	at spinal.lib.eda.xilinx.VivadoFlow$.doCmd(VivadoFlow.scala:11)
	at spinal.lib.eda.xilinx.VivadoFlow$.apply(VivadoFlow.scala:27)
	at spinal.lib.eda.bench.XilinxStdTargets$$anon$4.synthesise(Targets.scala:75)
	at spinal.lib.eda.bench.Bench$$anonfun$1$$anonfun$apply$1$$anonfun$2.apply(Bench.scala:30)
	at spinal.lib.eda.bench.Bench$$anonfun$1$$anonfun$apply$1$$anonfun$2.apply(Bench.scala:30)
	at scala.concurrent.impl.Future$PromiseCompletingRunnable.liftedTree1$1(Future.scala:24)
	at scala.concurrent.impl.Future$PromiseCompletingRunnable.run(Future.scala:24)
	at java.util.concurrent.ForkJoinTask$AdaptedRunnable.exec(ForkJoinTask.java:1265)
	at java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:334)
	at java.util.concurrent.ForkJoinWorkerThread.execTask(ForkJoinWorkerThread.java:604)
	at java.util.concurrent.ForkJoinPool.scan(ForkJoinPool.java:784)
	at java.util.concurrent.ForkJoinPool.work(ForkJoinPool.java:646)
	at java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:398)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
[error] (compile:runMain) Nonzero exit code: 1
[error] Total time: 22 s, completed 23 Aug, 2017 5:05:22 PM

Dolu1990, 2017-08-23:
It look like there is some missing / in paths ?
From the original Bench(rtls, targets, "E:/tmp/") , how did you modified it, did you removed the / before the " ?
(those script aren't realy smart about this kind of things XD

Let's me know ^^

riscveval, 2017-08-24:
Hi, I changed the path but still i am getting the same issue...
[Done] at 8.134
rmdir /S /Q \tmp\VexRiscvsmallestnoCSR_Artix7
rmdir /S /Q \tmp\VexRiscvsmallestnoCSR_CycloneV
rmdir /S /Q \tmp\VexRiscvsmallestnoCSR_CycloneIV
rmdir /S /Q \tmp\VexRiscvsmallestnoCSR_CycloneII
rmdir /S /Q \tmp\VexRiscvsmallest_Artix7
rmdir /S /Q \tmp\VexRiscvsmallest_CycloneV
rmdir /S /Q \tmp\VexRiscvsmallest_CycloneIV
rmdir /S /Q \tmp\VexRiscvsmallest_CycloneII
rmdir /S /Q \tmp\VexRiscvsmallandproductive_Artix7
rmdir /S /Q \tmp\VexRiscvsmallandproductive_CycloneV
rmdir /S /Q \tmp\VexRiscvsmallandproductive_CycloneIV
rmdir /S /Q \tmp\VexRiscvsmallandproductive_CycloneII
rmdir /S /Q \tmp\VexRiscvfullnoMMUnocache_Artix7
rmdir /S /Q \tmp\VexRiscvfullnoMMUnocache_CycloneV
rmdir /S /Q \tmp\VexRiscvfullnoMMUnocache_CycloneIV
rmdir /S /Q \tmp\VexRiscvfullnoMMUnocache_CycloneII
rmdir /S /Q \tmp\VexRiscvfullnoMMU_Artix7
rmdir /S /Q \tmp\VexRiscvfullnoMMU_CycloneV
rmdir /S /Q \tmp\VexRiscvfullnoMMU_CycloneIV
rmdir /S /Q \tmp\VexRiscvfullnoMMU_CycloneII
rmdir /S /Q \tmp\VexRiscvfull_Artix7
rmdir /S /Q \tmp\VexRiscvfull_CycloneV
rmdir /S /Q \tmp\VexRiscvfull_CycloneIV
rmdir /S /Q \tmp\VexRiscvfull_CycloneII
VexRiscv smallest no CSR ->
[error] (run-main-0) java.io.IOException: Cannot run program "cmd": error=2, No such file or directory
java.io.IOException: Cannot run program "cmd": error=2, No such file or directory

Dolu1990, 2017-08-24:
Hmmm, maybe it work on myone because i have some cygwin executable in my path, i will check on a clean windos os a tell you how it go

riscveval, 2017-08-24:
Hi, I am using Ubuntu 14.04, All other configurations are compile properly. Can you give some suggestion to solve in this issue.

Dolu1990, 2017-08-24:
Hoooo
Ok i get it, sorry in the precedent messages is wrote "I work only on windows,"  But i wanted to write "It work only on windows,"  So there is the issue.

It work only on windows, because basicaly i already had lot of issues/paines to get Xilinx/Altera EDA tools working on ubuntu linux distribution. I know there is ways to have it working on them, but for synthesis things, i'm working on windows to not waste time :) That's why it only work on windows.

Do you have access to a windows machine to try those things ?

riscveval, 2017-08-24:
Hi, All other builds are supported ubuntu, if you try to solve this everything is one common platform.

Dolu1990, 2017-08-24:
Ok ^^
I will take a look to make it compatible with linux platform, shouldn't be complicated

Dolu1990, 2017-08-24:
Hi,
I made an update in the SpinalHDL library which host the flow tools,
So if you git pull the last SpinalHDL and do a     "sbt clean publish-local"   then the flow will be linux ready 
I have tested for quartus, but not for vivado, i curenlty can't download it, as i'm in a hostel with crap crap wifi.

For instance, my current linux setup is => 

```scala
    val targets = XilinxStdTargets(
      vivadoArtix7Path = null
    ) ++ AlteraStdTargets(
      quartusCycloneIIPath = null,
      quartusCycloneIVPath = "/eda/intelFPGA_lite/17.0/quartus/bin",
      quartusCycloneVPath  = "/eda/intelFPGA_lite/17.0/quartus/bin"
    )

    Bench(rtls, targets, "/eda/tmp/")
```

Let's me know

Dolu1990, 2017-08-24:
Hooo Also, be carefull, it run multiple synthesis in parallel, if you have 8 CPU logical thread on your PC, it will run 6 synthesis at the same time (75% logical count). So if you don't have much RAM it could be an issue. If it the case you can create multiple targets lists and do multiple runs of the Bench tool

Dolu1990, 2017-08-25:
Hi,

I tested the vivado flow + quartus flow on mint 17.3, it worked fine with for instance :
```scala
    val targets = XilinxStdTargets(
      vivadoArtix7Path = "/eda/Xilinx/Vivado/2017.2/bin"
    ) ++ AlteraStdTargets(
      quartusCycloneIIPath = null,
      quartusCycloneIVPath = "/eda/intelFPGA_lite/17.0/quartus/bin",
      quartusCycloneVPath  = "/eda/intelFPGA_lite/17.0/quartus/bin"
    )

    Bench(rtls, targets, "/eda/tmp/")
```

Let's me know if you have any issue

closed 2017-08-25 by Dolu1990

riscveval, 2017-08-28:
Hi, Thanks for the update.  I am getting in this issue only for cyclone II Boards
[error] (run-main-0) java.io.FileNotFoundException: /nfs/iind/disks/i2r.disk.0001/RISCV_Evaluation/SpinalHDL/tmpMurax_CycloneII/Murax.sta.rpt (No such file or directory)
java.io.FileNotFoundException: /nfs/iind/disks/i2r.disk.0001/RISCV_Evaluation/SpinalHDL/tmpMurax_CycloneII/Murax.sta.rpt (No such file or directory)
        at java.io.FileInputStream.open0(Native Method)

And also I have two doubts to you :
1. How can your measure frequency?
 
2. How to measure Dhrystone performance? Can you run any simulation to measure the performance ?

Dolu1990, 2017-08-28:
**How can you measure frequency? give example.**
In all those synthesis to get Area and Frequancy, the target frequancy is set very high (unreachable), so the synthesis tool will be timing stressed (area usage will be higher than normal one).
Then i just parse the EDA tools reports to get those information. For instance in Xilinx i look into doit.log for :

```
1. Slice Logic
--------------

+----------------------------+------+-------+-----------+-------+
|          Site Type         | Used | Fixed | Available | Util% |
+----------------------------+------+-------+-----------+-------+
| Slice LUTs                 | 1021 |     0 |     41000 |  2.49 |
|   LUT as Logic             | 1001 |     0 |     41000 |  2.44 |
|   LUT as Memory            |   20 |     0 |     13400 |  0.15 |
|     LUT as Distributed RAM |   16 |     0 |           |       |
|     LUT as Shift Register  |    4 |     0 |           |       |
| Slice Registers            | 1291 |     0 |     82000 |  1.57 |
|   Register as Flip Flop    | 1291 |     0 |     82000 |  1.57 |
|   Register as Latch        |    0 |     0 |     82000 |  0.00 |
| F7 Muxes                   |    2 |     0 |     20500 | <0.01 |
| F8 Muxes                   |    0 |     0 |     10250 |  0.00 |
+----------------------------+------+-------+-----------+-------+
```

and

```
Timing Report

Slack (VIOLATED) :        -0.764ns  (required time - arrival time)
```

For altera i look into Murax.sta.rpt for :

```
+------------------------------------------------------------------------------------------------------------+
; Slow 1200mV 85C Model Fmax Summary                                                                         ;
+------------+-----------------+-------------+---------------------------------------------------------------+
; Fmax       ; Restricted Fmax ; Clock Name  ; Note                                                          ;
+------------+-----------------+-------------+---------------------------------------------------------------+
; 140.17 MHz ; 140.17 MHz      ; clk         ;                                                               ;
+------------+-----------------+-------------+---------------------------------------------------------------+
```
and into Murax.flow.rpt for

```
+----------------------------------------------------------------------------------+
; Flow Summary                                                                     ;
+------------------------------------+---------------------------------------------+
; Flow Status                        ; Successful - Mon Jul 31 13:49:42 2017       ;
; Quartus Prime Version              ; 15.1.0 Build 185 10/21/2015 SJ Lite Edition ;
; Revision Name                      ; Murax                                       ;
; Top-level Entity Name              ; Murax                                       ;
; Family                             ; Cyclone IV E                                ;
; Device                             ; EP4CE30F29C6                                ;
; Timing Models                      ; Final                                       ;
; Total logic elements               ; 1,893 / 28,848 ( 7 % )                      ;
;     Total combinational functions  ; 1,483 / 28,848 ( 5 % )                      ;
;     Dedicated logic registers      ; 1,250 / 28,848 ( 4 % )                      ;
; Total registers                    ; 1250                                        ;
; Total pins                         ; 104 / 533 ( 20 % )                          ;
; Total virtual pins                 ; 0                                           ;
; Total memory bits                  ; 67,840 / 608,256 ( 11 % )                   ;
; Embedded Multiplier 9-bit elements ; 0 / 132 ( 0 % )                             ;
; Total PLLs                         ; 0 / 4 ( 0 % )                               ;
+------------------------------------+---------------------------------------------+
```

**How are you running dhrystone in different configurations?**
For pure CPU runs (no SoC) i use the regression tests (https://github.com/SpinalHDL/VexRiscv#regression-tests) which also run the dhrystone and report how many cycle it used to do 200 dhrystones runs.

Also dhrystone compilation is done with -o3  -no-inline and some other flags, but for sure the -no-inline is there.

As Briey is a cached CPU, the above number are also valable for it.
For Murax which is an uncached CPU and were the IBus and DBus performance are realy important, i have a special Murax config with 256 KB of ram. Then i run the src/test/cpp/murax simulation environnement to load the dhrystone into it via GDB, then run it and look at the reported performance.

About 1.16 DMIPS/Mhz, by activating the branch into the execute stage in place of the memory stage + activating the memory load bypassing into the memory stage in place of the writeback stage you get 1.25 DMIPS/Mhz. If in addition (not implemented) i have a single cycle divide unit (in place of the 34 cycle that i have actualy) i would get 1.35 DMIPS/Mhz.
What i want to mean with those number is, the settings used in showed VexRiscv configs are FPGA/frequancy friendly, and it's why it get lower performance compared to other ASIC targeted simple 5 stages RISC-V implementation.

About calculation, i have this XLS 
[dmpis.xlsx](https://github.com/SpinalHDL/VexRiscv/files/1255404/dmpis.xlsx)

riscveval, 2017-08-29:
Hi Thank for your valuable support.
How are you calculating Fmax from slack time (Xilinx Instance). i am trying to calculate 1/t but the value is not matched. 
Can you share step by steps for calculating DMIPS/MHz Vexriscvfull. I am not understand from XLS.

Dolu1990, 2017-08-29:
For xilinx => Requirement - Slack  = 1000 / (2.5 -(-0.764))   MHz

```
Timing Report

Slack (VIOLATED) :        -0.764ns  (required time - arrival time)
  Source:                 system_ram_ram_symbol2_reg/CLKBWRCLK
                            (rising edge-triggered cell RAMB18E1 clocked by clk  {rise@0.000ns fall@1.250ns period=2.500ns})
  Destination:            system_cpu/RegFilePlugin_regFile_reg_1/ADDRARDADDR[6]
                            (rising edge-triggered cell RAMB18E1 clocked by clk  {rise@0.000ns fall@1.250ns period=2.500ns})
  Path Group:             clk
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            2.500ns  (clk rise@2.500ns - clk rise@0.000ns)
```

For DMIPS => https://github.com/SpinalHDL/VexRiscvSocSoftware/blob/master/projects/murax/dhrystone/src/dhry_1.c#L292
So  (1e6f/1757.0f) * Number_Of_Runs / User_Time   where Number_Of_Runs is in my setup 200 runs, and User_Time is the number of cycle to do all those run.
Practical example 98000 cycles for 200 run => 1.16 DMIPS/MHz

riscveval, 2017-08-30:
Hi For frequency calculation why are you using 1000 what is this value represent?
Similarly DMIPS Calculation  1757 VAX MIPS Machine value, Number of runs =200, User_time 97136, Why are you using 1e6f what is this value represent?

Note : Target Board is mandatory for to calculate dhrystone performance in all these cores?
All other values are displayed in simulation. Why DMIPS/MHz is not displayed in simulation.

Dolu1990, 2017-08-30:
1000 is to convert GHz to MHz as the time value is in ns

1e6 is because we want to scale into DMIPS/Mhz

no target board is mandatory at all, all those core / SoC can run dhrystone in verilator simulation.
Then CPU without cache are subject to the memory bus performance, it's why for Murax SoC i run the test in it, in place of the pure CPU sim.

Basicaly, the original dhrytone source doesn't show the DMIPS/Mhz, i modified it to do it only later.

riscveval, 2017-08-30:
Hi I am trying to run the simulation in Briey/ Murax. I am not getting any values in terminal.
Briey -->BOOT
SDRAM : MODE REGISTER DEFINITION CAS=3 burstLength=0
 VGA is connected but in this window i am not getting any value. After few seconds its automatically closed.

Murax --> ./obj_dir/VMurax
BOOT
after that i am not getting any value.

To connect OpenOCD to the simulation for Murax : I am getting this error 

adapter speed: 800 kHz
adapter_nsrst_delay: 260
jtag_ntrst_delay: 250
Info : set servers polling period to 50ms
Info : clock speed 800 kHz
Error: TRST/SRST error

To connect OpenOCD to the simulation for Briey: I am getting this error 

adapter speed: 4000 kHz
adapter_nsrst_delay: 260
jtag_ntrst_delay: 250
Info : set servers polling period to 50ms
Info : clock speed 4000 kHz
Error: TRST/SRST error

Note : Make clean run not shows any error.

Dolu1990, 2017-08-30:
Hi
Having a dark VGA is normal, it is not active when it boot. Also not geting any value is normal after boot, as no binary is loaded into.
Then about openOCD, how are you running it ? which command ?
Can you try to run it with the verbose argument ? (-v3 i think) and send me the log ?

I never had this TRST/SRST error, so i don't realy know what is happening.

A thing to be carfule, is the cpu0.yaml content is not the same for briey and murax, so be carfule to point the right cpu0.yaml to openocd.

I just tried briey yesterday in sim, and it was working fine, so please, give more info ^^

riscveval, 2017-08-31:
Hi, 
I am using the following command for to run openocd 

krradhak@krradhak:~/VexRiscv/openocd_riscv$ src/openocd -f tcl/interface/jtag_tcp.cfg -c "set BRIEY_CPU0_YAML /home/krradhak/VexRiscv/cpu0.yaml" -f tcl/target/briey.cfg

Note : After boot , not getting any value then how to know the passed tests and dhrystone performance?

When i am trying to run regression tests the following tests are failed in VexRiscv 

FAIL machineCsr
*** top->VexRiscv->writeBack_RegFilePlugin_regFileWrite_payload_data is 2 but should be 11 ***

FAIL DebugPluginTest
timeout

Herewith i am attaching the screen shot of errors for openocd debugger for your reference
 
[image]

[image]

Dolu1990, 2017-08-31:
Hi

So for regressions :
FAIL machineCsr
*** top->VexRiscv->writeBack_RegFilePlugin_regFileWrite_payload_data is 2 but should be 11 ***
Is normal if you aren't using the very full version of the CPU, as the machineCsr regression test is testing the fully implemented CSR.

If you use the GenFull to get the CPU it use a small CSR configuration which doesn't implement all things tested by the CSR regression (because most of them are useless into an embedded CPU for FPGA).
If you want the real full version of the CPU with all CSR implemented, you can generate the vexriscv.TestsWorkspace 
(src/main/scala/vexriscv/TestsWorkspace.scala)
=> SUCCESS machineCsr

Then about 
FAIL DebugPluginTest
timeout

It look it is the real issue, probably related to the environnement. 
I just tested now with a fresh git clone on linux mint 17.3 + the GenFull and this test is passing :
CONNECTED
SUCCESS DebugPluginTest

So something which is possible is that the TCP ports used by the simulation are already binded by another application in your PC ? port 7893   But it would be realy bad luck. You don't get this CONNECTED message before the failure of the DebugPluginTest ?
Just, if this DebugPluginTest isn't passing, there is no value trying with openocd. First have to fix this DebugPluginTest issue.

Can you send the VCD wave ? You can get it by running the test like this : 
make clean run REDO=1 TRACE=yes DHRYSTONE=no

Else is the dhrystone benchmarks passing in the regression tests ? (it should, or maybe the regression doesn't go futher the DebugPluginTest failure ?) 

For the Briey and Murax tests, nothing is loaded automaticaly in them, you have to connect with openocd + gdb and then load the dhrystone benchmark in them (https://github.com/SpinalHDL/VexRiscvSocSoftware)
Then it will print into the simulation the UART TX from the CPU, which will take the Dhrystone printf.

riscveval, 2017-09-04:
Hi I am getting the CONNECTED Message, After that how can i show the values in UART. I am trying for Gtkterm Serial console but i am unable to get any values.

Herewith I am sending some screen shots for your reference. 
[image]
[image]
[image]

Note : Can you share step by step instruction after getting CONNECTED Message.

Dolu1990, 2017-09-04:
Look all good ^^
To load something in Briey first you need to compile some RISC-V binaries => 

Install RISC-V gcc :
wget https://static.dev.sifive.com/dev-tools/riscv64-unknown-elf-gcc-20170612-x86_64-linux-centos6.tar.gz
tar -xzvf riscv64-unknown-elf-gcc-20170612-x86_64-linux-centos6.tar.gz
sudo mv riscv64-unknown-elf-gcc-20170612-x86_64-linux-centos6 /opt/riscv64-unknown-elf-gcc-20170612-x86_64-linux-centos6
sudo mv /opt/riscv64-unknown-elf-gcc-20170612-x86_64-linux-centos6 /opt/riscv
echo 'export PATH=/opt/riscv/bin:$PATH' >> ~/.bashrc 

Compile binaries :
git clone https://github.com/SpinalHDL/VexRiscvSocSoftware.git
cd VexRiscvSocSoftware/
make all

Run a binary :
cd projects/briey/vga/build/
/opt/riscv/bin/riscv64-unknown-elf-gdb vga.elf
target remote localhost:3333
set remotetimeout 60
monitor reset halt
load
continue

And there you will see something writing on the VGA, you can do the same for th dhrystone bencmark.
Just not it will take some time to load the binary, and the performance result of the dhrystone will be lower than expected as the compilated dhrystone is doing some cache trashing. (So performance down only because of the compilation, not the CPU)

Also, things emited by the CPU via the UARt are printed directly on the verilator shell, not a TTY

For murax and pure cpu it is the same flow. Just that binary aren't necessearly compatible.

But to get the dhystone of pure CPU regression (no SoC) better just use the regression make clean run and get the numbers there

Dolu1990, 2017-09-04:
Just, what had produced the "Error: TRST/SRST error" before ?

riscveval, 2017-09-04:
Hi  I am following the same flow as you mentioned in your Repo.
cd VexRiscvSocSoftware/
make all
But i am getting some files missed in riscv toolchain :
[image]

Dolu1990, 2017-09-04:
For me it's getting 
/opt/riscv/riscv64-unknown-elf/lib/rv32i/ilp32//libc.a
 not 
 /opt/riscvriscv64-unknown-elf/lib/rv32i/ilp32//libc.a

It's weird as i tried this morning with a clean git clone XD

Maybe you already have some environement variables set ?
Any way i just submited a fix which force to have a /  after the RISCV_PATH ^^
So git clone a fresh https://github.com/SpinalHDL/VexRiscvSocSoftware.git then it should be fine, let's me know

riscveval, 2017-09-04:
Hi for this issue resolved. 
cd projects/briey/vga/build/
/opt/riscv/bin/riscv64-unknown-elf-gdb vga.elf
i am not getting as you mentioned in the earlier solution. 
[image]

Dolu1990, 2017-09-04:
Hi,
This screen look good, then you have to type into the GDB terminal the following => 
target remote localhost:3333
set remotetimeout 60
monitor reset halt
load
continue

Have you tried ?

riscveval, 2017-09-05:
Hi i tried but i am getting localhost:3333: Connection timed out.

[image]
Herewith i am attaching screen shot for your reference.

Dolu1990, 2017-09-05:
O.o
So with for instance Briey verilator launched + openocd + finaly GDB you get localhost:3333 connection time out ? what about the openocd screen ?

That's realy weird, because realy, each time i sent you commands and things i tested time in the same time

Dolu1990, 2017-09-05:
To be sure, the order to do things is :
1) verilator
2) openocd
3) gdb