SpinalHDL/VexRiscv #4: GDB not working with FPGA

AaronKel, 2017-10-04:
Hello all,
I'm having a little problem with the FPGA implementation to test functionality on sillicon and the compilation is completely fine but when it comes to interfacing with the board.
The OpenOCD can connect, I'm using the command:

`openocd -f tcl/interface/altera-usb-blaster.cfg -c "set BRIEY_CPU0_YAML /home/aaron/VexRiscv/cpu0.yaml" -f tcl/target/briey.cfg`

 However, when then trying to connect to the GDB with:

`target remote localhost:3333`

It times out, am I missing something? I'm using the DE0-nano development board.

Dolu1990, 2017-10-04:
Hi,

How do you have linked the Altera USB blaster to the briey toplevel ?
(Currently I always have used a dedicated JTAG via some USB FTDI)

Also about the DE0-nano, i already have test Briey on it, be carefull the SDRAM configuration is different than the default one (DE1-SoC). You need to generate the toplevel via (for instance) this main : https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Briey.scala#L394

AaronKel, 2017-10-04:
I have used the default:

> 
> io_jtag_tck =>  GPIO_2(0),
>       io_jtag_tms =>  GPIO_2(3),
>       io_jtag_tdi =>  GPIO_2(1),
>       io_jtag_tdo =>  GPIO_2(2),

Then attached the correct slots on the blaster but it gets to:

> 
> YOLO riscv_spinal_target_create
> Info : set servers polling period to 50ms
> YOLO riscv_spinal_init_target
> Info : No lowlevel driver configured, will try them all
> Info : Altera USB-Blaster II found (Firm. rev. = 6�l)
> Info : This adapter doesn't support configurable speed

and stops and I cannot connect to it via GDB. What sort of FTDI chip/setup did you use. I think it would be helpful if I tried it your way, since you have already tested it.

Dolu1990, 2017-10-05:
I'm using the ft2232h chip, so, the very "standard" way, the only element on the jtag chain is the Briey SOC. Do you have multiple jtag tap on your JTAG ? Or the Briey is the only one ? Have you powered the vcc_trgt pin of the dongle ? (i had forget it many times XD)
Also could you give  a try with the verbose openocd ?  -v3   if i remember it right

Dolu1990, 2017-10-05:
Also, just to minimise risks of error, don't modify any thing of the de0-nano project/sources (for the moment :))

AaronKel, 2017-10-07:
I'll run through my setup, because I may have just made a wiring mistake!

So I tried ordering in the ft232h breakout board form adafruit:

> https://www.adafruit.com/product/2264

[image]

Reverted back into a clean DE0-nano file from here:

> https://drive.google.com/drive/folders/0B-CqLXDTaMbKZGdJZlZ5THAxRTQ?usp=sharing

Then I looked at the top level files for the JTAG and it seems like they are on:

>       io_jtag_tck =>  GPIO_2(0),
>       io_jtag_tms =>  GPIO_2(3),
>       io_jtag_tdi =>  GPIO_2(1),
>       io_jtag_tdo =>  GPIO_2(2),

Then looking at the pin mapping it shows:

> GPIO_2[0]   |   PIN_A14
> GPIO_2[1]   |   PIN_B16
> GPIO_2[2]   |   PIN_C14
> GPIO_2[3]   |   PIN_C16

So looking at the pin mapping of the DE0 nano:

[image][image]

Then looking in the FT232H datasheet it shows

> **Pin |  MPSSE Configuration Description** 
> 13     |  JTAG TCK, Test interface clock
> 14     |  JTAG TDI, Test Data Input
> 15     |  JTAG TDO, Test Data output
> 16     |  JTAG TMS, Test Mode Select

with pins 13, 14, 15, 16, 17, 18, 19, 20 on the ADBUS[7:0] and from adafruit:

> D0 through D7 - These are the ADBUS pins

So my final wiring is:

> 13 (D0) -> A14
> 14 (D1) -> B16
> 15 (D2) -> C14
> 16 (D3) -> C16

Then I got the OpenOCD from

> https://github.com/SpinalHDL/openocd_riscv

Compiled it ran it with the simulation okay, then I tried running;
`$ sudo openocd -f tcl/interface/ftdi/ft2232h_breakout.cfg -c 'set BRIEY_CPU0_YAML /home/aaron/VexRis/cpu0.yaml' -f tcl/target/briey.cfg`
Got an error that it wasn't found so changed ft2232h_breakout.cfg to

```
interface ftdi
#ftdi_device_desc "FT2232H Breakout Module"
ftdi_vid_pid 0x0403 0x6014

# Channel for using JTAG
ftdi_channel 0

# Initial state and direction setting (ref. AN_135_MPSSE_Basics.pdf)
# ftdi_layout_init [data] [direction]

#  name   signal  state      direction  JTAG name
# *DBUS0  TCK/SK  (low =0)   (out=1)    TDI
[... 249 lines cut ...]
Debug: 257 553 command.c:143 script_debug(): command - ocd_mflash ocd_mflash init
Debug: 259 556 mflash.c:1377 handle_mflash_init_command(): Initializing mflash devices...
Debug: 260 556 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_nand init
Debug: 261 556 command.c:143 script_debug(): command - ocd_nand ocd_nand init
Debug: 263 560 tcl.c:497 handle_nand_init_command(): Initializing NAND devices...
Debug: 264 560 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_pld init
Debug: 265 560 command.c:143 script_debug(): command - ocd_pld ocd_pld init
Debug: 267 563 pld.c:205 handle_pld_init_command(): Initializing PLDs...
Debug: 268 563 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_soft_reset_halt
Debug: 269 563 command.c:143 script_debug(): command - soft_reset_halt ocd_soft_reset_halt
User : 271 566 target.c:2948 handle_soft_reset_halt_command(): requesting target halt and executing a soft reset
Debug: 272 566 vexriscv.c:1448 vexriscv_soft_reset_halt(): -
Debug: 273 568 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_init
Debug: 274 568 command.c:143 script_debug(): command - init ocd_init

```

I think the error is similar to what you were saying about the target voltage:
```
Error: 237 536 core.c:949 jtag_examine_chain_check(): JTAG scan chain interrogation failed: all ones
Error: 238 536 core.c:950 jtag_examine_chain_check(): Check JTAG interface, timings, target power, etc.
```
Would I just supply 3.3v to one of the pins?

Many thanks

Dolu1990, 2017-10-07:
I don't think it is a supply issue in this case, have you flash the usb dongle ? often the default config isn't JTAG ready.

Anyways, the issue is realy in the low level jtag stuff, has the Briey jtag tap isn't detected.
Let's me know ^^

AaronKel, 2017-10-08:
I finally figured it out after a lot of head scratching. I decided to use the USB-blaster again, I was using this version;

[image]

These clones use a cheap ft423(?) I think so was causing the hangups. If I recompile OpenOCD with the following;

> ./bootstrap
> ./configure --enable-ftdi --enable-usb_blaster_libftdi --enable-dummy
> make
> #sudo make install

It all worked, I knew it was something simple! Thank you for helping me diagnose there was no error on the VexRiscv side!

AaronKel, 2017-10-08:
close

Dolu1990, 2017-10-08:
Good :)
Thanks for the info ^^

MindShow, 2019-04-29:
i tried a usb blaster clone，but return some errors
`Open On-Chip Debugger 0.10.0+dev-01202-gced8dcd6 (2019-04-23-21:53)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'usb_blaster' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
/home/csliu/Desktop/riscvwork/VexRiscvSocSoftware/openocd/cpu0.yaml
adapter speed: 10 kHz
adapter_nsrst_delay: 260
jtag_ntrst_delay: 250
Info : set servers polling period to 50ms
Info : usb blaster interface using libftdi
Error: unable to get latency timer
Info : This adapter doesn't support configurable speed
Info : JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
Error: fpga_spinal.bridge: IR capture error; saw 0x08 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
requesting target halt and executing a soft reset
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
Error: fpga_spinal.bridge: IR capture error; saw 0x08 not 0x01
Warn : Bypassing JTAG setup events due to errors
===== VexRiscv registers
(0) x0 (/32)
(1) x1 (/32)
(2) x2 (/32)
(3) x3 (/32)
(4) x4 (/32)
(5) x5 (/32)
(6) x6 (/32)
(7) x7 (/32)
(8) x8 (/32)
(9) x9 (/32)
(10) x10 (/32)
(11) x11 (/32)
(12) x12 (/32)
(13) x13 (/32)
(14) x14 (/32)
(15) x15 (/32)
(16) x16 (/32)
(17) x17 (/32)
(18) x18 (/32)
(19) x19 (/32)
(20) x20 (/32)
(21) x21 (/32)
(22) x22 (/32)
(23) x23 (/32)
(24) x24 (/32)
(25) x25 (/32)
(26) x26 (/32)
(27) x27 (/32)
(28) x28 (/32)
(29) x29 (/32)
(30) x30 (/32)
(31) x31 (/32)
(32) pc (/32): 0x80000000 (dirty)
(65) ustatus (/32)
(66) fflags (/32)
(67) frm (/32)
(68) fcsr (/32)
(69) uie (/32)
(70) utvec (/32)
(129) uscratch (/32)
(130) uepc (/32)
(131) ucause (/32)
(132) utval (/32)
(133) uip (/32)
(833) mstatus (/32)
(834) misa (/32): 0x00000000
(835) medeleg (/32)
(836) mideleg (/32)
(837) mie (/32)
(838) mtvec (/32)
(839) mcounteren (/32)
(897) mscratch (/32)
(898) mepc (/32)
(899) mcause (/32)
(900) mtval (/32)
(901) mip (/32)
(1985) mtohost (/32)
(1986) mfromhost (/32)
(1987) mreset (/32)
(1988) mipi (/32)
(1989) miobase (/32)
(3137) cycle (/32)
(3138) time (/32)
(3139) instret (/32)
(3265) cycleh (/32)
(3266) timeh (/32)
(3267) instreth (/32)
(3922) mvendorid (/32)
(3923) marchid (/32)
(3924) mimpid (/32)
`

Dolu1990, 2019-04-29:
unfortunately i don't have  jtag blaster to try :/

AaronKel, 2019-05-03:
@MindShow If you give me your openocd script I can try running it.

MindShow, 2019-05-03:
the usb blaster (only a mcu) can not work
another usb blaster (ft245 + cpld)  work well

altera-usb-blaster.cfg
```
#
# Altera USB-Blaster
#
# http://www.altera.com/literature/ug/ug_usb_blstr.pdf
#

interface usb_blaster
usb_blaster_lowlevel_driver ftdi
# These are already the defaults.
# usb_blaster_vid_pid 0x09FB 0x6001
# usb_blaster_device_desc "USB-Blaster"
# usb_blaster_lowlevel_driver ftdi
```

MindShow, 2019-06-05:
> the usb blaster (only a mcu) can not work
> another usb blaster (ft245 + cpld) work well
>
> altera-usb-blaster.cfg
>
> ```
> #
> # Altera USB-Blaster
> #
> # http://www.altera.com/literature/ug/ug_usb_blstr.pdf
> # [...]

edit the code and regenerate the firmware，now i can use the usb-blaster debug VexRiscv
[image]

mithro, 2019-06-05:
@MindShow What are the chips under your red text?

MindShow, 2019-06-05:
from top to bottom，
SL2.1
CH552G
CH552G
CH340G

mithro, 2019-06-05:
@MindShow - So it is firmware running on an E8051 processor which is trying to emulate the original BusBlaster system. The OpenOCD driver for the BusBlaster has a bunch of different modes. Maybe try them?

You can see the options at https://github.com/fjullien/openOCD/blob/master/src/jtag/drivers/usb_blaster/usb_blaster.c#L914-L943

MindShow, 2019-06-05:
yes, it is a 8051 processor,

mithro, 2019-06-05:
@MindShow Yes, and the frequency options.

Dolu1990, 2019-06-05:
@MindShow So things are working now ?

Else maybe i can compare your JTAG frame with my ones ?
Also, what are the logs of openocd with the -d option can help the diagnostic

MindShow, 2019-06-06:
@mithro 
i had set 
```
usb_blaster_lowlevel_driver ftdi
```
the USB Blaster does not support Configurable speed

@Dolu1990 
yes, it work well,

JTAG TDO change at TCK posedge in VexRiscv project, so the blaster should capture TDO before changed

Dolu1990, 2019-06-06:
Hooo right, i should  change TDO to be on falling edge.  Will do it :)
http://www.tij.co.jp/ods/images/SWRS166/SWAS031-069.gif

Thanks for the info

Dolu1990, 2019-06-24:
I merged the changes, now the JTAG changing TDO on the falling edge.
So, everything is fine ?
Can close the issue :) ?

MindShow, 2019-06-24:
got it，
https://github.com/SpinalHDL/SpinalHDL/blob/5b8c37718dbfdaeb70e294b6ece984f46733d077/lib/src/main/scala/spinal/lib/com/jtag/JtagTap.scala#L71
thanks