SpinalHDL/VexRiscv #7: Xilinx Vivado inferring block ram on register file

kaofishy, 2017-10-25:
Not really sure if this is a bug but I've noticed that by default, Vivado infers the CPU register file as a block ram. On one hand it saves LUTs when you have free block rams but it complicates timing a bit. Is this an intentional part of the design?

Dolu1990, 2017-10-25:
For compatibilities reasons, the default register file implementation is done in a syncronous way (=> block ram).
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/RegFilePlugin.scala#L54

Bus you can also ask the register file plugin to do register file access in a asyncronous way, which will then maybe help the xilinx tool to infer it as a LUT based ram (https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/RegFilePlugin.scala#L53)

So for that you need to change, for instance in the Briey SoC, SYNC into ASYNC there https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Briey.scala#L104

If it isn't enough, you can add an synthesis attribue to https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/RegFilePlugin.scala#L30 this way : 
      val regFile = Mem(Bits(32 bits),32) addAttribute(Verilator.public) addAttribute("RAM_STYLE","DISTRIBUTED")

kaofishy, 2017-10-25:
Thanks for your response, I didn't realize you could add synthesis attributes directly in the Scala files! I will experiment with the different options.

kaofishy, 2017-10-25:
BTW, which Artix-7 part did you run the timing analysis against? Just want to get a ballpark of what I could expect.

Dolu1990, 2017-10-25:
About performances numbers, they are obtained by using the fastest speed grade of each family.
https://github.com/SpinalHDL/SpinalHDL/blob/master/lib/src/main/scala/spinal/lib/eda/bench/Targets.scala#L81

closed 2017-10-25 by kaofishy