SpinalHDL/VexRiscv #11: Interactive debug of the simulated CPU

shengbintai, 2018-01-10:
Hello
 I have a question to ask，I am testing the Interactive debug of the simulated CPU .but failed.
log information is at fllowing
```
sbt "run-main vexriscv.demo.GenFull"
cd src/test/cpp/regression
make run DEBUG_PLUGIN_EXTERNAL=yes
```
### peter@ubuntu:~/workspace/VexRiscv/src/test/cpp/regression$ make run DEBUG_PLUGIN_EXTERNAL=yes
verilator -cc  ../../../../VexRiscv.v  -O3 -CFLAGS -std=c++11 -LDFLAGS -pthread  -CFLAGS -DIBUS_CACHED -CFLAGS -DDBUS_CACHED -CFLAGS -DREDO=10 -CFLAGS -pthread -CFLAGS -DTHREAD_COUNT=4 -CFLAGS -DDHRYSTONE -CFLAGS -DCSR -CFLAGS -DISA_TEST -CFLAGS -DMMU -CFLAGS -DMUL -CFLAGS -DDIV -CFLAGS -DDEBUG_PLUGIN -CFLAGS -DDEBUG_PLUGIN_STD -CFLAGS -DDEBUG_PLUGIN_EXTERNAL -CFLAGS -DTRACE_START=0 --gdbbt  -Wno-UNOPTFLAT -Wno-WIDTH --x-assign unique --exe main.cpp
No stack.
make  -j  -C obj_dir/ -f VVexRiscv.mk VVexRiscv
make[1]: Entering directory '/home/peter/workspace/VexRiscv/src/test/cpp/regression/obj_dir'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=0 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DCSR -DDBUS_CACHED -DDEBUG_PLUGIN -DDEBUG_PLUGIN_EXTERNAL -DDEBUG_PLUGIN_STD -DDHRYSTONE -DDIV -DIBUS_CACHED -DISA_TEST -DMMU -DMUL -DREDO=10 -DTHREAD_COUNT=4 -DTRACE_START=0 -pthread -std=c++11   -c -o main.o ../main.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VVexRiscv.cpp VVexRiscv_VexRiscv.cpp > VVexRiscv__ALLcls.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VVexRiscv__Dpi.cpp VVexRiscv__Syms.cpp > VVexRiscv__ALLsup.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=0 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DCSR -DDBUS_CACHED -DDEBUG_PLUGIN -DDEBUG_PLUGIN_EXTERNAL -DDEBUG_PLUGIN_STD -DDHRYSTONE -DDIV -DIBUS_CACHED -DISA_TEST -DMMU -DMUL -DREDO=10 -DTHREAD_COUNT=4 -DTRACE_START=0 -pthread -std=c++11   -c -o VVexRiscv__ALLsup.o VVexRiscv__ALLsup.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=0 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DCSR -DDBUS_CACHED -DDEBUG_PLUGIN -DDEBUG_PLUGIN_EXTERNAL -DDEBUG_PLUGIN_STD -DDHRYSTONE -DDIV -DIBUS_CACHED -DISA_TEST -DMMU -DMUL -DREDO=10 -DTHREAD_COUNT=4 -DTRACE_START=0 -pthread -std=c++11   -c -o VVexRiscv__ALLcls.o VVexRiscv__ALLcls.cpp
      Archiving VVexRiscv__ALL.a ...
ar r VVexRiscv__ALL.a VVexRiscv__ALLcls.o VVexRiscv__ALLsup.o
ranlib VVexRiscv__ALL.a
g++    main.o verilated.o verilated_dpi.o VVexRiscv__ALL.a   -pthread  -o VVexRiscv -lm -lstdc++  2>&1 | c++filt
make[1]: Leaving directory '/home/peter/workspace/VexRiscv/src/test/cpp/regression/obj_dir'
./obj_dir/VVexRiscv
BOOT
```
#Then I did the following
src/openocd -c 'set VEXRISCV_YAML ../VexRiscv/cpu0.yaml' -f tcl/target/vexriscv_sim.cfg
```
### peter@ubuntu:~/workspace/openocd_riscv$ src/openocd -c "set VEXRISCV_YAML ../VexRiscv/cpu0.yaml" -f tcl/target/vexriscv_sim.cfg
Open On-Chip Debugger 0.10.0+dev-00293-ga3421ba (2018-01-10-11:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
../VexRiscv/cpu0.yaml
Info : only one transport option; autoselect 'jtag'
adapter speed: 4000 kHz
adapter_nsrst_delay: 260
jtag_ntrst_delay: 250
Info : set servers polling period to 50ms
Info : clock speed 4000 kHz
Info : TAP fpga_spinal.bridge does not have IDCODE
Info : TAP auto0.tap does not have IDCODE
Info : TAP auto1.tap does not have IDCODE
Info : TAP auto2.tap does not have IDCODE
Info : TAP auto3.tap does not have IDCODE
Info : TAP auto4.tap does not have IDCODE
Info : TAP auto5.tap does not have IDCODE
Info : TAP auto6.tap does not have IDCODE
Info : TAP auto7.tap does not have IDCODE
Info : TAP auto8.tap does not have IDCODE
Info : TAP auto9.tap does not have IDCODE
Info : TAP auto10.tap does not have IDCODE
Info : TAP auto11.tap does not have IDCODE
Info : TAP auto12.tap does not have IDCODE
Info : TAP auto13.tap does not have IDCODE
Info : TAP auto14.tap does not have IDCODE
Info : TAP auto15.tap does not have IDCODE
Info : TAP auto16.tap does not have IDCODE
Info : TAP auto17.tap does not have IDCODE
Info : TAP auto18.tap does not have IDCODE
Info : TAP auto19.tap does not have IDCODE
Warn : Unexpected idcode after end of chain: 21 0xfffff800
Error: double-check your JTAG setup (interface, speed, ...)
Error: Trying to use configured scan chain anyway...
Error: fpga_spinal.bridge: IR capture error; saw 0x0f not 0x01
Warn : Bypassing JTAG setup events due to errors
requesting target halt and executing a soft reset

### #Then VexRiscv log show CONNECTED
......
make[1]: Leaving directory '/home/peter/workspace/VexRiscv/src/test/cpp/regression/obj_dir'
./obj_dir/VVexRiscv
BOOT
CONNECTED

What should I do?

Dolu1990, 2018-01-10:
Hi,

No worries to have, everything is fine, i just tried now with the same commands than you, i get the same result, and then successfuly connect on it with GDB via eclipse :)

So if you want some raw GDB debugging : 
riscv32-unknown-elf-gdb myExecutable.elf
target remote localhost:3333
set remotetimeout 60
set arch riscv:rv32
monitor reset halt
load
continue

If you want some Eclipse debugging : 
https://github.com/SpinalHDL/VexRiscv#using-eclipse-to-run-the-software-and-debug-it

Let's me know if things are going wrong :)
So just for you to know, this VexRiscv testbench has nearly now peripheral mapped in memory, you can get more fun with Murax and Briey testbenches

shengbintai, 2018-01-11:
Hi
 I get it.Thank you so much.

closed 2018-01-11 by shengbintai