SpinalHDL/VexRiscv #12: 48/64 bit instructions

drom, 2018-01-19:
Awesome project!
I am trying to add custom SIMD instructions.
Is it possible to add 48 / 64 bit instructions?
Can `iBus_rsp_inst` bus be extended to 64 / 128?

Dolu1990, 2018-01-19:
Hi
It would mainly require some upgrades on the program counter managment, instruction fetch and instruction decode. Nothing impossible, but would need some works.

I would say for this kind of upgrade the best is to start by patching the uncached instruction implementation to keep things simple (Not the cached one)

Dolu1990, 2018-03-22:
It isn't 48/64 bits. but there is a working RV32IMC in the reworkFetcher branch. All the fetching architecture was reworked, as it was the main obstacle to variable width instruction encoding.