SpinalHDL/VexRiscv #13: Unresolved dependency on eclipse plugin

roman3017, 2018-01-28:
When building I am getting error:
```sh
sbt run
...
[error] sbt.librarymanagement.ResolveException: unresolved dependency: com.typesafe.sbteclipse#sbteclipse-plugin;4.0.0: not found
...
```
Commenting out eclipse from build.sbt and plugins.sbt seems to fix it for me but I wonder what would be the correct way:
```diff
diff --git a/build.sbt b/build.sbt
index 038d4ca..69558d9 100644
--- a/build.sbt
+++ b/build.sbt
@@ -6,10 +6,10 @@ version := "1.0"
 
 scalaVersion := "2.11.8"
 
-EclipseKeys.withSource := true
+//EclipseKeys.withSource := true
 
 libraryDependencies ++= Seq(
   "com.github.spinalhdl" % "spinalhdl-core_2.11" % "0.10.15",

diff --git a/project/plugins.sbt b/project/plugins.sbt
index ef6ceea..e348df3 100644
--- a/project/plugins.sbt
+++ b/project/plugins.sbt
@@ -1,2 +1,2 @@
-addSbtPlugin("com.typesafe.sbteclipse" % "sbteclipse-plugin" % "4.0.0")
+//addSbtPlugin("com.typesafe.sbteclipse" % "sbteclipse-plugin" % "4.0.0")
```

Dolu1990, 2018-01-28:
I think you are using a old version of this repository, as currently it use the 5.2.1. See : 
https://github.com/SpinalHDL/VexRiscv/blob/master/project/plugins.sbt

So just update it or do a git pull would fix it.

Also be sure to have this too : 
https://github.com/SpinalHDL/VexRiscv/blob/master/project/build.properties

roman3017, 2018-01-28:
You are right, it works now. Thank you very much.

closed 2018-01-28 by roman3017