SpinalHDL/VexRiscv #14: GenFullNoMmuNoCache throws exception with fence.i and fence

esromneb, 2018-02-02:
I venerated a Vex core with the `GenFullNoMmuNoCache` option.  I am trying to compile this program: 

https://github.com/gcc-mirror/gcc/blob/master/gcc/testsuite/gcc.c-torture/execute/931002-1.c 

which generates a `fence.i` before the call to `t1()`.  When the core hits this, it jumps to the trap location with `CsrPlugin_mcause_exceptionCode = 2`

How can I reconfigure vex to treat `fence.i` and `fence` as `nop`. Or alternatively is there a gcc compiler flag to prevent it from use `fence.i` instructions.  Thanks.

Dolu1990, 2018-02-03:
I just added a plugin to thread fence.i and fence instruction as nop : 
https://github.com/SpinalHDL/VexRiscv/commit/4729e467631006b6d495ccc32483a906c6aa0a73
Just add it into the list of plugins of GenFullNoMmuNoCache : 
new DummyFencePlugin,
You have to know that the 4 following instruction will not be coherent with the data bus, to work as intended some pipeline flush + jump + timer should be added

Else another way is to support it into the exception handler.

Dolu1990, 2018-02-05:
Sorry i just noticed that i forgot to submit a file XD
Now it's fine.
Also notice that the memory mapping of Briey/Murax has changed, the onChipMemory is now at address 0x80000000

Dolu1990, 2018-02-12:
Let's me know how it is going with torture : D

closed 2018-02-12 by Dolu1990