SpinalHDL/VexRiscv #17: VexRiscV fails to meet timing as soon as you connect it to IBus/DBus memory

ameetgohil, 2018-03-13:
[image]

This on Lattice ecp5 @ 125MHz
Worst path info:
```Path information for path number 1: 
      Requested Period:                      7.264
    - Setup time:                            1.777
    + Clock delay at ending point:           0.000 (ideal)
    = Required time:                         5.487

    - Propagation time:                      6.769
    - Clock delay at starting point:         0.000 (ideal)
    = Slack (critical) :                     -1.282

    Number of logic level(s):                10
    Starting point:                          VexRiscv._zz_237[2] / Q
    Ending point:                            mem.mem0_1_mem0_0_0 / ADA1
    The start point is clocked by            q_engine|clk [rising] on pin CK
    The end   point is clocked by            q_engine|clk [rising] on pin CLKA

Instance / Net                                           Pin      Pin               Arrival     No. of    
Name                                        Type         Name     Dir     Delay     Time        Fan Out(s)
----------------------------------------------------------------------------------------------------------
VexRiscv._zz_237[2]                         FD1S3AX      Q        Out     0.930     0.930       -         
_zz_237[2]                                  Net          -        -       -         -           14        
VexRiscv.un3__zz_408                        ORCALUT4     A        In      0.000     0.930       -         
VexRiscv.un3__zz_408                        ORCALUT4     Z        Out     0.754     1.685       -         
un3__zz_408                                 Net          -        -       -         -           43        
VexRiscv._zz_159_f0                         ORCALUT4     D        In      0.000     1.685       -         
VexRiscv._zz_159_f0                         ORCALUT4     Z        Out     0.608     2.293       -         
_zz_159                                     Net          -        -       -         -           3         
VexRiscv.decode_arbitration_haltItself9     ORCALUT4     B        In      0.000     2.293       -         
VexRiscv.decode_arbitration_haltItself9     ORCALUT4     Z        Out     0.608     2.901       -         
decode_arbitration_haltItself9              Net          -        -       -         -           3         
VexRiscv._zz_99_1_a0_4_2_sx                 ORCALUT4     C        In      0.000     2.901       -         
VexRiscv._zz_99_1_a0_4_2_sx                 ORCALUT4     Z        Out     0.523     3.424       -         
_zz_99_1_a0_4_2_sx                          Net          -        -       -         -           1         
VexRiscv._zz_99_1_a0_4_2                    ORCALUT4     A        In      0.000     3.424       -         
VexRiscv._zz_99_1_a0_4_2                    ORCALUT4     Z        Out     0.523     3.947       -         
_zz_99_1_a0_4_2                             Net          -        -       -         -           1         
VexRiscv._zz_99_1_a0_4_2_RNIU39D2           ORCALUT4     D        In      0.000     3.947       -         
VexRiscv._zz_99_1_a0_4_2_RNIU39D2           ORCALUT4     Z        Out     0.523     4.470       -         
_zz_1_0                                     Net          -        -       -         -           1         
VexRiscv._zz_99_1_a1_RNIEP4H3               ORCALUT4     B        In      0.000     4.470       -         
VexRiscv._zz_99_1_a1_RNIEP4H3               ORCALUT4     Z        Out     0.523     4.993       -         
_zz_sx                                      Net          -        -       -         -           1         
VexRiscv._zz_99_1_a1_RNIUEUP6               ORCALUT4     C        In      0.000     4.993       -         
VexRiscv._zz_99_1_a1_RNIUEUP6               ORCALUT4     Z        Out     0.743     5.736       -         
_zz_99                                      Net          -        -       -         -           34        
VexRiscv.iBus_cmd_payload_pc_m2[2]          ORCALUT4     A        In      0.000     5.736       -         
VexRiscv.iBus_cmd_payload_pc_m2[2]          ORCALUT4     Z        Out     0.523     6.259       -         
iBus_cmd_payload_pc_m2[2]                   Net          -        -       -         -           1         
VexRiscv.iBus_cmd_payload_pc[2]             PFUMX        BLUT     In      0.000     6.259       -         
VexRiscv.iBus_cmd_payload_pc[2]             PFUMX        Z        Out     0.509     6.769       -         
iBus_cmd_payload_pc[2]                      Net          -        -       -         -           18        
mem.mem0_1_mem0_0_0                         DP16KD       ADA1     In      0.000     6.769       -         
==========================================================================================================```

drom, 2018-03-13:
@Dolu1990 Could you recommend some solution to brake this critical path?

Dolu1990, 2018-03-13:
Did you tried to synthetise Murax ?

Dolu1990, 2018-03-13:
With the current version of murax i get 168 Mhz with Murax on speed grade 6
But i spoted that the way the synthetiser is inferring the pc incrementation is dump, so, with a simple syn_keep attribut it speed up to 214 Mhz
I will submit that trick

ameetgohil, 2018-03-13:
Thank you for the quick response. We'll try all the suggestions.
We're not using the Murax. We're generating the VexRiscv without mmu/cache and attaching to our own custom verilog.

Where should I put the syn_keep attribute?

Dolu1990, 2018-03-13:
I just submited the improvment now. With it, the syn_keep attribute will come by itself, you don't need to manualy add it. Let's me know if you have issues later ^^

Dolu1990, 2018-03-13:
But you still need to enable the relaxedPcCalculation (the most impostant). Also adding a pipeline stage to the data bus could help (not the instruction bus)

closed 2018-03-13 by Dolu1990

drom, 2018-03-14:
Thank you for the great RISCV design!

Dolu1990, 2018-03-14:
Hoo wait, i'm not used to diamond tool, and probably readed the timing report wrongly XD I have to look again (facepalm)

Dolu1990, 2018-03-14:
So finaly it look like Murax is going at 135 Mhz, not 214. At which frequancy were going the CPU without the memory attached to it in your case ?

ameetgohil, 2018-03-14:
I'm getting 142MHz with the just the vexriscv w/o memory and relaxedPcCalculation=true