SpinalHDL/VexRiscv #19: Add iCE40 stats to "Area usage and maximal frequency" section?

mithro, 2018-04-24:
VexRiscv looks pretty awesome! 

The [Area usage and maximal frequency](https://github.com/SpinalHDL/VexRiscv#area-usage-and-maximal-frequency) section is really useful for seeing how the parametrization works. 

It also looks like [Murax SoC](https://github.com/SpinalHDL/VexRiscv#murax-soc) supports the iCE40 FPGA.

Would it be possible to add the iCE40 info to the [Area usage and maximal frequency](https://github.com/SpinalHDL/VexRiscv#area-usage-and-maximal-frequency) section too? 

It looks like you already have some details in the Murax SoC section;
```
Murax interlocked stages (0.45 DMIPS/Mhz) ->
  Artix 7    -> 305 Mhz 1004 LUT 1297 FF 
  Cyclone V  -> 160 Mhz 744 ALMs
  Cyclone IV -> 148 Mhz 1,522 LUT 1,255 FF 
  ICE40-HX   ->  51 Mhz 2402 LC (icestorm)

MuraxFast bypassed stages (0.65 DMIPS/Mhz) ->
  Artix 7    -> 312 Mhz 1240 LUT 1330 FF 
  Cyclone V  -> 159 Mhz 884 ALMs
  Cyclone IV -> 142 Mhz 1,755 LUT 1,289 FF 
  ICE40-HX   ->  50 Mhz, 2787 LC  (icestorm)
```

I'm very interested in know how big `VexRiscv smallest` is on the iCE40. It would also be interested to know the biggest configuration of VexRiscv that will fit on the various iCE40 parts.

Thanks!

Dolu1990, 2018-04-25:
Thanks :)

**Would it be possible to add the iCE40 info to the Area usage and maximal frequency section too?**
Yes, i have to do that, it would be also usefull.

**It also looks like Murax SoC supports the iCE40 FPGA.**
Right, tested on the ice40 hx8k-breakout board, also about supporting multiplication on DSP less FPGA, i recently added a iterative multiplication plugin which only use adders (https://github.com/SpinalHDL/VexRiscv#muldiviterativeplugin)

**I'm very interested in know how big VexRiscv smallest is on the iCE40. It would also be interested to know the biggest configuration of VexRiscv that will fit on the various iCE40 parts.**
I never toke a exact look to the VexRiscv area usage into iCE40, so i don't realy know the usage for this FPGA family, but i will add iCE40 into the synthesis benchmark.

mithro, 2018-05-12:
@Dolu1990 It would also be really interested to see this on the iCE40 UltraPlus UP5K device. It's a super low power device but has significantly lower performance.

@daveshah1 added support for these parts to icestorm a while black.

Dolu1990, 2018-05-14:
@mithro I will give a try with the up5k ! Do you have some idea about the performance ratio between the ICE40 HX and the UP5K ?

Dolu1990, 2018-05-14:
@mithro 

```
Murax ->
  iCE40         -> 50 Mhz 2313 LC
  iCE40Ultra -> 18 Mhz 2313 LC
MuraxFast (more dhrystone/cycle)->
  iCE40         -> 49 Mhz 2644 LC
  iCE40Ultra -> 21 Mhz 2644 LC
```

mithro, 2018-05-14:
Is this with the IceCube2 / Radiant or Icestorm?

Dolu1990, 2018-05-14:
All iCE40 numbers on the readme/here are from icestorm. I remember that a 1 years ago when i give some try with icecube2, i got up to 50 % timings improvment on VexRiscv, but maybe that was a special case, i never tried with Murax.

What is the delta between those tools from your experience ?

daveshah1, 2018-05-15:
I would say Radiant/iCEcube2 is about 30% better than the icestorm flow for a typical application.

mithro, 2018-07-21:
This all looks good! Closing the issue.

closed 2018-07-21 by mithro