SpinalHDL/VexRiscv #20: AXI Bridge

yannickl96, 2018-05-24:
Hello, I am new to SpinalHDL but found a liking in your VexRiscv. Since I need an AXI interface to access my memory, it would be nice to just use the core and the AXI bridge that was mentioned in your README, so I can convert your simple bus to AXI. Could you provide me with any info on how to achieve this or where I could learn how to achieve this?
Have a nice day!

Dolu1990, 2018-05-24:
Hi,

you want to generate the Verilog/Vhdl of the CPU and then continue in your flow ? or do a full design in SpinalHDL ?

If that's the first one, there is an example which generate the CPU and rework it's IO into AXI.

https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/VexRiscvAxi4WithIntegratedJtag.scala

Sound good ?

yannickl96, 2018-05-24:
Hello Dolu, I want to generate the Verilog of the CPU and then continue with Vivado to package the core and use it in a block design, therefore your example helped very much. I wonder how I could miss that. Thank you for your quick reply!

closed 2018-05-26 by Dolu1990