SpinalHDL/VexRiscv #22: Briey DE0 Nano SoC, not enough I/O pin locations

Lestandin, 2018-05-29:
Hi,

I intend to test some of the softwares provided for Briey on the DE0 Nano SoC. I first ran the TCL scripts as explained by the readme in the devkits folder, but now when running the fitter on Quartus I get the following logs :

> Info: *******************************************************************
Info: Running Quartus Prime Fitter
	Info: Version 18.0.0 Build 614 04/24/2018 SJ Lite Edition
	Info: Processing started: Tue May 29 11:31:26 2018
Info: Command: quartus_fit --read_settings_files=on --write_settings_files=off DE0_NANO -c DE0_NANO
Info: qfit2_default_script.tcl version: #1
Info: Project  = DE0_NANO
Info: Revision = DE0_NANO
Warning (18236): Number of processors has not been specified which may cause overloading on shared machines.  Set the global assignment NUM_PARALLEL_PROCESSORS in your QSF to an appropriate value for best performance.
Info (20030): Parallel compilation is enabled and will use 8 of the 8 processors detected
Info (119006): Selected device 5CSEMA4U23C6 for design "DE0_NANO"
Info (21077): Low junction temperature is 0 degrees C
Info (21077): High junction temperature is 85 degrees C
Warning: RST port on the PLL is not properly connected on instance pll0:pll0_inst|altpll:altpll_component|pll0_altpll:auto_generated|generic_pll2. The reset port on the PLL should be connected. If the PLL loses lock for any reason, you might need to manually reset the PLL in order to re-establish lock to the reference clock.
	Info: Must be connected
Warning (21300): LOCKED port on the PLL is not properly connected on instance "pll0:pll0_inst|altpll:altpll_component|pll0_altpll:auto_generated|generic_pll1". The LOCKED port on the PLL should be connected when the FBOUTCLK port is connected. Although it is unnecessary to connect the LOCKED signal, any logic driven off of an output clock of the PLL will not know when the PLL is locked and ready.
Info (171003): Fitter is performing an Auto Fit compilation, which may decrease Fitter effort to reduce compilation time
Warning (292013): Feature LogicLock is only available with a valid subscription license. You can purchase a software subscription to gain full access to this feature.
Warning (15714): Some pins have incomplete I/O assignments. Refer to the I/O Assignment Warnings report for details
Info (176045): Design uses memory blocks. Violating setup or hold times of memory block address registers for either read or write operations could cause memory contents to be corrupted. Make sure that all memory block address registers meet the setup and hold time requirements.
Critical Warning (169085): No exact pin location assignment(s) for 138 pins of 154 total pins. For the list of pins please refer to the I/O Assignment Warnings table in the fitter report.
Error (179000): Design requires 154 user-specified I/O pins -- too many to fit in the 145 user I/O pin locations available in the selected device
	Info (179001): Current design requires 154 user-specified I/O pins -- 154 normal user-specified I/O pins and 0 programming pins that have been constrained to use dual-purpose I/O pin locations
	Info (179002): Targeted device has 145 I/O pin locations available for user I/O -- 117 general-purpose I/O pins and 28 dual-purpose I/O pins
Error (12289): An error occurred while applying the periphery constraints. Review the offending constraints and rerun the Fitter.
Info (11798): Fitter preparation operations ending: elapsed time is 00:00:10
Warning (171167): Found invalid Fitter assignments. See the Ignored Assignments panel in the Fitter Compilation Report for more information.
Warning (169064): Following 82 pins have no output enable or a GND or VCC output enable - later changes to this connectivity may change fitting results
Warning (169069): Following 90 pins have nothing, GND, or VCC driving datain port -- changes to this connectivity may change fitting results
Error (11802): Can't fit design in device. Modify your design to reduce resources, or choose a larger device. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number.
Error: Quartus Prime Fitter was unsuccessful. 3 errors, 9 warnings
	Error: Peak virtual memory: 1226 megabytes
	Error: Processing ended: Tue May 29 11:31:48 2018
	Error: Elapsed time: 00:00:22
	Error: Total CPU time (on all processors): 00:00:21

I'm not really used to Quartus and I don't see how I could solve these I/O issue. Moreover when looking to the Compilation Report it states 154/314 for Total Pins and the right device is selected.

Dolu1990, 2018-05-29:
Hi,

Error (179000): Design requires 154 user-specified I/O pins -- too many to fit in the 145 user I/O pin locations available in the selected device

This FPGA => Number of I/O | MCU - 181, FPGA - 145
So only 145 are usable by the FPGA (apparently)

You can remove some of the GPIO (briey as 64 gpio, that's proably a bit much XD)
So i would say, remove all reference to GPIOB  in the Briey.scala should solve the issue.
I should also give more parametrization to the Briey config.

Let's me know how it go ^^

Lestandin, 2018-05-30:
Thanks for the really fast answer, the problem is what I was thinking. For the moment I'll be trying with less GPIO's and see what happens.

I'll let you know if get any good results out of this.

FlyingxDutch, 2019-05-03:
Hello,
I tried to launch BrieySoC (generated from SpinalHDL) on Mimas V.2 FPGA board from "Numato Lab" with Spartan6 (Spartan XC6SLX9 in CSG324 package) and had similiar issues. I remove from project one GPIO port and shortened bus in left GPIOs. Also the fact that Briey.v design required three externals clocks is not helping when one would like to run project on real hardware (FPGA board).
Best Regards