SpinalHDL/VexRiscv #25: Setup instructions don't work

xobs, 2018-06-08:
The setup instructions say to do:

````sh
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 2EE0EA64E40A89B84B2DF73499E82A75642AC823
````

However, this does not actually work:

````sh
xobs@Pazuzu:~$ sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 2EE0EA64E40A89B84B2DF73499E82A75642AC823
Executing: /tmp/apt-key-gpghome.QoPbFivw9k/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv 2EE0EA64E40A89B84B2DF73499E82A75642AC823
gpg: connecting dirmngr at '/tmp/apt-key-gpghome.QoPbFivw9k/S.dirmngr' failed: IPC connect call failed
gpg: keyserver receive failed: No dirmngr
xobs@Pazuzu:~$
````

This prevents the support packages from getting installed further on.

xobs, 2018-06-08:
I was able to install the key by manually copying http://keyserver.ubuntu.com/pks/lookup?op=get&fingerprint=on&search=0x99E82A75642AC823 into a file and doing "sudo apt-key add [file]".

Note that they keyserver.ubuntu.com search is broken -- I wasn't able to search for `2EE0EA64E40A89B84B2DF73499E82A75642AC823`

Dolu1990, 2018-06-08:
That's weird, i made a new ubuntu 18.04 vm, and everything worked fine. I can't reproduce your issue.

xobs, 2018-06-09:
I'm running Ubuntu 18.04 on Linux-on-Windows, so perhaps the "apt-key adv" is expecting some service to be running.  I just performed a fresh install, and got the same issue.  Even if I open up a sudo session (`sudo -i`) and run "dirmngr" by default, running `apt-key adv` fails.

Also, searching for "2EE0EA64E40A89B84B2DF73499E82A75642AC823" at keyserver.ubuntu.com doesn't turn up anything, making it hard to add your key to apt.

Dolu1990, 2018-06-09:
I made a new virutal box vm again, and run directly those command (two times), second time it work ^^
An alternative way to install it, is to install SBT from it's zip (see http://www.codesynch.com/hadoop_spark_blog/sbt-scala-build-tool-unix-install-sbteclipse-plugin-sample-wordcount/)

Else all the toolchain can also work on Windows, including verilator. (Verilator install => http://spinalhdl.github.io/SpinalDoc/spinal/sim/introduction/#windows, sbt install from MSI)

xobs, 2018-06-09:
The command that worked for me was:

````sh
curl 'http://keyserver.ubuntu.com:11371/pks/lookup?op=get&search=0x99E82A75642AC823' | sudo apt-key add -
````

This issue can probably be closed, because it's likely due to a misconfiguration on Ubuntu-on-Windows.  Hopefully the solution posted here will help others who have run into the same issue.  Thanks for taking a look!

closed 2018-06-09 by xobs

Dolu1990, 2018-06-09:
Ok, thanks for the feedback :D