SpinalHDL/VexRiscv #26: Setup instructions: scala is a requirement

xobs, 2018-06-08:
````sh
xobs@Pazuzu:/mnt/c/Users/smcro/Documents/Code/fpga/SpinalHDL$ sbt clean compile publish-local
Copying runtime jar.
Getting org.scala-sbt sbt 0.13.16  (this may take some time)...

:: problems summary ::
:::: WARNINGS
                module not found: org.scala-sbt#sbt;0.13.16

        ==== local: tried

          /home/xobs/.ivy2/local/org.scala-sbt/sbt/0.13.16/ivys/ivy.xml

          -- artifact org.scala-sbt#sbt;0.13.16!sbt.jar:

          /home/xobs/.ivy2/local/org.scala-sbt/sbt/0.13.16/jars/sbt.jar

        ==== local-preloaded-ivy: tried

          file:////home/xobs/.sbt/preloaded/org.scala-sbt/sbt/0.13.16/ivys/ivy.xml

        ==== local-preloaded: tried

          file:////home/xobs/.sbt/preloaded/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.pom

          -- artifact org.scala-sbt#sbt;0.13.16!sbt.jar:

          file:////home/xobs/.sbt/preloaded/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.jar

        ==== Maven Central: tried

          https://repo1.maven.org/maven2/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.pom

          -- artifact org.scala-sbt#sbt;0.13.16!sbt.jar:

          https://repo1.maven.org/maven2/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.jar

        ==== sbt-maven-releases: tried

          https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.pom

          -- artifact org.scala-sbt#sbt;0.13.16!sbt.jar:

          https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.jar

        ==== sbt-maven-snapshots: tried

          https://repo.scala-sbt.org/scalasbt/maven-snapshots/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.pom

          -- artifact org.scala-sbt#sbt;0.13.16!sbt.jar:

          https://repo.scala-sbt.org/scalasbt/maven-snapshots/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.jar

        ==== typesafe-ivy-releases: tried

          https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/0.13.16/ivys/ivy.xml

        ==== sbt-ivy-snapshots: tried

          https://repo.scala-sbt.org/scalasbt/ivy-snapshots/org.scala-sbt/sbt/0.13.16/ivys/ivy.xml

                ::::::::::::::::::::::::::::::::::::::::::::::

                ::          UNRESOLVED DEPENDENCIES         ::

                ::::::::::::::::::::::::::::::::::::::::::::::

                :: org.scala-sbt#sbt;0.13.16: not found

                ::::::::::::::::::::::::::::::::::::::::::::::

:::: ERRORS
        Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo1.maven.org/maven2/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.pom

        Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo1.maven.org/maven2/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.jar

        Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.pom

        Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.jar

        Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo.scala-sbt.org/scalasbt/maven-snapshots/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.pom

        Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo.scala-sbt.org/scalasbt/maven-snapshots/org/scala-sbt/sbt/0.13.16/sbt-0.13.16.jar

        Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/0.13.16/ivys/ivy.xml

        Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo.scala-sbt.org/scalasbt/ivy-snapshots/org.scala-sbt/sbt/0.13.16/ivys/ivy.xml

:: USE VERBOSE OR DEBUG MESSAGE LEVEL FOR MORE DETAILS
unresolved dependency: org.scala-sbt#sbt;0.13.16: not found
Error during sbt execution: Error retrieving required libraries
  (see /home/xobs/.sbt/boot/update.log for complete log)
Error: Could not retrieve sbt 0.13.16
xobs@Pazuzu:/mnt/c/Users/smcro/Documents/Code/fpga/SpinalHDL$
````

Installing scala with `sudo apt-get install scala` does not solve the issue.

xobs, 2018-06-08:
This is due to a conflict between `sbt` and Ubuntu 18.04.  Workaround: https://github.com/sbt/sbt/issues/2295#issuecomment-394490499

Dolu1990, 2018-06-08:
Same than for the other issue,  i made a new ubuntu 18.04 vm, and everything worked fine. I can't reproduce your issue. (openjdk 8)

Anyway, the current VexRiscv master commit doesn't need to build the SpinalHDL library.
Also, you don't need to install Scala :) SBT will download scala himself.

Dolu1990, 2018-06-11:
Is it ok to close the issue ?

xobs, 2018-06-12:
I'm trying it again on a fresh install.  The behavior I'm seeing at the moment is that `sbt "run-main vexriscv.demo.GenFull"` gives no output but does spawn a `java` binary that uses a lot of CPU and doesn't exit when I hit Control-C.

````sh
xobs@Cuboid:~$ java -version
openjdk version "1.8.0_171"
OpenJDK Runtime Environment (build 1.8.0_171-8u171-b11-0ubuntu0.18.04.1-b11)
OpenJDK 64-Bit Server VM (build 25.171-b11, mixed mode)
xobs@Cuboid:~$
````

Dolu1990, 2018-06-12:
You are doing this command in the root of the VexRiscv repository right ?
Maybe the ubuntu on windows is broken ? I never used it, i don't realy know.

The issue that i have is that i can't reproduce the issue.
Maybe you can give a try on raw windows ?
What happend if you just try the sbt command without arguments ?

xobs, 2018-06-13:
It does work after a reinstall.  Apologies, I must have somehow had a corrupted Ubuntu 18.04 install.  Thank you for your support.

closed 2018-06-13 by xobs