SpinalHDL/VexRiscv #28: Can VexRISCV based SoC be made to fit on an iCE40 1k?

mithro, 2018-06-25:
So, this is a pretty big challenge I think....

The iCE40 1K has the following resources;
  * 1280 Logic Cells
  * 64kbits Embedded RAM bits
  * 1 PLL

However, there might be hope, your stats seem to indicate the following;
```
VexRiscv smallest (RV32I, 0.52 DMIPS/Mhz, no datapath bypass, no interrupt) ->
  iCE40      -> 81 Mhz 1130 LC

VexRiscv smallest (RV32I, 0.52 DMIPS/Mhz, no datapath bypass) ->
  iCE40      -> 71 Mhz 1278 LC
```

But that is obviously *only* the CPU.

The SoC seems to use about double the LC;
```
Murax interlocked stages (0.45 DMIPS/Mhz) ->
  ICE40-HX   ->  51 Mhz 2402 LC (icestorm)

MuraxFast bypassed stages (0.65 DMIPS/Mhz) ->
  ICE40-HX   ->  50 Mhz, 2787 LC  (icestorm)
```

Thoughts?

Dolu1990, 2018-06-27:
Actualy, you VexRiscv smallest no interrupt could be a little smaller if you set the earlyInjection from the DBusSimplePlugin. It will write the data read access back into the pipeline into the memory stage instead of the writeBack stage.
It reduce the sice to 1064 LC.

Then someother optimisation have impacte on none ICE40 FPGA, as in ice40 if you use only the LUT of an LC you lose the FF (and the reverse too)

Hope it can help ^^

Dolu1990, 2018-09-23:
Would maybe be possible to have a 4 stage RISCV (no memory stage) + moving the register file read into the execute stage to avoid having to pipeline the reg files values to the execute stage.

It would for sure benefits regular FPGA area, but for ice40 the fact that LUT can't be reused if the FF of the LE is used nullify many possible optimisation.

Dolu1990, 2018-10-03:
Doing some experiments, by removing 1 fetch stage and moving the register file read into the execute stage i got : 

MicroNoCsr ->
iCE40 -> 57 Mhz 999 LC   <- 999 is actualy smaller than 1K XD
Artix 7 -> 330 Mhz 537 LUT 389 FF 
Cyclone V -> 144 Mhz 366 ALMs
Cyclone IV -> 120 Mhz 714 LUT 314 FF

mithro, 2018-10-03:
@Dolu1990 That is pretty awesome! Just need to figure out how to shove the rest of a SoC into the 1k too :-P

regymm, 2021-10-06:
I've just fitted the GenSmallestNoCsr into iCE40-HX1K with 2KB ROM, 4KB RAM, UART RX, and some memory-mapped GPIO for processing VT100 sequences -- VexRISCV is very promising!

The most difficult part for me is the block ram to iBus/dBus interface, as I didn't find any detailed documents about the buses. So it'll be nice if there're some examples for directly using block ram for instruction/data(Murax's arbitrator seems too much for the smallest design).

Dolu1990, 2021-10-11:
hmm using a true dual ported ram would produce the best results i guess.

For the ibus => https://github.com/SpinalHDL/VexRiscv#ibussimpleplugin

Stream and Flow are specified in the SpinalDoc.

For the dbus => https://github.com/SpinalHDL/VexRiscv#dbussimpleplugin

It is nearly similar, excepted for the rsp arbitration.