SpinalHDL/VexRiscv #32: What is the difference between iCE40HX8K-EVB and iCE40-hx8k_breakout_board

mithro, 2018-07-21:
My board says;
```
iCE40-HX8K Breakout Board
Rev A P/N: ICE40HX8K-B-EVN
```
Not sure which one that is... However iCE40HX8K-EVB uses `iceprogduino` while iCE40-hx8k_breakout_board uses `iceprog` which works with my board.

If this is all correct, once you run `iceprog` what is suppose to happen?

I'll try and send an updated README once I have this all figured out...

mithro, 2018-07-21:
I /think/ you want to connect to `/dev/ttyUSB1` and get something?

mithro, 2018-07-21:
```
    .io_uart_txd(io_B12),
    .io_uart_rxd(io_B10)
```

mithro, 2018-07-21:
The trick seems to be the way the jumpers on the board need to be configured.

The board seems to need to be in the "SPI Peripheral Mode" as described below;
> In SPI Peripheral Mode the SPI signals loads the program file into the CRAM (configuration ram) of the FPGA directly. In this mode the FPGA will lose its configuration when power is removed and must be re-configured. Jumpers must be in locations J6:1-2 and J6:3-4. Jumper J7 is not installed. See Figure 5

Dolu1990, 2018-07-21:
iCE40HX8K-EVB is another board which was added by somebody else, nothing related to the official hx8k breakout one.

Once you run the iceprog, the led[6:0] should count up, and the led 7 should blink. You should also be able to have the uart echo by using the right channel of the USB. It is 115200 baud rate with single top bit no parity.

I remember having issue when using the USB for both programming and serial uart, but i was on a virtual box machine running linux.

Let's me know if there is issues

PR #33 (2018-07-21): Improve the Murax example for the iCE40-hx8k_breakout_board
 * Small fixes to the Makefile
 * Adding README with images.

Pretty much fixes #32.
closed 2018-07-21 by Dolu1990