SpinalHDL/VexRiscv #35: Problems when writing to memory

yannickl96, 2018-07-26:
Hello, I have a little problem while using your processor and maybe you can point me into the right direction. 
So consider the following setup: I use a PYNQ-Z1 board with a xc7z020clg400-1 FPGA from Xilinx. On this I have an instance of your VexRiscv, built with the AXI and JTAG extension (from your demo folder). The iBusAxi is connected to a AXI BRAM Controller which is connected to a Block Memory Generator with 16KiB memory. The dBusAxi is connected to a separate controller and a separate BRAM, therefore instruction and data memory are separated. The data memory is mapped to address 0x4000. So what I basically want to achieve, is to write the integer 0xDEADBEEF to the address 0x4000 (first position in data memory). I wrote a C program for that and compiled it with the riscv32-unknown-elf-gcc with march=rv32im and mabi=ilp32. The code is then turned into binary format by invoking riscv32-unknown-elf-objcopy -O binary [...]. The relevant assembler code looks like this:
`                                               lui	a5,0x4
   10068:	fef42623          	sw	a5,-20(s0)
   1006c:	fec42783          	lw	a5,-20(s0)
   10070:	deadc737          	lui	a4,0xdeadc
   10074:	eef70713          	addi	a4,a4,-273 # deadbeef <__global_pointer$+0xdeaca573>
   10078:	00e7a023          	sw	a4,0(a5) # 4000 <_start-0xc054>
`
Now what happens is, that the processor writes to 0x4000 in the memory, but unfortunately it writes the bytes 0xFFFFFEEF. It would be nice if you could provide any idea what could be going wrong. 
Cheers!

yannickl96, 2018-07-28:
Hi,
thanks for your reply. The binary is loaded to the instruction BRAM by using AXI. The iBus is connected to the same BRAM. Then I have another BRAM which is connected to the dBus. To this BRAM I want to write the value.

Dolu1990, 2018-07-28:
hmm Do you have a wave to share ? vcd file ?

yannickl96, 2018-07-28:
I just looked through the whole assembler again and noticed that the stack pointer is not initialized (which makes sense, since I don't have an OS running on the processor). Therefore the processor wants to write to an address at 0xFFFF..... which is not mapped. So I'll edit the linker script to provide a stack pointer that is initialized fitting to the size of my BRAM. I'll let you know how it goes.
Thank you for your time, anyways.

Dolu1990, 2018-07-28:
Have you checked https://github.com/SpinalHDL/VexRiscvSocSoftware/tree/master/projects/murax/demo/src ? There is maybe some usefull ressources to setup things.

yannickl96, 2018-07-28:
Thank you, I'll take a look :)

closed 2018-08-10 by Dolu1990