SpinalHDL/VexRiscv #36: QSys interprets clock and reset as conduits

kaofishy, 2018-08-02:
Hi,

After compiling the VexRiscvAvalonForSim example, I imported the TCL component into Quartus but trying to instantiate it inside Platform Designer I receive the following errors:

```
Error: unsaved.VexRiscv_0: VexRiscv_0.debug_resetOut has an associatedClock of "clk" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.timerInterrupt has an associatedReset of "reset" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.timerInterrupt has an associatedClock of "clk" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.externalInterrupt has an associatedReset of "reset" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.externalInterrupt has an associatedClock of "clk" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.iBusAvalon has an associatedReset of "reset" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.iBusAvalon has an associatedClock of "clk" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.dBusAvalon has an associatedReset of "reset" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.dBusAvalon has an associatedClock of "clk" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.debugBusAvalon has an associatedReset of "debugReset" with incompatible type "conduit"
Error: unsaved.VexRiscv_0: VexRiscv_0.debugBusAvalon has an associatedClock of "clk" with incompatible type "conduit"
```
This happens with VexRiscvAvalanWithIntegratedJtag as well. I am using Quartus II 18.0 .

Tony

Dolu1990, 2018-08-05:
Hi,

When i made this autogenerated TCL stuff that was with Quartus II 14.1 (if i remember well) So maybe they had some changes which broke some of the compatibility ?

I will check what happened.

Charles

Dolu1990, 2018-08-05:
Ok so in fact i made it for qsys 13.1 ^^

Anyway, i tried with 15.1, and i remember it worked with it, and it is broken, something happend on my side, i will lets you know when it is fixed.

kaofishy, 2018-08-05:
No worries, I'm working with a MAX10 (very cheap dev boards!) so unfortunately 13.1 doesn't work for me.

Dolu1990, 2018-08-05:
I identified the issue and submited a fix to the SpinalHDL repo. but currently it isn't released.

But you can already use it : 
- clone the SpinalHDL "dev" branch, go in it, do a sbt clean publish-local
- Update the SpinalHDL versions in the VexRiscv/build.sbt file to be 1.1.7
- do a sbt clean in the VexRiscv repo
- regenerate the VexRiscv hardware and now that should be fine ^^

By importing things in Qsys 17.0, things look fine for me, should be fine for qsys 18.0.

Just be carefull with resets : 
[image]

kaofishy, 2018-08-05:
Just tried it in 18.0 and it works! Thanks for the quick response.

closed 2018-08-05 by kaofishy