SpinalHDL/VexRiscv #38: File name <> Module name inconsistent

tomverbeure, 2018-08-08:
File name: GenSmallAndPerformant.scala
Module name: GenSmallAndProductive

Same thing for: GenSmallAndProductiveWithICache

Which is the right name: *performant or *productive?

Once I know, I'll submit a patch.

Tom

tomverbeure, 2018-08-08:
Also: the GenFullNoMmuNoCache.scala and GenNoCacheNoMmuMaxPerf.scala files are identical.

Dolu1990, 2018-08-10:
Right, some of the naming is twisted, including with the one reported and not 100% coherent with the one reported in the readme. I will have to fix it.
For  GenFullNoMmuNoCache.scala and GenNoCacheNoMmuMaxPerf.scala, the difference is in the branch plugin. DYNAMIC_TARGET vs STATIC prediction, isn't it ?