SpinalHDL/VexRiscv #41: Processor not writing to data memory

yannickl96, 2018-10-19:
Hello everyone. I seem to have encountered a bug when I try to use the VexRiscv processor in a Vivado block design. The benchmarks dhrystone and towers from https://github.com/riscv/riscv-tests/tree/master/benchmarks work fine, however when I try to run the multiply benchmark, the processor only reads from the data memory and never writes. This gets even weirder because the addresses that it reads, are the addresses that it shall write to. The dBusAxi is connected to an AXI Bram controller IP which is connected to a block memory generator with 32KiB of memory. I attached the data that I gathered with an ILA core and the assembler generated by the RISC-V GCC. Please let me know if you need any more info.

Cheers!
[multiply_vexriscv.zip](https://github.com/SpinalHDL/VexRiscv/files/2496739/multiply_vexriscv.zip)
[multiply_vexriscv.txt](https://github.com/SpinalHDL/VexRiscv/files/2496745/multiply_vexriscv.txt)

Dolu1990, 2018-10-21:
Hi,

Would it be possible to have the vcd with all the signals ? (including the CPU signals)

Thanks :)

yannickl96, 2018-10-21:
You mean the internal processor signals?

Dolu1990, 2018-10-21:
Yes ^^

Dolu1990, 2018-10-21:
Hooo are you debugging directly on the physical target ?

yannickl96, 2018-10-21:
Yes, sir ^^. In theory, the processor should work fine (at least according to the regression tests). However, on the Xilinx PYNQ-Z1 there are some problems.

yannickl96, 2018-10-29:
@Dolu1990 I have to add the signals post-synthesis which results in about 1200 signals (and my PYNQ doesn't have nearly enough BRAM to save them all). Are there any signals which are more helpful to you than others?

yannickl96, 2018-10-29:
Thank you for the hint! I will give it a try.

Dolu1990, 2018-10-29:
@yannickl96 Sorry had lost the track of this issue.

So, the thing is you realy need to do a simulation to sort this kind of things out, especialy if you have a whole system/third party stuff in the project. Debuging it the half blind is realy a paine full process :)

About synthesis issues, i have to say that this kind of issue already happend to me on Zynq. That's quite a paine in the ass if that's the case there :/

yannickl96, 2018-10-29:
Well, it would not be the first time that Vivado would do something unfortunate. -.-^^

yannickl96, 2018-11-15:
@Dolu1990 Did your synthesis issue on your Zynq by any chance emerge due to an error in the memory map?

Dolu1990, 2018-11-15:
Yes, exactly, it was the issue. Bad memory synthesis, and it was somehow quite random.

closed 2019-05-25 by Dolu1990