SpinalHDL/VexRiscv #42: RV32E support

xobs, 2018-11-08:
After a discussion with @Dolu1990 , it was revealed that 50% of the gate count of a minimal VexRiscv is the register file, which is 32x32-bits.  That's many registers.

The RV32E version of the specification cuts the number of registers down to 16, which should shrink the number of gates required by ~25%.

How much work is required to add support for RV32E?

Dolu1990, 2018-11-13:
Got a two stages VexRiscv running zephyr and compliance tests :)

tomverbeure, 2018-11-13:
> Got a two stages VexRiscv running zephyr and compliance tests :)

Very nice work!

FYI: Clifford Wolf checked in a riscv-formal change for a serial RISC-V CPU => super small size.

Dolu1990, 2018-11-14:
Hooo yeah, it will be a funny one XD