SpinalHDL/VexRiscv #43: Can the Briey SoC run on a iCE40-HX8K-CT256?

ndbroadbent, 2018-11-22:
The README seems to suggest that the Murax SoC is the only one that can be run on an iCE40 FPGA. 

But I was a bit confused by the Briey SoC timings and area:

```
  Artix 7    -> 239 Mhz 3227 LUT 3410 FF
  Cyclone V  -> 125 Mhz 2,207 ALMs
  Cyclone IV -> 112 Mhz 4,594 LUT 3,620
```

I think I am mistunderstanding these numbers, because the iCE40-HX8K chip has 7,680 LUTs, which is greater than 3227, 2,207, and 4,594.

I'm new to FPGAs so sorry for the question. But how many LUTs does the Briey SoC require?

(I've just bought a iCE40-HX8K breakout board. But I might also get a Artix 7 (or Zynq 7020) board to play with VexRiscv.)

Dolu1990, 2018-11-22:
So, there is three potential things about the ICE40 : 
- Each LC has only one output, can be the FF or the LUT, but not both, so if you do much pipelining with register, you lose a lot of LUT that you can't reuse (in Cyclone IV you can)
- Icestorm could fail with the place and route when the occupancy is high (if you consider the above point, it is very likely the case)
- Maybe it failed because the SoC use to many IO (GPIO)

I have to check it ^^

Maybe this could interrest you, there is Up5kPerf and Up5kArea : 
https://github.com/SpinalHDL/riscvSoftcoreContest

ndbroadbent, 2018-11-22:
Thanks @Dolu1990! That makes sense about the place and route failing when the occupancy is high. I didn't realize there would need to be some leftover LUTs as a "buffer" so that the circuit can fit.

I really like the open source tooling around Project IceStorm, so I think I will start with the iCE40 chips, and I'll just be happy with the Murax SoC for now. But maybe in the future I will also get a snickerdoodle black (Xilinx Zynq-7020 SoC with an Artix 7), and then I can play around with the full RISC-V CPU (and run it at a higher speed.) 

Also the riscvSoftcoreContest looks awesome! Would these also run on a iCE40-HX8K?

I'm a bit confused - From what I read it sounds like the iCE40 UltraPlus is newer and faster, but it only has 5k LUTs? But the HX8K has 8k LUTs. So it's hard for me to tell which one is better.

Would you recommend that I buy a development board with an UltraPlus or a HX8K?

Dolu1990, 2018-11-22:
So, to run on iCE40 HX8K, you would probably need to enable the instruction cache or having a faster XIP flash controller. You can still run the code from RAM, but there is not so much of it in the HX8K ^^
About the UltraPlus (UP5K) That's probably the slowest modern FPGA that you can get, they are about 3 time slower than the HX8K.
They have only 5K lut but they have a loooooooot of ram via their SPRAM. which is "ground breaking XD"
Yesteday i made some power consumption with murax on it =>  1.2V comsumption was 4 mA to run at 12 Mhz, which is realy not much, comparable to ASIC MCU, and 200 uA when the clock was off. That's the point of Up5k : ram and consumption.

So between ultraplus and hx8k, honnestly, i don't know what is the best, the memory ressource of the HX8K hurt a bit, but you can have external memories, that's fine ^^ and the hx8k is way faster.
If you want to do a tiny tiny thing without minimal external component with low power => Up5k

ndbroadbent, 2018-11-22:
Interesting, thanks so much! That was really helpful!

I'm just learning, so I don't really need a lot of speed or RAM. So it sounds like the iCE40 HX8K is still a great chip to learn on. Thanks again!

closed 2018-11-22 by ndbroadbent