SpinalHDL/VexRiscv #48: JTAG/Reset Lock Up (Avalon)

cutephoton, 2018-12-22:
So I've been banging my head on the desk trying to understand the debug/jtag. I think the problem is related to reset.

For reference, this is my qsys diagram:

[image]

`monitor reset halt` -> locks up
`monitor reset resume` -> locks up
`monitor halt` -> works
`monitor resume` -> works

This is what halt/resume cycles look like:

[image]

And now `monitor reset run/halt`

[image]

Thoughts?

OCD config:
```
#-----------------------------------------
# JTAG Adapter
#
interface ftdi
ftdi_vid_pid 0x0403 0x6014
#ftdi_layout_init 0x0c08 0x0f1b
ftdi_layout_init 0xfff8 0xfffb
ftdi_layout_signal nTRST -data 0x0100 -oe 0x0100
ftdi_layout_signal nSRST -data 0x0200 -oe 0x0200
transport select jtag

#-----------------------------------------
# Target
#
set  _ENDIAN little
set _TAP_TYPE 1234
set _CPUTAPID 0x10001FFF

adapter_khz 4000
# have tried 25,100,1000 too
adapter_nsrst_delay 260
jtag_ntrst_delay 250

set _CHIPNAME fpga_spinal
jtag newtap $_CHIPNAME bridge -expected-id $_CPUTAPID -irlen 4 -ircapture 0x1 -irmask 0xF

target create $_CHIPNAME.cpu0 vexriscv -endian $_ENDIAN -chain-position $_CHIPNAME.bridge -coreid 0 -dbgbase 0xF00F0000
vexriscv readWaitCycles 12
vexriscv cpuConfigFile ../cpu0.yaml

poll_period 50

init
#echo "Halting processor"
soft_reset_halt
```
GDB:
```
target remote localhost:3333
set arch riscv:rv32
monitor reset halt
load
continue
```

Dolu1990, 2018-12-22:
Hi,
One thing, acordignly to https://github.com/SpinalHDL/VexRiscv#cpu-clock-and-resets, is that all the peripherals and the interconnect is also reseted by the debug_resetOut.

The reason of that is, if the CPU get a reset while a transaction was pending, and the arbitration isn't reseted, the interconnect/peripherals can get a lockup.

Maybe that's the issue :)
Let's me know.

Dolu1990, 2018-12-22:
Also, what is your IBus configuration ? if that's the uncached one, it is important to have https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/IBusSimplePlugin.scala#L160 set to true.

cutephoton, 2018-12-22:
Well, I think this is it:

`case plugin: DebugPlugin => {`
should be:
`case plugin: DebugPlugin => plugin.debugClockDomain {`

That really broke my head. This problem seems to exist in both avalon demos. I'll try prepare a pull request once I recover from this HDL hangover. :)

cutephoton, 2018-12-22:
FYI, was able to get cache up and running. Pretty satisfying. Thanks for the help.

cutephoton, 2018-12-23:
Maybe I spoke too soon. I'm finding with the cache enabled that after stepping the first two instructions (GP, and SP loads) that reads appear to be referencing data 2 words off. If I issue a write, it does write back to the correct address.

memory map:
```
                0x0000000080000200                time_t1        
                0x0000000080000204                time_t0              -> 0xF0002010
                0x0000000080000208                time_systick      -> 0xF0002000
                0x000000008000020c                gpio_a               -> 0xF0003010
                0x0000000080000210                gpio_led            -> 0xF0003000
```

GDB:

```
(gdb)     monitor reset halt
JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
(gdb)     load
Loading section .vector, size 0x104 lma 0x80000000
Loading section .memory, size 0x80 lma 0x80000104
Loading section .text.startup, size 0x58 lma 0x80000184
Loading section .rodata, size 0x24 lma 0x800001dc
Loading section .data, size 0x18 lma 0x80000200
Start address 0x80000000, load size 536
Transfer rate: 52 KB/sec, 107 bytes/write.
(gdb) print &gpio_led
$7 = (volatile struct IO_gpio_registers **) 0x80000210 <gpio_led>
(gdb) x /1ix 0x80000210
0x80000210 <gpio_led>:	0xf0003000                                        **CORRECT**
(gdb) set gpio_led->value=0xFF
(gdb) step
crtInit () at src/crt.S:64
64	  mret                                                           *exiting reset*
(gdb) step
77	     la gp, __global_pointer$
(gdb) x /1ix 0x80000210
0x80000210 <gpio_led>:	0xf0003010                                         **now reading gpio_a**
(gdb) step
crtInit () at src/crt.S:84
84	  la sp, _stack_start
(gdb) x /1ix 0x80000210
0x80000210 <gpio_led>:	0xf0003010
(gdb) step

Breakpoint 1, 0x800000c0 in bss_init () at src/crt.S:84
84	  la sp, _stack_start
(gdb) x /1ix 0x80000210
0x80000210 <gpio_led>:	0xf0002000                                        **now time_systick**
```
Here are the settings - is there some magic setting I'm missing?
```
          new IBusCachedPlugin(
            resetVector = 0x80000000l,
            prediction = STATIC,
            config = InstructionCacheConfig(
              cacheSize = 4096,
              bytePerLine =32,
              wayCount = 1,
              addressWidth = 32,
              cpuDataWidth = 32,
              memDataWidth = 32,
              catchIllegalAccess = true,
              catchAccessFault = true,
              catchMemoryTranslationMiss = true,
              asyncTagMemory = false,
              twoCycleRam = true,
              twoCycleCache = true
            )
            //            askMemoryTranslation = true,
            //            memoryTranslatorPortConfig = MemoryTranslatorPortConfig(
            //              portTlbSize = 4
            //            )
          ),
          //                    new DBusSimplePlugin(
          //                      catchAddressMisaligned = true,
          //                      catchAccessFault = true
          //                    ),
          new DBusCachedPlugin(
            config = new DataCacheConfig(
              cacheSize         = 4096,
              bytePerLine       = 32,
              wayCount          = 1,
              addressWidth      = 32,
              cpuDataWidth      = 32,
              memDataWidth      = 32,
              catchAccessError  = true,
              catchIllegal      = true,
              catchUnaligned    = true,
              catchMemoryTranslationMiss = true
            ),
            memoryTranslatorPortConfig = null
            //            memoryTranslatorPortConfig = MemoryTranslatorPortConfig(
            //              portTlbSize = 6
            //            )
          ),
```

cutephoton, 2018-12-24:
Hmm... I disabled the Dcache but I still get read issues wrt the debugger. I'm running at 80MHz (up from 50MHz). At some point, I had it running reliably at 125MHz without the cache. No timing violations noted. Peripheral bus is 50MHz.

```
(gdb) x /8ix 0x80000200
0x80000200 <time_t1>:	0x00000000	0x00000000	0xf0002000	0x00000000
0x80000210 <gpio_led>:	0x00000000	0x00000000	0x00000000	0x00000001
(gdb) x /8ix 0x80000200
0x80000200 <time_t1>:	0xf0002200	0x00000000	0x00000000	0x00000000
0x80000210 <gpio_led>:	0xf0003000	0x00000000	0x00000000	0x00000001
```

Dolu1990, 2018-12-24:
There is something getting wrong :/

So, by hazard, which Altera kit are you using ? i have some altera device here, DE1-SOC. Maybe i can give a try with your project on it ?

Becaus has far i know, the debug implementation was stable one all target i tried.

What JTAG bridge are you using ?

cutephoton, 2018-12-24:
I do have the DE1-SOC but been using Max10s lately. Besides, where's the fun if just works?

It's all good now. I hope. It was (maybe) Intels fault! (I updated the top of the last comment, but maybe it wasn't clear.)  QSys started throwing null pointer exceptions and other strange behaviors (UIs not working). The SignalTap output was surprising. ¯\_(ツ)_/¯ that's all I got.

Dolu1990, 2018-12-24:
hmm
Ok, let's me know if things a turning bad again, i just tested in sim the debug with the Briey SoC which also use caches, (haven't use it since at least 1 year) and it worked fine :)

Dolu1990, 2018-12-31:
Hooo yes, you are right, i just fixed the issue in the dev branch of SpinalHDL.
Do you know the flow to use the unreleased stuff ?

Dolu1990, 2018-12-31:
Stupid mistake :)

cutephoton, 2018-12-31:
Thanks. I think we can mark this one resolved, eh? A good way to end the year!

:balloon:  :balloon: :balloon:  :balloon:

closed 2018-12-31 by cutephoton

Dolu1990, 2018-12-31:
Yes ^^