SpinalHDL/VexRiscv #51: Can't access the code memory when a breakpoint is set on it

LukasJaegerSIT, 2019-01-10:
Greetings!
I am basing my cybersecurity research on VexRiscV and need to build a firmware that can hash parts of itself. I noticed that the hash algorithm failed when a certain point in the code was reached and I noticed that this point had the same address as the breakpoint I had set for debugging. I've stripped away the hash code and tried this on a completely unmodified branch of VexRiscV (Murax, simulation only):

```c
// Here we copy program code with a breakpoint
// into another part of the RAM.
uint8_t * ram_ptr = (uint8_t*) (0x80000450);
uint8_t arr [32] = {0}; 
for (int i = 0; i < 32; i++){
	arr[i] = ram_ptr[i];
}
``` 

I noticed that the word at the address where the breakpoint was set (0x80000454) was not copied correctly. Instead of the expected piece of RISC-V assembly, I got those four bytes: 0x73, 0x0, 0x10, 0x0. I've attached the full demo if you want to check this for yourself.
[ram_err_demo.tar.gz](https://github.com/SpinalHDL/VexRiscv/files/2745290/ram_err_demo.tar.gz)

The steps I executed:

* Built Murax using `sbt`
* Started Verilator
* Started `gdb` and connected to the remote target
* Did `monitor reset halt` and `load`
* Set a breakpoint at `main.c:46` and continued
* Printed `arr`

This is a pretty special use case but for me it is crucial that I can access the code memory and hash it without that glitch. Do you have any idea what could cause this? I will look into the debugging plugin myself and try to determine the reason but if you know it from the top of your head it would be a great help. If I can find a solution, I will inform you and create a pull request.

Dolu1990, 2019-01-10:
Hi,

That's not a bug. What is happening, is for Murax, the code is executed from RAM, so GDB is using software breakpoints, it modify the programm memory with a RISC-V EBREAK (0x00100073, as you have seen) in order to make the CPU break when the instruction is executed.

To avoid that you have to do two things : 
- Adding hardware breakpoints in Murax by forcing https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Murax.scala#L70 to 3
- Modify the linkerscript stuff of your software to define the code region as a read-only one. (it will force GDB to use hardware breakpoint)

Does that seem good ?

LukasJaegerSIT, 2019-01-10:
Ah... that makes sense. I will try hardware breakpoints. Thank you very much.

closed 2019-01-10 by LukasJaegerSIT

Dolu1990, 2019-01-10:
Just be sure to have pull the openocd_riscv again, and recompile it to have the last version :)