SpinalHDL/VexRiscv #54: The smallest verilog simulation does not commit SW before RET. Anything wrong?

wye1102, 2019-01-23:
The disassembled code is as follows:
0xbfc80268:  0007a703          lw            a4,0(a5)
0xbfc8026c:  00071863          bnez          a4,16                           # 0xbfc8027c
0xbfc80270:  800007b7          lui           a5,-2147483648
0xbfc80274:  00a7a823          sw            a0,16(a5)
0xbfc80278:  00008067          ret           
0xbfc8027c:  0007a683          lw            a3,0(a5)
0xbfc80280:  00100713          addi          a4,zero,1
0xbfc80284:  00e69863          bne           a3,a4,16                        # 0xbfc80294

The SW at 0xBFC80274 is just not output.
The core connection is like a puzzle: the dbus_ready and rsp_valid makes me frustrated. So this is my connection:
	wire	[31:0]	w_instr_addr;
	wire	[31:0]	w_instr_data;
	wire			w_instr_ready;
	
	wire	[31:0]	w_data_addr;
	wire	[31:0]	w_data_rdata;
	wire			w_data_rd;
	wire	[31:0]	w_data_wdata;
	reg	[3:0]		w_data_wr;
	wire			w_data_ready;

	wire	[31:0]	w_data_addr_pre;
	wire			w_data_wr_pre;
	wire	[1:0]		w_rv32_cpu_wrsize;		//	0:byte, 1:word, 2:dword
	
	wire			w_ibus_cmd_valid, w_dbus_cmd_valid;
	
	generate
		if(SOC_SIMULATION == "FALSE") begin: Gen_SOC
			VexRiscvSmallestNoCsrSynp rv32_cpu (
				.iBus_cmd_valid			(w_ibus_cmd_valid),
				.iBus_cmd_ready			(w_instr_ready),
				.iBus_cmd_payload_pc		(w_instr_addr),
				.iBus_rsp_valid			(0),
				.iBus_rsp_payload_error		(0),
				.iBus_rsp_payload_inst		(w_instr_data),
				.dBus_cmd_valid			(w_dbus_cmd_valid),
				.dBus_cmd_ready			(w_data_ready || (|w_data_wr) || (~w_dbus_cmd_valid)),
				.dBus_cmd_payload_wr		(w_data_wr_pre),
				.dBus_cmd_payload_address	(w_data_addr_pre),
				.dBus_cmd_payload_data		(w_data_wdata),
				.dBus_cmd_payload_size		(w_rv32_cpu_wrsize),
				.dBus_rsp_ready			(0),
				.dBus_rsp_error			(0),
				.dBus_rsp_data			(w_data_rdata),
				.clk					(clk_i),
				.reset				(r_soc_rst[0])
			);
			assign w_data_addr = {w_data_addr_pre[31:2], 2'b0};
		end else begin: Gen_Sim
			
		end
	endgenerate
	always @(*) begin
		if(w_rv32_cpu_wrsize == 2'b10)
			w_data_wr <= {4{w_data_wr_pre}};
		else if(w_rv32_cpu_wrsize == 2'b01) begin
			w_data_wr[1:0] <= (w_data_addr_pre[1] == 0) ? {2{w_data_wr_pre}} : 2'b00;
			w_data_wr[3:2] <= (w_data_addr_pre[1] == 1) ? {2{w_data_wr_pre}} : 2'b00;
		end else begin
			w_data_wr[0] <= (w_data_addr_pre[1:0] == 2'b00) && w_data_wr_pre;
			w_data_wr[1] <= (w_data_addr_pre[1:0] == 2'b01) && w_data_wr_pre;
			w_data_wr[2] <= (w_data_addr_pre[1:0] == 2'b10) && w_data_wr_pre;
			w_data_wr[3] <= (w_data_addr_pre[1:0] == 2'b11) && w_data_wr_pre;
		end
	end
	
I just got stuck here. Thanks for any suggestions.

tomverbeure, 2019-01-23:
The problem is that iBus_rsp_valid and dBus_rsp_ready are tied to zero.

Not that dBus_rsp_ready and iBus_rsp_valid have the same meaning. They are high when requested read data is ready to be accepted by the CPU. (I think that difference in naming is inconsistent and unfortunate.)

You need to understand the way a bus like this works: there are 2 phases. Accepting a request and returning the data for that request.

Both need to be acknowledged once per request by the outside world through req_ready and rsp_valid rsp.

The reason it is done this way is so that you can have multiple requests for data in flight at the same time.

Say it takes 5 clock cycles to fetch an instruction from an external cache when you have a cache hit, but a huge amount of time for a cache miss.

In that case, you need to have both req_ready and rsp_valid to achieve maximum performance.

req_ready indicates that your cache has consumed a request, which offers the CPU the opportunity to immediately issue a request for the next instruction (if its pipeline allows it to do so). 
Meanwhile rsp_ready allows the cache to return the data of this request when the data is available. In case of a hit, it will be after 5 cycles, in case of a miss it will be after much longer.

There exist busses that don’t have this split request ready and response ready (such as the picorv32 RISC-V or the Arm APB3 bus), but those are inherently much slower when memory reads take multiple cycles.

Tom

Tom

wye1102, 2019-01-24:
I tried another 4 hours but still cannot get it work properly. I used this soc wrapper on an old MIPS core project which works perfectly (fully optimized pipeline but pretty big). When ICACHE is disabled, the processor won't stall (PC goes invalid) however it seems something is wrong and enters infinite loop. When ICACHE is enabled, the processor just hang after sometime at simulation. The register file is fully compatible with original implementation.

Looks like req / ready / valid signals is going to make it very complicated for migration. I would rather use iready and dready (as used in the current MIPS core) which is very simple and easy to implement and there's no ambiguity. 

Here's the demo ISE project which may be used for analysis. SpinalHDL is a good tool to help reduce complicated wiring. I'll try to use spinalhdl to write some modules later for some test.

[rv32_soc.tar.gz](https://github.com/SpinalHDL/VexRiscv/files/2790152/rv32_soc.tar.gz)