SpinalHDL/VexRiscv #57: Two lint issues with the vexriscv.demo.GenFull configuration

secworks, 2019-02-22:
Aloha!

I generated Verilog for the vexriscv.demo.GenFull configuration. I then used the verilator linter to check the code. There were two warnings:

%Warning-WIDTH: VexRiscv.v:2565: Operator ASSIGNW expects 30 bits on the Assign RHS, but Assign RHS's SHIFTR generates 32 bits.
%Warning-WIDTH: VexRiscv.v:4422: Operator ASSIGNW expects 30 bits on the Assign RHS, but Assign RHS's SHIFTR generates 32 bits.

Line 2656 is:
  assign _zz_331_ = (IBusCachedPlugin_iBusRsp_stages_0_input_payload >>> 2);

And line 4422 is:
  assign _zz_164_ = (IBusCachedPlugin_fetchPc_output_payload >>> 2);

So both seems to be related to the instruction cache bus. 

Is this a known issue? 

Looking at the code it seems

Dolu1990, 2019-02-22:
Hi,
The Verilog generator of SpinalHDL always assign the result of a shift operator into a variable to resize the result to the wanted size. (There is the list of all expression for which it do that : https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/internals/ComponentEmitterVerilog.scala#L1092)

So it isn't an issue, but a generator coding style ^^
Is that an issue in some cases ? Is that linting config standard ?
(I'm not used to linting tools)

secworks, 2019-02-23:
Ok, if I look at the Verilog code I see:

On line 4422:  assign _zz_164_ = (IBusCachedPlugin_fetchPc_output_payload >>> 2);
_zz_164_ is defined on row 2036 to "wire [29:0] _zz_164_;".  So the assignment looses two bits.
But then on row 2064 the wire is used: "assign _zz_330_ = _zz_164_[9:0];" So we are actually only using the lowest 10 bits. No problem.

It is possible to disable these types of warnings, or build a list of known false errors to be ignored. In general I find that these warnings where the LHS and RHS widths of an assignment don't match usually points to some real issue. And esp if LHS width < RHS width you do loose info. So disabling lint checking assignment width mismatch should not be done. Generally.

I'm closing this issue.

BTW: We are looking at using VexRisc in the Cryptech Open HSM project. I'll be trying to use it at first as a SW controlled DMA/local master. So first core will be smallish, but include the debug port.
https://cryptech.is/

closed 2019-02-23 by secworks