SpinalHDL/VexRiscv #59: Incorrect path to examples

secworks, 2019-02-26:
Aloha!

In README.md, under section CPU Generation (https://github.com/SpinalHDL/VexRiscv#cpu-generation) it states that:
>You can find two example CPU instances in:
>
>    src/main/scala/vexriscv/GenFull.scala
>    src/main/scala/vexriscv/GenSmallest.scala

I guess they have moved. The correct path seems to be "src/main/scala/vexriscv/demo/"
And it contains many more examples than the two mentioned.

secworks, 2019-02-26:
At least for me most of the examples in the demo directory fails to build. If I try with sbt "runMain vexriscv.demo.GenSmallAndPerformantICache" I get:

[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running (fork) vexriscv.demo.GenSmallAndPerformantICache
[error] Error: Could not find or load main class vexriscv.demo.GenSmallAndPerformantICache
[error] Caused by: java.lang.ClassNotFoundException: vexriscv.demo.GenSmallAndPerformantICache
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 1 s, completed 26 Feb 2019, 12:57:29

Dolu1990, 2019-02-26:
Hi,
I just fixed the readme path to GenFull and GenSmallest in the dev branch

About sbt "runMain vexriscv.demo.GenSmallAndPerformantICache", it should be  sbt "runMain vexriscv.demo.GenSmallAndProductiveICache"
That's my fault, the object into the scala file isn't named the same way than the file itself XD a fix is also submited to match the file name with the object name.

Thanks for the feed back ^^

Dolu1990, 2019-02-26:
I will merge this in master soon.

closed 2019-02-26 by Dolu1990

secworks, 2020-02-10:
Cool! Thanks!