SpinalHDL/VexRiscv #63: Provide MMU / MemoryTranslatorPlugin example

xobs, 2019-03-16:
VexRiscv implements a simple MMU.  This MMU is effective, but nonstandard, and requires extended instructions to do refilling.

Would it be possible to give a simple example of how to use it?  For example, how would mappings be added and removed in order to create a simple context switch for a protected memory operating system?

Dolu1990, 2019-03-17:
The instruction are :
    def TLBW0  = M"0000000----------111-----0001111"
    def TLBW1  = M"0000001----------111-----0001111"

For instance, those 5 line : 
https://github.com/Dolu1990/pinsecSoftware/blob/VexRiscv/tests/mmu/src/crt.S#L121
Will map the virtual address 0xC2000000 to 0x20000000  in execute read write mode without user rights via the second shared TLB entry

The mapping of the instructions into the shared TLB is done there : https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/MemoryTranslatorPlugin.scala#L137

The number of shared TLB entry is defined by the plugin arguments.

Let's me know if something isn't clear :)

Also, i'm currently implementing a hardware refilled MMU : 
https://github.com/SpinalHDL/VexRiscv/issues/60#issuecomment-473652087

Dolu1990, 2019-03-23:
In the Linux branch, there is a Linux config in the demo folder, cacheless iBus dBus with self refilled MMU. It pass all my tests, the debug interface is broken at the moment.
Maybe it would be better than using the self refilled MMU for your use case ?
In that branch, i also fixed, and bring back many fixes from daveshah1 and kgugala.
In medium term, it will be merged back into master.

xobs, 2019-03-24:
Self-refilled should be fine, whichever has the lower gate count is preferable.

Dolu1990, 2019-03-24:
The overhead should be high, about 100-150 lut max for the same sizeof associative  iTLB dTLB cache, but it can even turn into a area gain, because you can probably reduce the size of the fully associative  iTLB dTLB cache (self refilling is fast :D)

Dolu1990, 2019-05-25:
@xobs Hi,
How things are going ?
Did you tried the self refilled MMU ?