SpinalHDL/VexRiscv #65: Murax debug on FPGA

sebastien-riou, 2019-03-18:
I had some success using Debug with verilator, now I switched to FPGA (iCE40-HX8K) and a  JLink JTAG probe. I am using the project [iCE40-hx8k_breakout_board](https://github.com/SpinalHDL/VexRiscv/tree/master/scripts/Murax/iCE40-hx8k_breakout_board).
**What works:**
- The code execution in general, without connection to debug I mean.
- The JTAG connection seems fine, the target is found everytime, I can read/write registers and run from the first instruction to the first breakpoint (which I can place anywhere in the code).

**What does not work:**
- I cannot step (just trying intruction level for now). when I try to instruction step from the first instruction, instead of jumping to 0x800000b0 I land at 0xb0. If I set a break point at 0x800000b0 and just run then I stop there. Then if I instruction step I land at 0xb4 instead of 0x800000b4... It seems the instruction step is discarding some most significant bits. 
- I cannot run to a second breakpoint

**What I tried:** 
- removing all break points -> the first instruction step still lands at 0xb0
- reduce JTAG frequency to 400kHz -> same behavior
- openocd debug message does not contain obvious error message

openocd_riscv is up-to-date, Murax is from yesterday, I just changed the RAM size to 8k (it is 4k in the object "MuraxWithRamInit").

log of openocd for the trial "reset, run with break at , instruction step"

````
user@lafite:~/Downloads/openocd_riscv$ src/openocd -d -f tcl/interface/jlink.cfg -c "set MURAX_CPU0_YAML /home/user/Downloads/VexRiscv/cpu0.yaml" -f tcl/target/murax.cfg
Open On-Chip Debugger 0.10.0+dev-01202-gced8dcd (2019-02-25-22:00)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 5 command.c:544 command_print(): debug_level: 3
Debug: 14 5 options.c:184 add_default_dirs(): bindir=/usr/local/bin
Debug: 15 5 options.c:185 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 16 5 options.c:186 add_default_dirs(): exepath=/home/user/Downloads/openocd_riscv/src
Debug: 17 5 options.c:187 add_default_dirs(): bin2data=../share/openocd
Debug: 18 5 configuration.c:42 add_script_search_dir(): adding /home/user/.openocd
Debug: 19 5 configuration.c:42 add_script_search_dir(): adding /home/user/Downloads/openocd_riscv/src/../share/openocd/site
Debug: 20 5 configuration.c:42 add_script_search_dir(): adding /home/user/Downloads/openocd_riscv/src/../share/openocd/scripts
Debug: 21 5 configuration.c:97 find_file(): found tcl/interface/jlink.cfg
Debug: 22 5 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface jlink
Debug: 23 5 command.c:143 script_debug(): command - interface ocd_interface jlink
Debug: 25 5 command.c:323 command_new(): BUG: command 'jlink' does not have the '.usage' field filled out
Debug: 26 5 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 27 5 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 28 5 command.c:323 command_new(): BUG: command 'jlink freemem' does not have the '.usage' field filled out
Debug: 29 5 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 30 5 command.c:323 command_new(): BUG: command 'jlink hwstatus' does not have the '.usage' field filled out
Debug: 31 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 32 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 33 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 34 6 command.c:323 command_new(): BUG: command 'jlink config' does not have the '.usage' field filled out
Debug: 35 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 36 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 37 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 38 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 39 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 40 6 command.c:323 command_new(): BUG: command 'config reset' does not have the '.usage' field filled out
Debug: 41 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 42 6 command.c:323 command_new(): BUG: command 'config write' does not have the '.usage' field filled out
Debug: 43 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 44 6 command.c:323 command_new(): BUG: command 'jlink emucom' does not have the '.usage' field filled out
Debug: 45 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 46 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
Debug: 47 6 command.c:364 register_command_handler(): registering 'ocd_jlink'...
User : 48 6 command.c:693 command_run_line(): /home/user/Downloads/VexRiscv/cpu0.yamlUser : 49 6 command.c:695 command_run_line(): 
Debug: 50 6 configuration.c:97 find_file(): found tcl/target/murax.cfg
Debug: 51 6 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_adapter_khz 400
Debug: 52 6 command.c:143 script_debug(): command - adapter_khz ocd_adapter_khz 400
Debug: 54 6 core.c:1645 jtag_config_khz(): handle jtag khz
Debug: 55 6 core.c:1612 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 56 6 core.c:1612 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 57 6 command.c:544 command_print(): adapter speed: 400 kHz
Debug: 58 6 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_adapter_nsrst_delay 260
Debug: 59 6 command.c:143 script_debug(): command - adapter_nsrst_delay ocd_adapter_nsrst_delay 260
User : 61 6 command.c:544 command_print(): adapter_nsrst_delay: 260
Debug: 62 6 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
[... 432 lines cut ...]
Debug: 2847 19514 vexriscv.c:1479 vexriscv_resume_or_step(): Target resumed at 0x800000b0
Debug: 2857 19514 vexriscv.c:710 vexriscv_debug_entry(): -
Debug: 2867 19515 vexriscv.c:636 vexriscv_save_context(): -
Debug: 2886 19539 target.c:1581 target_call_event_callbacks(): target event 0 (gdb-halt)
Debug: 2887 19539 target.c:1581 target_call_event_callbacks(): target event 1 (halted)
Debug: 2888 19539 target.c:1581 target_call_event_callbacks(): target event 6 (gdb-end)
Debug: 2898 19540 gdb_server.c:3160 gdb_input_inner(): received packet: 'p20'
Debug: 2899 19540 vexriscv.c:1249 vexriscv_get_gdb_reg_list(): vexriscv_get_gdb_reg_list 0

Debug: 2909 19541 gdb_server.c:3160 gdb_input_inner(): received packet: 'g'
Debug: 2910 19541 vexriscv.c:1249 vexriscv_get_gdb_reg_list(): vexriscv_get_gdb_reg_list 1

Debug: 2911 19541 gdb_server.c:3160 gdb_input_inner(): received packet: 'qXfer:threads:read::0,fff'
Debug: 2912 19541 gdb_server.c:3160 gdb_input_inner(): received packet: 'qXfer:threads:read::0,fff'
Debug: 2913 19593 gdb_server.c:3160 gdb_input_inner(): received packet: 'm80,40'
Debug: 2914 19593 gdb_server.c:1444 gdb_read_memory_packet(): addr: 0x0000000000000080, len: 0x00000040
Debug: 2915 19593 target.c:2200 target_read_buffer(): reading buffer of 64 byte at 0x00000080
Debug: 2925 19610 gdb_server.c:3160 gdb_input_inner(): received packet: 'mc0,40'
Debug: 2926 19610 gdb_server.c:1444 gdb_read_memory_packet(): addr: 0x00000000000000c0, len: 0x00000040
Debug: 2927 19610 target.c:2200 target_read_buffer(): reading buffer of 64 byte at 0x000000c0
Debug: 2937 19626 gdb_server.c:3160 gdb_input_inner(): received packet: 'm100,40'
Debug: 2938 19626 gdb_server.c:1444 gdb_read_memory_packet(): addr: 0x0000000000000100, len: 0x00000040
Debug: 2939 19626 target.c:2200 target_read_buffer(): reading buffer of 64 byte at 0x00000100

````

Dolu1990, 2019-03-21:
Hi,

Hmm that's is weird.
I never tried with that jtag key. Do you have another one ?
Can you also try to connect to the simulated Murax via the SpinalSim ? (sbt "test:runMain vexriscv.MuraxSim" + jtagtcp interface for openocd)

Those two things can help to diagnostic :)

sebastien-riou, 2019-03-23:
Hi,

Thanks for the suggestion, I was not aware of SpinalSim.

SpinalSim works well.
the first relevant difference in the logs (as far as I can see) is this one:
SpinalSim:
Debug: 404 9045 gdb_server.c:3160 gdb_input_inner(): received packet: 'm80000080,40'
JLINK Hardware:
Debug: 2913 19593 gdb_server.c:3160 gdb_input_inner(): received packet: 'm80,40'

I ordered another jtag hardware

Dolu1990, 2019-03-23:
Good :)
So, just to say, i'm using a FTDI2232H based breakout, i already tried with some others, as JtagKey, JtagKey2.
Some people also used raspberry pi by the past.

Let's me know how it go

sebastien-riou, 2019-03-31:
I confirm nothing is wrong with Murax, I tried the "FT2232H_Mini_Module" with the exact same FPGA image and everything else, it works like a charm. The problem seems on JLink side or just a weird electrical connection problem, sorry about that.

closed 2019-03-31 by sebastien-riou

Dolu1990, 2019-05-07:
One thing. Today, i had to make one board working with the c232hm_ddhsl_0. And with that specific dongle, i had some issues. It was because of electrical reasons, bad impedance matching / crosstalk.
So what i finaly did, is adding a 380 ohm resistance between the FPGA pin and the dongle TCK pin (close to the FPGA) to filter that out. then it worked fine :)