SpinalHDL/VexRiscv #68: Add a pipeline description to gcc

futaris, 2019-03-27:
https://gcc.gnu.org/onlinedocs/gccint/Processor-pipeline-description.html#Processor-pipeline-description

Add something like:
https://github.com/gcc-mirror/gcc/commit/8b1090c1eaf28580aeac2de46401eaf4fc91bb5a

for VexRiscV

Dolu1990, 2019-03-27:
Hoo i didn't knew about it.
It would be interresting :D

Thanks for the links.

mithro, 2019-03-27:
Maybe VexRISCV should auto generate this? The pipeline can be very different with different configs?

Dolu1990, 2019-03-27:
Yes right it could be autogenerated, as it can very much,
- Branch predication can be turned off
- Miss prediction penality can vary much
- MUL can be iterative or single cycle with late result
- Shifts register can be iterative isntead of single cycle
- ...

mithro, 2019-03-27:
@futaris Do you know if this configuration is compiled into gcc or can it be provided at runtime?

tomverbeure, 2019-03-27:
I was coincidentally looking at this yesterday as well.

When you read the paper where the automata was introduced (in 2003), the performance improvement can be significant in some cases (~10%) but that's for super-scalar architectures with asymmetric pipelines. 

Even with all the configuration options, the VexRiscv still has a very conventional pipeline. I wonder what kind of upside there might be to have a custom version for each major configuration?

It might be sufficient to just have a version with the 3-stage deep MUL and other high clock speed settings. If you're going to use the iterative shift or multiply operations, performance was clearly not a priority to begin with, and the potential upside for a custom automata will probably very low. (<1%?)

Tom

futaris, 2019-03-27:
@mithro I think it's compile time only. Best to look at what microblaze and openrisc do. 

Best to look at what microblaze and openrisc do in gcc.

https://github.com/Xilinx/meta-xilinx/tree/master/meta-xilinx-bsp/recipes-microblaze/gcc/gcc-7