SpinalHDL/VexRiscv #70: No trap generated for `divu` on core without divide

xobs, 2019-04-23:
On a VexRiscv core without a Multiply or Divide stage, the core will execute a `divu` instruction.  It does... _something_.  I'm not sure what.

The VexRiscv source file is at https://github.com/xobs/VexRiscv-verilog/blob/riscv-contest-20190315/src/main/scala/vexriscv/GenCoreDefault.scala.  This particular core was instantiated with the following command:

    run-main vexriscv.GenCoreDefault --iCacheSize 1024 --dCacheSize 0 --mulDiv false --singleCycleMulDiv false --outputFile 2-stage-1024-cache --pipelining false --memoryStage false --writeBackStage false

As you can see, `mulDiv` is `false`, which causes no divider to be added.

Here is the result of running that instruction on an FPGA:

```
(gdb) disassemble $pc-4,$pc+8
Dump of assembler code from 0x20040530 to 0x2004053c:
   0x20040530 <gc_init+8>:      li      a2,65
=> 0x20040534 <gc_init+12>:     divu    a4,a4,a2
   0x20040538 <gc_init+16>:     addi    sp,sp,-16
End of assembler dump.
(gdb) p/x $a2
$5 = 0x41
(gdb) p/x $a4
$6 = 0xfffc
(gdb) stepi
0x20040538      122         MP_STATE_MEM(gc_alloc_table_byte_len) = total_byte_len / (1 + BITS_PER_BYTE / 2 * BYTES_PER_BLOCK);(gdb) disassemble $pc-4,$pc+8
Dump of assembler code from 0x20040534 to 0x20040540:
   0x20040534 <gc_init+12>:     divu    a4,a4,a2
=> 0x20040538 <gc_init+16>:     addi    sp,sp,-16
   0x2004053c <gc_init+20>:     sw      s0,8(sp)
End of assembler dump.
(gdb) p/x $a2
$7 = 0x41
(gdb) p/x $a4
$8 = 0x7ffe
(gdb)
```

As you can see, it tries to execute $a4 = $a4 / $a2, but rather than raising an exception it's done something strange and gotten 0x7ffe as the result.

Dolu1990, 2019-04-23:
https://github.com/xobs/VexRiscv-verilog/blob/riscv-contest-20190315/src/main/scala/vexriscv/GenCoreDefault.scala#L142 :) ?

xobs, 2019-04-23:
...I retract the statement.  Sorry for the noise!

closed 2019-04-23 by xobs

Dolu1990, 2019-04-23:
No worries :)

Anyway, i will very soon merge the VexRiscv linux branch into master, it bringed a lot of fixes and compatibility improvements and better regressions test (including running linux on cache-less designs)

https://travis-ci.org/SpinalHDL/VexRiscv/builds/523267998

Just need to release the related SpinalHDL version, and update the readme

xobs, 2019-04-23:
Right now I'm mostly size-constrained.  If I enable illegal instruction traps, it actually doesn't fit.

On a UP5K I have:

- 2-stage VexRiscV with 1024 bytes cache, 4 hardware breakpoints, and SIMPLE branch prediction
- Memory-mapped Quad SPI Flash
- Wishbone bus
- Basic Timer
- USB with five endpoints
- Lattice LED Hard IP (memory-mapped)
- 8 kB ROM
- Debugging via USB packets

The last part is the most exciting. I can debug a VexRiscV entirely without external tools over USB, while it's in the USB port.  This includes debugging the USB stack, as long as it enumerates properly (i.e. the basic descriptors have to be transferred.)

Eventually I'll shrink the USB core, which will let me fit a more complicated VexRiscV.  And hopefully Yosys will be able to infer the DSP blocks to do multiply.

But smaller is always better.

Dolu1990, 2019-04-23:
Hoo nice :D
The cache is for the instruction ? The IBusCachedPlugin ?

Dolu1990, 2019-04-23:
How much LC at the moment :D ?
Which frequancy ?

xobs, 2019-04-23:
Yes, IBusCachedPlugin as 1024 bytes.  DBusCachedPlugin has 0 bytes.

```
Info: Device utilisation:
Info:            ICESTORM_LC:  5266/ 5280    99%
Info:           ICESTORM_RAM:    29/   30    96%
Info:                  SB_IO:    14/   96    14%
Info:                  SB_GB:     8/    8   100%
Info:           ICESTORM_PLL:     1/    1   100%
Info:            SB_WARMBOOT:     1/    1   100%
Info:           ICESTORM_DSP:     0/    8     0%
Info:         ICESTORM_HFOSC:     0/    1     0%
Info:         ICESTORM_LFOSC:     0/    1     0%
Info:                 SB_I2C:     0/    2     0%
Info:                 SB_SPI:     0/    2     0%
Info:                 IO_I3C:     0/    2     0%
Info:            SB_LEDDA_IP:     1/    1   100%
Info:            SB_RGBA_DRV:     1/    1   100%
Info:         ICESTORM_SPRAM:     4/    4   100%
...
Info: Max frequency for clock     'clk48': 200.80 MHz (PASS at 48.00 MHz)
Info: Max frequency for clock 'crg_clk12': 12.58 MHz (PASS at 12.00 MHz)
Info: Max frequency for clock 'crg_clk48': 49.27 MHz (PASS at 48.00 MHz)
```

I think I mentioned that the VexRiscV has a very strange critical path that looks like a 32-bit adder.  That's another thing I need to look into...

Dolu1990, 2019-04-23:
wtf 99 % full, and still getting 12 Mhz
Nice :D

About weird 32 bits path, I do not remember about it, can you send and critical path report ?

Dolu1990, 2019-04-23:
And the related netlist :D

xobs, 2019-04-23:
I'm pleasantly surprised that it's even able to route, let alone meet timing!

Here's the full output:

```
Info: Importing module top
Info: Rule checker, verifying imported design
Info: Checksum: 0x262d3f8b

Info: constrained 'clk48' to bel 'X6/Y0/io1'
Info: constrained 'spiflash_cs_n' to bel 'X24/Y0/io1'
Info: constrained 'spiflash_clk' to bel 'X24/Y0/io0'
Info: constrained 'spiflash_miso' to bel 'X23/Y0/io1'
Info: constrained 'spiflash_mosi' to bel 'X23/Y0/io0'
Info: constrained 'spiflash_wp' to bel 'X22/Y0/io1'
Info: constrained 'spiflash_hold' to bel 'X21/Y0/io1'
Info: constrained 'led_rgb0' to bel 'X4/Y31/io0'
Info: constrained 'led_rgb1' to bel 'X5/Y31/io0'
Info: constrained 'led_rgb2' to bel 'X6/Y31/io0'
Info: constrained 'usb_d_p' to bel 'X13/Y31/io1'
Info: constrained 'usb_d_n' to bel 'X13/Y31/io0'
Info: constrained 'usb_pullup' to bel 'X12/Y31/io1'
Info: constrained 'touch_t1' to bel 'X7/Y0/io0'
Info: constrained 'touch_t2' to bel 'X6/Y0/io0'
Info: constrained 'touch_t3' to bel 'X5/Y0/io0'
Info: constrained 'touch_t4' to bel 'X7/Y0/io1'
Warning: net 'usb_48_clk' does not exist in design, ignoring clock constraint
Warning: net 'usb_48_raw_clk' does not exist in design, ignoring clock constraint
Warning: net 'sys_clk' does not exist in design, ignoring clock constraint
Warning: net 'usb_12_clk' does not exist in design, ignoring clock constraint
Info: constraining clock net 'crg_clk48' to 48.00 MHz
Info: constraining clock net 'clk48' to 48.00 MHz
Warning: net 'crg_clk12_raw' does not exist in design, ignoring clock constraint

Info: Packing constants..
Info: Packing IOs..
Info: spiflash_cs_n feeds SB_IO SB_IO_4, removing $nextpnr_iobuf spiflash_cs_n.
Info: spiflash_clk feeds SB_IO SB_IO_5, removing $nextpnr_iobuf spiflash_clk.
Info: spiflash_miso feeds SB_IO SB_IO_3, removing $nextpnr_iobuf spiflash_miso.
Info: spiflash_mosi feeds SB_IO SB_IO_2, removing $nextpnr_iobuf spiflash_mosi.
Info: spiflash_wp feeds SB_IO SB_IO_6, removing $nextpnr_iobuf spiflash_wp.
Info: spiflash_hold feeds SB_IO SB_IO_7, removing $nextpnr_iobuf spiflash_hold.
Info: led_rgb0 use by SB_RGBA_DRV SB_RGBA_DRV, not creating SB_IO
Info: led_rgb1 use by SB_RGBA_DRV SB_RGBA_DRV, not creating SB_IO
Info: led_rgb2 use by SB_RGBA_DRV SB_RGBA_DRV, not creating SB_IO
Info: usb_d_p feeds SB_IO SB_IO, removing $nextpnr_iobuf usb_d_p.
Info: usb_d_n feeds SB_IO SB_IO_1, removing $nextpnr_iobuf usb_d_n.
Info: touch_t1 feeds SB_IO SB_IO_8, removing $nextpnr_iobuf touch_t1.
Info: touch_t2 feeds SB_IO SB_IO_9, removing $nextpnr_iobuf touch_t2.
Info: touch_t3 feeds SB_IO SB_IO_10, removing $nextpnr_iobuf touch_t3.
Info: touch_t4 feeds SB_IO SB_IO_11, removing $nextpnr_iobuf touch_t4.
Info: Packing LUT-FFs..
Info: Packing non-LUT FFs..
Info: Packing carries..
Info: Packing RAMs..
Info: Placing PLLs..
Info:   constrained PLL 'SB_PLL40_CORE' to X12/Y31/pll_3
Info: Packing special functions..
Info:   constrained SB_LEDDA_IP 'SB_LEDDA_IP' to X0/Y31/ledda_ip_2
Info:   constrained SB_RGBA_DRV 'SB_RGBA_DRV' to X0/Y30/rgba_drv_0
Info: Promoting globals..
Info: promoting picorvspi_reset1 [reset] (fanout 763)
Info: promoting VexRiscv.reset [reset] (fanout 53)
Info: promoting $abc$53614$auto$rtlil.cc:1817:NotGate$53502 [reset] (fanout 32)
Info: promoting $abc$53614$auto$rtlil.cc:1817:NotGate$53490 [reset] (fanout 18)
Info: promoting VexRiscv.IBusCachedPlugin_iBusRsp_cacheRspArbitration_input_ready [cen] (fanout 85)
Info: promoting $abc$53614$auto$dff2dffe.cc:158:make_patterns_logic$40950 [cen] (fanout 81)
Info: Constraining chains...
Info: Checksum: 0x4527e47a

[... 620 lines cut ...]
Info: Slack histogram:
Info:  legend: * represents 23 endpoint(s)
Info:          + represents [1,23) endpoint(s)
Info: [    -4,   4006) |*+
Info: [  4006,   8016) |*+
Info: [  8016,  12026) |**+
Info: [ 12026,  16036) |***+
Info: [ 16036,  20046) |********+
Info: [ 20046,  24056) |*+
Info: [ 24056,  28066) |**+
Info: [ 28066,  32076) |***+
Info: [ 32076,  36086) |********+
Info: [ 36086,  40096) |***+
Info: [ 40096,  44106) |*****+
Info: [ 44106,  48116) |*****+
Info: [ 48116,  52126) |**********+
Info: [ 52126,  56136) |****************+
Info: [ 56136,  60146) |***************************+
Info: [ 60146,  64156) |*****************************************************+
Info: [ 64156,  68166) |*************************+
Info: [ 68166,  72176) |***********************+
Info: [ 72176,  76186) |******************************+
Info: [ 76186,  80196) |************************************************************
5 warnings, 0 errors
```

Dolu1990, 2019-04-23:
Right, there is 32 bits adder -> some stuff, 32 bits adder, and it was a mistake from my side, some combinatorial link leaking where it shoudln't have (Branch plugin target calculation was depending combinatorily from the success of the branch for STATIC and DYNAMIC prediction), you are using one of those right ?

It was fixed in the linux branch :D
https://github.com/SpinalHDL/VexRiscv/commit/63cd5f42aff00661f68c1139228ac1800af4ee14 + https://github.com/SpinalHDL/VexRiscv/commit/41ff87f83b4d82b22723fef4b17f8f826823214f

And then it also looked like one Quartus, the synthesis was reducing the area by doing some tricks which where greatly reducing the fmax by 40 %, fixed in : 
https://github.com/SpinalHDL/VexRiscv/commit/13b774b535626ea9ab469ad6029ca3f5cbb3e984

xobs, 2019-04-23:
I use STATIC.  Oddly, when I selected NONE, it didn't route at all.

Let me try those newer branches to see if it improves timing at all.

Dolu1990, 2019-04-23:
Maybe you will need a script which try new seeds untill it success ^^ ?