SpinalHDL/VexRiscv #73: Issues with built-in DDR memory controller Briey SoC

FlyingxDutch, 2019-05-03:
Hello,

for some time I am trying to run "Briey SoC" (generated from SpinalHDL according to given instruction) on Spartan6 based FPGA board "Mimas V.2" from "Numat Lab". Here is link to this board description (on Numato website):
[https://numato.com/product/mimas-v2-spartan-6-fpga-development-board-with-ddr-sdram](url)

In order to launch Briey.v project on real FPGA board (Mimas V.2) I altered the original design. The major changes are:
1) Added (PLL) instance CLKGEN (Xilinx IP core) for generating clocks: AXI bus clock, VGA clock, DDR memory clock 

2) Added input buffer (IBUFG) for input JTAG clock (signal: io_jtag_tck)

3) Generation of differential clock (two lines) for DRR memory from single clock (100 MHz PLL). Added primitive OBUFDS.

4) Two directional buffer (instance DDRDataBuffer - "SelectI/O Wizard IP core") added to Briey bus and DDR memory data bus (16 bit wide)

5) In "User constraints file" (BrieySOC.ucf) maped properly most of ports of top module (BrieySOC). There is lack of few important signals - I will write about this later.

The main issue I had encountered is related to internal Briey DDR memory controller.
My FPGA board Mimas v.2 has on-board SDRAM (DDR1 RAM memory) chip: MT46H32M16LF.
This chip requires diferential clock (two lines) and also two-directional 16-bit wide data bus.

In the Briey design thera are two one-directional 16-bit buses (one for write and second for read). To solve that issue I added Xilinx IPcore named "Select I/O Wizard"

I reconstructed the original design Briey.v with new top module named "BrieySOC", here is it's code (Verilog):
`/* Top module: Briey SOC */

module BrieySOC(
      input   io_asyncReset,
      input   Clk100Mhz,
      input   io_jtag_tms,
      input   io_jtag_tdi,
      output  io_jtag_tdo,
      input   io_jtag_tck,
      output [12:0] io_sdram_ADDR,
      output [1:0] io_sdram_BA,
      //input  [15:0] io_sdram_DQ_read,
      //output [15:0] io_sdram_DQ_write,
		inout  [15:0] io_sdram_DATA,
      output  io_sdram_DQ_writeEnable,
      output [1:0] io_sdram_DQM,
      output  io_sdram_CASn,
      output  io_sdram_CKE,
      output  io_sdram_CSn,
      output  io_sdram_RASn, 
      output  io_sdram_WEn,
		output  io_sdram_clk,
		output  io_sdram_clk_n,
      input  [3:0] io_gpioA_read,
      output [3:0] io_gpioA_write,
      output [3:0] io_gpioA_writeEnable,
      output  io_uart_txd,
      input   io_uart_rxd,
      output  io_vga_vSync,
      output  io_vga_hSync,
      output  io_vga_colorEn,
      output [4:0] io_vga_color_r,
      output [5:0] io_vga_color_g,
      output [4:0] io_vga_color_b,
      input   io_timerExternal_clear,
      input   io_timerExternal_tick,
      input   io_coreInterrupt
    );
//----------------------------------------------------

//-- Connecting wires
wire clkAXIOut;
wire clkLOCKED;
wire clkVGAOut;
wire clkBuf100MHz;
wire clkDDR100MHz;
wire clkJTAGBuf;
[... 193 lines cut ...]
NET "io_vga_color_g[5]"          LOC = C11     | IOSTANDARD = LVCMOS33 | DRIVE = 8 | SLEW = FAST ;
NET "io_vga_color_g[4]"          LOC = A10     | IOSTANDARD = LVCMOS33 | DRIVE = 8 | SLEW = FAST ;
NET "io_vga_color_g[3]"          LOC = C10     | IOSTANDARD = LVCMOS33 | DRIVE = 8 | SLEW = FAST ;

NET "io_vga_color_b[4]"          LOC = A11     | IOSTANDARD = LVCMOS33 | DRIVE = 8 | SLEW = FAST ;
NET "io_vga_color_b[3]"          LOC = B11     | IOSTANDARD = LVCMOS33 | DRIVE = 8 | SLEW = FAST ;
  `
But I still don't known how to map few important DDR memory signals. The problem is that DDR chip on Mimas V2 has different internal arrangment. I don't know how to map/generate in Verilog code these signals for MT46H32M16LF memory chip:
`           #NET  "mcb3_dram_dm"              LOC = K3      | IOSTANDARD = MOBILE_DDR;
           #NET  "mcb3_dram_dqs"             LOC = L4      | IOSTANDARD = MOBILE_DDR;
            #NET  "mcb3_dram_udm"             LOC = K4      | IOSTANDARD = MOBILE_DDR;
            #NET  "mcb3_dram_udqs"            LOC = P2      | IOSTANDARD = MOBILE_DDR;
            #NET  "mcb3_rzq"                  LOC = N4      | IOSTANDARD = MOBILE_DDR;

   But I don't know how to generate these signals from original signals of Briey DDR controler. I am aware that these signals are important for proper timing of DDR memory.

 The last issue is that I am not sure if added two-directional buffer for busses between original Briey design and on-board DDR memory is proper (I am not sue if it lack of control direction of data-flow in it, but there isn't any otion for that in Xilinx "IP core" named "Select I/O Wizard).

I had posted and detailed describe my attempts to launch Briey SOC on Mimas V.2 Board on edaboard.com forum, but I didn't get answer solving my issues. Here is link to my post related to this topic:

[https://www.edaboard.com/showthread.php?383756-Briey-SoC-(RISCV)-and-Mimas-V-2-Numato-FPGA-board](url)

I hope that5 someone will be able to help me with these problems.

Thanks and regards

FlyingxDutch, 2019-05-03:
Here is ziped full project - ISE 14,7 :
[BrieySoCNext06.zip](https://github.com/SpinalHDL/VexRiscv/files/3141859/BrieySoCNext06.zip)

FlyingxDutch, 2019-05-03:
And here is diagram of new top module BrieySoC.v
[image]

And here internal organisation of Mimas V.2 DDR memory chip (MT46H32M16LF)
[image]

mithro, 2019-05-03:
@FlyingxDutch You might want to look at using LiteX with VexRISC-V we already have support for DDR on the MimasV2 board. 

There is a tutorial for getting micropython running on MimasV2 here -> https://fupy.github.io/MimasV2.html

Dolu1990, 2019-05-04:
@FlyingxDutch Briey is only support SDRAM-SDR. Sorry :)

FlyingxDutch, 2019-05-05:
Hello,
now that i am aware, how it is implemented I think it is time to close that issue. I will look at LiteX  VexRISC-V implementation as @mithro adviced me.

Thanks all for quick answers and Regards

closed 2019-05-05 by FlyingxDutch