SpinalHDL/VexRiscv #76: Support for native BRAM instruction fetch bus

smunaut, 2019-05-22:
I'm working on the ice40 and running all application out of block ram.

I think I could win in both speed and resources if instead of having a full wishbone interface to fetch instructions I could have a simple interface with address and where data is always provided on the next cycle (and so you can also provide a new address at each cycle and always get the data 1 cycle later).

Dolu1990, 2019-05-22:
Hi,

Actualy, the cacheless VexRiscv isn't a Wishbone native CPU,  but as it's own instruction bus which work in a very similar way of what you specified.

The VexRiscv-verilog repo was created for Migen/Litex users which need Wishbone, so wishbone bridge was added :)

There is the instruction bus definition:
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/IBusSimplePlugin.scala#L77

And bellow in that class, you can see all sort of toWishbone, toXXX function which bridge it into something else.

smunaut, 2019-05-22:
Thanks, yeah the PipelinedMemoryBus seems to work nicely :)

closed 2019-05-22 by smunaut