SpinalHDL/VexRiscv #81: Add support for fast register saving

mithro, 2019-07-26:
VexRISC-V uses blockrams on the iCE40. This should make it easy to support saving the whole register file by changing one of the upper address bits. This would be super useful for making interrupt handling a lot faster.

I'm unsure how this should be exposed? Maybe as a custom instruction or something?

Dolu1990, 2019-07-26:
I hade some discution about that with @gmarkall
So, it look like the fast interrupt workgroup isn't looking that way, but you are right, it would be a realy easy way to have fast interrupt (for free in FPGA)

I already explored a bit this idea in the actual RegFilePlugin (https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/RegFilePlugin.scala#L21)
But never realy tested it / used it.

This would save 16 load + 16 store + 2 pc write on  C compatible interrupts, which is much.

Actualy the way it is implemented is by adding new CSR which allow switching between the two register file banks with two modes : Permanant switch / Single instruction switch. 
It could have been custom instruction instead of additional CSR, it do not realy matter.

Dolu1990, 2019-07-26:
Also, nothing prevents to swap the default RegFilePlugin for another implementation for new experiments.

hpax, 2023-08-14:
This is indeed a *very* common situation on FPGAs; most FPGA block RAMs are much larger than 1024 bits (the standard RISC-V register file size.)

I experimented with this on picorv32, and ended up settling on a way to use this for a tiny embedded real time OS in this manner:

1. A new CSR represents the current user (as opposed to interrupt) context.
2. In interrupt context, register bank 0 is always used.
3. In user context, the CSR determines the bank used.
4. A separate instruction (or mapping to CSR space) for copying data to/from user context registers.