SpinalHDL/VexRiscv #82: illegal instruction with Zephyr

FrankBuss, 2019-08-10:
I'm trying to use VexRiscv with Zephyr, but it generates an illegal instruction exception when trying to call the z_arch_irq_enable function. I could debug it, and this is the instruction which caused the exception:
```
csrr   a4,0xbc0
```
I'm using the litex_vexriscv BOARD definition in Zephyr, but with a modified version of Murax.scala, without LiteX. If I comment the instruction, it works, at least I can start the hello world application and see the text on the UART output. But of course interrupts wouldn't work. How do I need to change Murax.scala to add the missing instruction, or is this a bug? Or do I need LiteX for this?

I couldn't find a tutorial how to create a working configuration with VexRiscv and LiteX for Zephyr. But if this missing instruction would work, I think then I don't need LiteX.

Dolu1990, 2019-08-10:
0xbc0 is a custom CSR added for the Litex SoC to handle the interrupt into the misoc way.
=> https://github.com/m-labs/VexRiscv-verilog/blob/master/src/main/scala/vexriscv/GenCoreDefault.scala#L205

So probably you should use a fully RISC-V compliant zephyr board definition.

For Zephyr with VexRiscv + litex, the only doc i know about is this one : https://docs.zephyrproject.org/latest/boards/riscv/litex_vexriscv/doc/litex_vexriscv.html

About more standard VexRiscv board def in zephyr, i just know that those people are doing it with that branch : 
https://github.com/limaconaudio/zephyr/commits/incus

Sorry, can't realy provide more data about it, i'm personnaly havn't used zephyr since 8 months

FrankBuss, 2019-08-10:
I tried to add the ExternalInterruptArrayPlugin, but now it doesn't compile. Attached is the full configuration.
[Murax.zip](https://github.com/SpinalHDL/VexRiscv/files/3489290/Murax.zip)