SpinalHDL/VexRiscv #83: RiscV A "AMO" with DBusSimplePlugin

BlamKiwi, 2019-08-12:
I've been playing around with RiscVexV for use as an embedded FPGA processor. The ```DBusSimplePlugin``` supports LR/SC but not AMO instructions to enable the full "A" extension. Are there plans to provide an implementation or pointers for how I would implement them myself? I just want to flesh out the implementation to allow full support for the Rust embedded atomics library.

Dolu1990, 2019-08-12:
Right about the dBusSimplePlugin. would you like a real atomic implemention (atomic request going on the memory bus) or a fake atomic extension, which is only local into the CPU which emit a standard read and then a standard write for each atomic extention ?

BlamKiwi, 2019-08-12:
Local to the CPU is fine for my purposes. I mostly use atomics for interrupt safe code in embedded systems.

Dolu1990, 2019-08-12:
That's realy shame that the RISC-V do not split the LR/SC and Atomics operations in two different extension.

So, there is no plan to implement Atomic in the cache-less design. as LR/SC can be used for the same purpose with nearly zero hardware cost, and i personnaly do not have usages of full A for cacheless design.

But if you have time, you can improve the DBusSimplePlugin or implement your own with the plugin system it is totaly fesable :)