SpinalHDL/VexRiscv #84: vexriscv.MuraxSim error in Windows 10 using the latest github clone

flylandcs, 2019-09-09:
after SpinalSim + Verilator windows setup according to https://spinalhdl.github.io/SpinalDoc/spinal/sim/introduction/

sbt "test:runMain vexriscv.MuraxSim"    (using the latest github clone)

I got the following error:
...
[info] [Progress] Simulation workspace in C:\Dev\VexRiscv\.\simWorkspace\Murax
[info] [Progress] Verilator compilation started
[error] Nonzero exit code returned from runner: -1073741819
[error] (Test / runMain) Nonzero exit code returned from runner: -1073741819

stepping into the code, it crashed in "VerilatorBackend.scale" at the following line:
val nativeInstance: IVerilatorNative = nativeImpl.newInstance().asInstanceOf[IVerilatorNative]

nativeImpl.newInstance() will load "verilator_1.dll", which does exists. I don't have any clue on why calling nativeImpl.newInstance() will crash. Since nativeImpl is dynamically compiled, I cannot debug it further.

BTW, the SpinalSim + Verilator windows setup instruction may not be complete on some systems. I have two additional findings:

1) In C:\Program Files\Java\jdk1.8.0_201\include\win32\jni_md.h ,  I added "typedef int64_t __int64;" in order to compile

2) I have to use "export CPLUS_INCLUDE_PATH=/mingw64/include/c++/9.2.0:$CPLUS_INCLUDE_PATH" to solve stdlib.h issue.

Dolu1990, 2019-09-10:
hmm was working on W7. 

about 1) I should be able to workaround that
2) weird :/
Anyway, thanks for the info !

Dolu1990, 2019-10-21:
@flylandcs 
Now trying to fix that in SpinalHDL

workaround the __int64 by having the following in the main cpp file : 

```
#include <stdint.h>
#include <string>
typedef int64_t __int64;
#include <jni.h>
```

Didn't hit the stdlib.h issue, but  :
``` 
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x0000000052ebb6ab, pid=22448, tid=0x0000000000004fac
#
# JRE version: Java(TM) SE Runtime Environment (8.0_201-b09) (build 1.8.0_201-b09)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.201-b09 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# V  [jvm.dll+0x2db6ab]
#
```

What's a paine in the ass ^^

Dolu1990, 2019-10-21:
ahh by reinstalling msys that was with verilator 4 that is now fine

Dolu1990, 2019-10-21:
So at the end in my case, the only things 1) and 2) were not required, my step is : 
- W10
- msys64 with https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Simulation/install.html#from-the-mingw-packet-manager
- OpenJDK 13

sj0000e, 2020-01-15:
my case. icarus verilog's dll files is problem.
change path ;C:\msys64\usr\bin;C:\msys64\mingw64\bin;c:\iverilog\bin