SpinalHDL/VexRiscv #87: Does mdd and mwd commands work with openocd_riscv?

shufps, 2019-09-24:
Sorry for this question - I'm just trying to use the VexRiscV as drop-in-replacement for a Cortex M1.

I'm using the AXI4 variant. The first Master is connected to a BRAM controller for ROM, the second Master is connected to an AXI4 interconnect on which a BRAM controller is connected for RAM.

Starting debugging seems to work but unfortunately, I can't access memory via telnet:4444, using the commands mdw (memory dump word) and mww (memory write word).

I mirrored the ROM (Dual-Port BRAM) to another address (0xd0000000) to which it should be possible to upload the binary via:

monitor load_image Debug/rv2.elf 0xd0000000

But after this I always read 0-values in ROM, in mirrored ROM and in RAM - actually everywhere not matter where I try it.

Now I'm - more or less - clueless if mdw and mww actually is supported by the debugging interface of VexRiscV.

What do you think? :-)

Thank you in advance!
Thomas

shufps, 2019-09-24:
Small Update ... it seems I did something wrong and I can now read/write from/to my RAM (also load_image works) and reading from my mirrored ROM but reading from ROM gives only 0xdec0de1c which seems to be a value for bad address decoding.

I wonder how the system knows that "mdw" is supposed to read from the ROM via the first AXI4 Master ...

Do you have any ideas for me?

shufps, 2019-09-24:
Issue can be closed - I got it working.

I found out two things: The AXI4-demo doesn't have hardware-breakpoints enabled but it uses an read-only instruction bus. I had to enable hardware-breakpoints. And the reset-vector of IBusCachedPlugin was set to 0x80000000 which actually is the RAM address.

Probably it was tested with code running from RAM.

Thank you for your awesome work! I love your project very much :)

Dolu1990, 2019-09-25:
Hoo nice catch :D

Thanks :)