SpinalHDL/VexRiscv #88: Illegal instruction div, rem

shufps, 2019-09-25:
I got the VexRiscV (from the demo with AXI4) almost running ... the only thing which seems to not work is divu, div, rem which triggers the Illegal Instruction Trap.

I'm not sure if I'm using the wrong compiler-flags though... I used a framework of a SiFive E310.

Eclipse says: RV32I, Integer ABI ILP32. (flags -march=rv32i -mabi=ilp32)

There doesn't seem to be a lot more which could influence code generation for integer arithmetic.

The AXI-demo has the Divider-Plugin enabled ... So, I'm quite clueless why it's not working.

What do you think, what could cause the Illegal Instruction Trap?

shufps, 2019-09-25:
sorry, stupid me ... wasn't div or rem ... both are working in minimum program without optimization fine ...

I don't know what's causing the trap though ... it only happens near places in code where it was calculated with "%"-C operator ... I'll try to figure this out (and close this issue)

closed 2019-09-25 by shufps

Dolu1990, 2019-09-25:
About GCC 
 -march=rv32i   => no mul/div/rem instruction generated
 -march=rv32im  =>  include mul/div/rem instructions.

Just one thing which is important to know, is that the CSR plugin of VexRiscv is generaly configured to work with a very small subset of the full spec. for instance, MTVEC in generaly configured to be not readable to spare logic, and even most of the time configured to not be even writable and only constant.

If you have a trap, you can use the MEPC CSR to know where the issue was exactly.

let's me know how it goes :D

Dolu1990, 2019-09-25:
Do not be affraid to run a simulation of the core, if you want i can point you to some signal which give a good clue of what is happening.

shufps, 2019-09-26:
Thank you for your very kind comments! :)

The illegal-instruction is gone after compiling the toolchain for the rv32im ... I guess the binary version from sifive is buggy.

For instance, I also had problems with objcopy. I nailed the trap-handler to 0x20 via linker script and disassembly (via objdump) showed it's on the right place but it was not in the right place in the binary (was shifted about 12 bytes further back) which was created by objcopy. Objcopy seems also to be used before uploading the program to the memory via the debugger, the data read via mdw and openocd was the same.

So, I think I would recommend to always compile the toolchain^^

:)

shufps, 2019-09-26:
I finally - and successfully - migrated my cortex M1 project to the VexRiscV!

Had just one small error but this probably is my fault ... I implemented a AXI4-to-localbus bridge and when memcpying data to the periphery the data got corrupted. Out of 0x5851f42d always became 0x58585858 - for all data. But a for-loop works great. I didn't use byte-strom signals ... That could be the explanation but I don't care.

Everything else worked instantly :)

I have to say this VexRiscV project is INSANE ... It's the most genious I've ever seen!

What I like most is the plugin concept ... RISC-V stands for being flexible and your concept totally reflects this!

Awesome and thank you very much for your great work :)

shufps, 2019-09-26:
I forgot ... *in less than 5 days from start to finish! Incredible :)

shufps, 2019-09-29:
for completeness: after rebuilding the rv32im toolchain, everything worked flawlessly.

I think the binary toolchain has some bugs ... one thing I figured out was that objcopy also seemed to behave strange ... for instance, I nailed the trap-handler via linker-script to 0x20 and disassembly with objdump clearly showed this but when converting the ELF to a BIN, the trap-handler was moved by about 12 byte. Perhaps I did something wrong, I don't know.

But I would recommend in any case to rebuild the toolchain before compiling.