SpinalHDL/VexRiscv #90: Issue connecting debug to simulated CPU

sebastien-riou, 2019-10-02:
I cloned this repo and openocd_riscv repo, then I followed the steps below (which are exactly what is in the README.md):
```
#in the VexRiscv repository, to run the simulation on which one OpenOCD can connect itself =>
sbt "runMain vexriscv.demo.GenFull"
cd src/test/cpp/regression
make run DEBUG_PLUGIN_EXTERNAL=yes
#In the openocd git, after building it =>
src/openocd -c 'set VEXRISCV_YAML ../VexRiscv/cpu0.yaml' -f tcl/target/vexriscv_sim.cfg
```
At this point verilator says "CONNECTED" however openocd says:
```
Open On-Chip Debugger 0.10.0+dev-01214-g0ace94f (2019-10-02-18:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
../VexRiscv/cpu0.yaml
Info : only one transport option; autoselect 'jtag'
adapter speed: 4000 kHz
adapter_nsrst_delay: 260
jtag_ntrst_delay: 250
Info : set servers polling period to 50ms
Info : clock speed 4000 kHz
Info : TAP fpga_spinal.bridge does not have IDCODE
Info : TAP auto0.tap does not have IDCODE
Info : TAP auto1.tap does not have IDCODE
Info : TAP auto2.tap does not have IDCODE
Info : TAP auto3.tap does not have IDCODE
Info : TAP auto4.tap does not have IDCODE
Info : TAP auto5.tap does not have IDCODE
Info : TAP auto6.tap does not have IDCODE
Info : TAP auto7.tap does not have IDCODE
Info : TAP auto8.tap does not have IDCODE
Info : TAP auto9.tap does not have IDCODE
Info : TAP auto10.tap does not have IDCODE
Info : TAP auto11.tap does not have IDCODE
Info : TAP auto12.tap does not have IDCODE
Info : TAP auto13.tap does not have IDCODE
Info : TAP auto14.tap does not have IDCODE
Info : TAP auto15.tap does not have IDCODE
Info : TAP auto16.tap does not have IDCODE
Info : TAP auto17.tap does not have IDCODE
Info : TAP auto18.tap does not have IDCODE
Info : TAP auto19.tap does not have IDCODE
Warn : Unexpected idcode after end of chain: 21 0xfffff800
Error: double-check your JTAG setup (interface, speed, ...)
Error: Trying to use configured scan chain anyway...
Error: fpga_spinal.bridge: IR capture error; saw 0x0f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
requesting target halt and executing a soft reset
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```
After that if I try to connect with GDB anyway, I get that:
```
Info : accepting 'gdb' connection on tcp/3333
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: fpga_spinal.bridge: IR capture error; saw 0x0f not 0x01
Warn : Bypassing JTAG setup events due to errors
```

The generated yaml file looks like that:
```
debug: !!vexriscv.DebugReport {hardwareBreakpointCount: 0}
iBus: !!vexriscv.BusReport
  flushInstructions: [4111, 19, 19, 19]
  info: !!vexriscv.CacheReport {bytePerLine: 32, size: 4096}
  kind: cached
```
The weird thing is that the generated VexRiscv.v does not seems to have JTAG (no match for "tdo" for example), but it does have signals debug_bus*

Dolu1990, 2019-10-02:
Hoo right, i improved the JTAG recently, let's me check if that's related.

Dolu1990, 2019-10-02:
Ok, i checked it. 
So basicaly, everything is fine. the CPU do not necessearly host the jtag itself but a dedicated memory mappable debug port which give more freedoom and allow to connect multiple CPU throug a single common jtag in the SoC
.
If you see all those warning from openocd, it is because as there is no real JTAG but only a dirrect access to the debug port between openocd and the simulation    =>  openocd panic a bit.
But no worries to have, i just tested with eclipse i can load binaries and debug them as expected using those commands.
Just be carefull to not load a binary with RVC as this configuration do not implement it.

If you simulate SoC as Murax, they do implement a real jtag, and connect it to openocd without the trick mentioned above => openocd is happy ^^

Tell me if something isn't working fine

sebastien-riou, 2019-10-02:
Thanks for checking. Indeed it does work ! Sorry for that, I did tried to go on despite the warning earlier with disappointing result but now I cannot reproduce the problem. I guess I had something still running in the background or something like that. Closing this, not a real issue.

closed 2019-10-02 by sebastien-riou