SpinalHDL/VexRiscv #97: Spec-compliant debug interface

MarekPikula, 2019-11-08:
Hi, are there any plans to create spec-compliant debug interface for VexRiscv? Although current debug interface is really nice and has nice performance it would be great to have standard solution in place, which works with upstream OpenOCD provided by RISC-V Foundation and possibly other tools, which support or will support spec.

Dolu1990, 2019-11-09:
Currently there is no plans for it.
It's hard for me to evaluate the official debug spec, between what is optional and what isn't i should probably take a look in the last revision.

MarekPikula, 2019-11-12:
I'd say that pretty nice reference for a execution-based core would be [riscv-dbg](https://github.com/pulp-platform/riscv-dbg) from PULP Platform. It's written in relatively good SystemVerilog and has nice documentation.

xobs, 2019-11-19:
I know this isn't the right place to hold a discussion about it, but we use the DebugPlugin by placing it on the general [wishbone] memory bus and tunneling it via a bridge, such as UART, USB, PCIe, or SPI.

We then use a piece of software called `wishbone-tool` to create a gdbserver bridge over your given medium:

- https://github.com/xobs/wishbone-utils/blob/master/wishbone-tool/src/riscv/mod.rs
- https://github.com/xobs/wishbone-utils/blob/master/wishbone-tool/src/gdb.rs

This has the nice property that we can get debugging even on an ICE40UP5k.

rapgenic, 2023-09-13:
Hi @Dolu1990, is there any news on this?

We are currently using litex to implement a SoC, and we'd like to have a softcore with a spec compliant debug interface. Looking at the commit history I've found the following commit:

[cores/vexriscv_smp add risc-v official debug support via --with-privileged-debug](https://github.com/enjoy-digital/litex/commit/112f78bde37c454209a25de7465e96d76cf87552)

Is this related to this issue? Also the commit refers to the SMP version of the core, is it applicable to the non SMP version as well?

Thanks!

bunnie, 2023-09-13:
fwiw I'm doing a tape-out using the default Vex JTAG interface and so far it's been fine. The main things I've learned are you have to make sure your TRST and system reset are independently controllable for reliable GDB attachment, and that the TRST has a pull-down on it so when you're not using it you don't get any X-prop issues.

Yes, I have to run on a custom build of openocd but it's rare that I don't have to do a custom build of openocd anyways for one reason or the other...

Dolu1990, 2023-09-13:
Hi,

Sorry, i forgot.

Here is some doc : 
https://github.com/SpinalHDL/VexRiscv/tree/dev#embeddedriscvjtag

In short, from a raw VexRiscv perspective, there is a all in one plugin you can just add, you also need to set withPrivilegedDebug in the CsrPlugin config.

> Is this related to this issue? 

Yes

that all in one plugin is only for non-smp version.
For smp version thing need to be a bit more outside the CPU
In litex, no port was made for the non smp version.

Dolu1990, 2023-09-13:
Also, note that often, the upstream openocd riscv version kind of break, so better using the vexriscv openocd fork.

rapgenic, 2023-09-14:
Hi, 

Thanks for the information!

The documentation link you provided is in a dev branch, does it mean that it is still not released or is it only the documentation that was missing?

We're using the non SMP version for now, so the all in one plugin should be fine for us. 

If we wanted to port the litex version we should be modifying the https://github.com/litex-hub/pythondata-cpu-vexriscv/ repository, correct?

Dolu1990, 2023-09-14:
I think the master branch should be good aswell, documentation wasn't merged yet in master.

Yes, the https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/master/pythondata_cpu_vexriscv/verilog/src/main/scala/vexriscv/GenCoreDefault.scala file

Adding the plugin and setting withPrivilegedDebug in the CsrPlugin config
https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/a36d99eeea984c261e16b744fc1d287c81099777/pythondata_cpu_vexriscv/verilog/src/main/scala/vexriscv/GenCoreDefault.scala#L200
.copy(withPrivilegedDebug=true)

rapgenic, 2023-09-14:
@Dolu1990, thank you, I've been able to take that repository and make some modifications to build some additional variants of the cpu that have the spec compliant jtag interface!!

https://github.com/protech-engineering/pythondata-cpu-vexriscv/commit/a384b014bd9c4fb76da0d0dfca897bf9bc3ae20c#diff-654a256b989fb4c943d91df96864500e8e23919463ab8dc38c233be9f3ff4a37

This is the patch if you're interested.

I have:
 * Updated the vexriscv module to the master
 * Changed the `-d` flag from a boolean to a string: `vexriscv` for the custom one and `riscv` for the spec compliant one.
 * Added another variant to the makefile, that generates `______DebugRiscv.v` files

I hopefully will be testing the CPU and its integration with litex the next days. 

If everything is ok would you be then open to review a pull request to the `pythondata_cpu_vexriscv`?

Dolu1990, 2023-09-18:
> If everything is ok would you be then open to review a pull request to the pythondata_cpu_vexriscv?

Sure :D
This can be userfull for many.

MarekPikula, 2025-06-27:
Hi, I've heard that the spec-compliant debug interface is already there for VexRiscv. Is that true?

Dolu1990, 2025-06-27:
@MarekPikula  Hi, yes there is. here is an example : https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/GenFullWithOfficialRiscvDebug.scala

MarekPikula, 2025-06-27:
Nice! I guess this issue is solved then?