SpinalHDL/VexRiscv #99: How can i debug the customInstruction

Khaiduy, 2019-11-30:
I saw that we have a section for custom instruction and also the assembly code to debug it. But how can i run the .S file and show the flow of program.

Dolu1990, 2019-12-07:
Hi,

Basicaly, in the cpp regression, there is a mode to run a hex file instead of the tests : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/test/cpp/regression/makefile#L121

There is for instance some example of make configs : 
https://github.com/SpinalHDL/VexRiscv/blob/6fc5406901fcd34d2fa166761a588fea1b38cddf/src/test/scala/vexriscv/DhrystoneBench.scala#L62

So generate that config, add the RUN_HEX=absolute_path_to_your_hex, and that should do it.

Khaiduy, 2019-12-11:
Thank you for the reply. I can debug it now.

closed 2019-12-11 by Khaiduy

adz0612, 2022-09-22:
Hello, it may be trivial but how do I generate the mentioned HEX file? @Dolu1990

Dolu1990, 2022-09-22:
@adz0612 For instance : 
riscv64-unknown-elf-objcopy -O ihex bios.elf bios.hex

adz0612, 2022-09-22:
@Dolu1990  Okay. Thanks! And then, to test with the hex I run something like:

`make clean run REDO=10 IBUS=SIMPLE DBUS=SIMPLE CSR=no MMU=no DEBUG_PLUGIN=no MUL=no DIV=no  COREMARK=yes RUN_HEX=absolute_path_to_your_hex `
?

Dolu1990, 2022-09-23:
Yes, i guess you don't need REDO=10 MMU=no MUL=no DIV=no COREMARK=yes  But it is fine with it too ^^

Dolu1990, 2022-09-23:
@adz0612

adz0612, 2022-09-28:
Hey, @Dolu1990 I had tried the same but it still seems to be running the Dhrystone benchmark and the other things, am I doing something wrong?
I generate the custom core using 

`sbt "runMain vexriscv.demo.GenCustomSimdAdd" `
and then run
`make clean run IBUS=SIMPLE DBUS=SIMPLE CSR=no DEBUG_PLUGIN=no RUN_HEX=/home/aditya/Desktop/vexRiscv_tutorial/VexRiscv/src/main/c/murax/hello_world/build/hello_world.hex
`

Dolu1990, 2022-09-29:
@adz0612 Hmm that's weird, because in the main.cpp dhrystone come after RUN_HEX

Like the the 		#if defined(DEBUG_PLUGIN_EXTERNAL) || defined(RUN_HEX) wasn't triggered ?

adz0612, 2022-09-29:
I assumed it would work too, I manually forced all the flags to 'no', and now it's only running what I want. So that way it worked.

adz0612, 2022-09-29:
@Dolu1990  I had another similar issue while trying to debug using openOCD and verilator

 I generate my custom CPU using ( as shown under [Adding custom intruction](https://github.com/SpinalHDL/VexRiscv#add-a-custom-instruction-to-the-cpu-via-the-plugin-system) )
`sbt "runMain vexriscv.demo.GenCustomSimdAdd"`

And then when I try to run 9 DEBUG_PLUGIN=yes
`make clean run IBUS=SIMPLE DBUS=SIMPLE CSR=no MMU=no DEBUG_PLUGIN=yes MUL=no DIV=no DHRYSTONE=no REDO=2 CUSTOM_SIMD_ADD=yes
`

I get the following error:
`./main.cpp: In constructor ‘DebugPlugin::DebugPlugin(Workspace*)’:
../main.cpp:2407:8: error: ‘class VVexRiscv’ has no member named ‘debugReset’
 2407 |   top->debugReset = 0;
      |        ^~~~~~~~~~
../main.cpp: In member function ‘virtual void DebugPlugin::onReset()’:
../main.cpp:2458:8: error: ‘class VVexRiscv’ has no member named ‘debugReset’
 2458 |   top->debugReset = 1;
      |        ^~~~~~~~~~
../main.cpp: In member function ‘virtual void DebugPlugin::postReset()’:
../main.cpp:2464:8: error: ‘class VVexRiscv’ has no member named ‘debugReset’
 2464 |   top->debugReset = 0;
      |        ^~~~~~~~~~
../main.cpp: In member function ‘virtual void DebugPlugin::postCycle()’:
../main.cpp:2479:21: error: ‘class VVexRiscv’ has no member named ‘debug_resetOut’
 2479 |   top->reset = top->debug_resetOut;
      |                     ^~~~~~~~~~~~~~
`

How do I resolve this? I want to be able to debug applications as I can on the GenFull as shown [here](https://github.com/SpinalHDL/VexRiscv#interactive-debug-of-the-simulated-cpu-via-gdb-openocd-and-verilator)

Dolu1990, 2022-09-29:
@adz0612 

GenCustomSimdAdd has no DebugPlugin, so i think that just adding 
new DebugPlugin(ClockDomain.current.clone(reset = Bool().setName("debugReset"))),

in the plugin list of the GenCustomSimdAdd should do the trick ^^