SpinalHDL/VexRiscv #101: Can not include CustomCsrDemoPlugin to Briey

Khaiduy, 2019-12-18:
When i try to include the CustomCsrDemoPlugin to the Briey plugin list, and compile the Briey with sbt, it leads to the error 
NO DRIVER ON (toplevel/axi_core_cpu/gpio_read : in Bits[32 bits]), define at
How can i fix this problem?

Dolu1990, 2020-01-08:
Hi, sorry for the delay, i completly missed a bunch of issue XD

So, basicaly that's normal. the plugin will put signal in the toplevel for you to do something with, for instance connecting them to some toplevel IO to make them going in/out of the SoC

And the error you have, is SpinalHDL seeing you have some floating wire. That's a lint check :)

Khaiduy, 2020-01-09:
Thanks for the feedback, actually I have many questions for you =)), but let's get into the main topics. First, can you explain a little bit about CsrInterface? I don't see the difference between the usage of onRead and onWrite.
The second is the use of both Csr and custom instruction in a plugin (my article is to create a hardware accelerator module controlled by custom instrucstion via CSR), but the more I do it, the more I find there is not much benefit from Combining the two. What do you think about this.

Khaiduy, 2020-01-09:
Thanks for the feedback, actually I have many questions for you =)), but let's get into the main topics. First, can you explain a little bit about CsrInterface? I don't see the difference between the usage of onRead and onWrite.
The second is the use of both Csr and custom instruction in a plugin (my article is to create a hardware accelerator module controlled by custom instrucstion via CSR), but the more I do it, the more I find there is not much benefit from combining the two. What do you think about this.

closed 2021-03-24 by Khaiduy

Dolu1990, 2021-03-24:
Ahhh damned, just ping me when i have more than 3 days delay,  i easily lose tracks of issues.

> I don't see the difference between the usage of onRead and onWrite.

Basicaly, not all CSR access do read nor write, ex if the target register is x0 => no read.

> The second is the use of both Csr and custom instruction in a plugin 

The FpuPlugin is quite close to that. basicaly there is CSR to controle the rounding mode and check the flags of the FPU + the integration of the FPU itself. So the advantage of having all together is that you can localise a feature in the source code instead of spliting it too much.
https://github.com/SpinalHDL/VexRiscv/blob/fpu/src/main/scala/vexriscv/plugin/FpuPlugin.scala

Sorry again for the bad issue tracking