SpinalHDL/VexRiscv #102: Murax in JTAG daisy chain

sebastien-riou, 2019-12-21:
I daisy chain Murax with another tap. I get correct operation when I do:
`
TDI -> Murax -> other -> TDO`
I get incorrect operation for the opposite:
`
TDI -> other -> Murax -> TDO`
So I guess we cannot daisy chain 2 Murax together.

To investigate the issue I customized Murax's JTAG:

```
class TesicChip1JtagBridge(c: SystemDebuggerConfig) extends Component{
  val io = new Bundle {
    val jtag = slave(Jtag())
    val remote = master(SystemDebuggerRemoteBus(c))
  }

  val system = new Area{
    val cmd = Flow Fragment(Bits(1 bit))
    io.remote.cmd << cmd.toStream

    val rsp = Reg(Flow(SystemDebuggerRsp(c)))
    when(io.remote.cmd.valid){
      rsp.valid := False
    }
    when(io.remote.rsp.fire){
      rsp.valid := True
      rsp.payload := io.remote.rsp.payload
    }
    io.remote.rsp.ready := True
  }

  val jtag = ClockDomain(io.jtag.tck)(new Area{
    val testReg = Reg(UInt(75 bits))
    val tap = new JtagTap(io.jtag, 4)
    val idcodeArea = tap.idcode(B"x0FFFFAAB")(1)
    val writeArea = tap.flowFragmentPush(system.cmd,TesicChip1JtagBridge.this.clockDomain)(2)
    val readArea = tap.read(system.rsp)(3)
    val treadArea = tap.read(testReg)(instructionId = 4)
    val twriteArea = tap.write(testReg)(instructionId = 5)
    readArea.shifter.addTag(crossClockDomain)
  })
}
```

Instructions 1,4 and 5 are working in both cases, only instruction 2 and 3 don't.

Dolu1990, 2020-01-08:
Hi, Sorry, i realy missed some issue. Please ping me when you have no responses (stack overflow).

So, i never tested JTAG with multiple tap, but that's something i'm interrested in.

I will try it in simulation via the tcp_jtag, this will allow to have a good visibility on what is happening :)

Dolu1990, 2020-01-08:
Then about having multiple VexRiscv in the same Soc, basicaly, it is supported, what you have to do is to manualy instanciate the SystemDebugger + JtagBridge (hidden in https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/DebugPlugin.scala#L72)

And connect the SystemDebugger memory interface to all the VexRiscv that you have with different address, and then in openocd config script, define multiple target (https://github.com/SpinalHDL/openocd_riscv/blob/riscv_spinal/tcl/target/murax.cfg#L18)
but with their corresponding address (-dbgbase XXX)

Then that should be fine. Be aware, i tested it probably 2 years ago, but should work fine.

Dolu1990, 2020-01-08:
I can reproduce it in sim

Dolu1990, 2020-01-08:
Have a fix, basicaly, that was because the bypass register of the JTAG was only sampling the input in DR_SHIFT, but it should always sample the input. Changing that i have the following working in sim : 
3 bits custom tap -> 4 bits CPU tap -> 5 bits custom tap

```scala
      case plugin : DebugPlugin         => plugin.debugClockDomain{
        resetCtrl.systemReset setWhen(RegNext(plugin.io.resetOut))
        class SimpleJtagTap(ilen : Int, opcode : Long) extends Component {
          val io = new Bundle {
            val jtag    = slave(Jtag())
            val switchs = in  Bits(8 bit)
            val keys    = in  Bits(4 bit)
            val leds    = out Bits(8 bit)
          }

          val ctrl = new ClockingArea(ClockDomain(io.jtag.tck)) {
            val tap = new JtagTap(io.jtag, ilen)
            val idcodeArea = tap.idcode(B(opcode))(instructionId = 1)
            val switchsArea = tap.read(io.switchs)(instructionId = 5)
            val keysArea = tap.read(io.keys)(instructionId = 6)
            val ledsArea = tap.write(io.leds)(instructionId = 3)
          }
        }

        val cpuTap = plugin.io.bus.fromJtag()
        val customTap0 = new SimpleJtagTap(3, 0x87654321l)
        val customTap2 = new SimpleJtagTap(5, 0x80022001l)
        customTap0.io.switchs := 0
        customTap0.io.keys := 0
        customTap2.io.switchs := 0
        customTap2.io.keys := 0

        customTap0.io.jtag.tck := io.jtag.tck
        customTap0.io.jtag.tms := io.jtag.tms
        cpuTap.tck := io.jtag.tck
        cpuTap.tms := io.jtag.tms
        customTap2.io.jtag.tck := io.jtag.tck
        customTap2.io.jtag.tms := io.jtag.tms

        customTap0.io.jtag.tdi := io.jtag.tdi
        cpuTap.tdi := customTap0.io.jtag.tdo
        customTap2.io.jtag.tdi := cpuTap.tdo
        io.jtag.tdo := customTap2.io.jtag.tdo
```

I just have to test things a bit more and then will commit

Dolu1990, 2020-01-08:
Tested on hardware in the same configuration, it work fine. I pushed the fix, that's in SpinalHDL dev branch, so to use it, you need to use the VexRiscv dev branch too.

Dolu1990, 2020-01-08:
Hoo for the above case, i used the following openocd config : 

```
set  _ENDIAN little
set _TAP_TYPE 1234

if { [info exists CPUTAPID] } {
   set _CPUTAPID $CPUTAPID
} else {
  # set useful default
   set _CPUTAPID 0x10001fff 
}

adapter_khz 800
adapter_nsrst_delay 260
jtag_ntrst_delay 250

set _CHIPNAME fpga_spinal
jtag newtap miaou bridge -expected-id 0x80022001 -irlen 5 -ircapture 0x1 -irmask 0x1F 
jtag newtap $_CHIPNAME bridge -expected-id $_CPUTAPID -irlen 4 -ircapture 0x1 -irmask 0xF 
jtag newtap miaou bridge -expected-id 0x87654321 -irlen 3 -ircapture 0x1 -irmask 0x7 

target create $_CHIPNAME.cpu0 vexriscv -endian $_ENDIAN -chain-position $_CHIPNAME.bridge -coreid 0 -dbgbase 0xF00F0000

vexriscv readWaitCycles 12
vexriscv cpuConfigFile $MURAX_CPU0_YAML

poll_period 50

init
#echo "Halting processor"
soft_reset_halt
sleep 1000
```

sebastien-riou, 2020-01-08:
Thanks, it works on my side now, can debug under Eclipse no matter the order in the JTAG chain :-)

closed 2020-01-08 by sebastien-riou