SpinalHDL/VexRiscv #104: Ping me if you are waiting a feedback/answer to one of your issues :)

Dolu1990, 2020-01-08:


curliph, 2020-10-28:
vexRiscV support up to 5 pipeline stage.   how to configure to instance each stage?
for example, which plugin to instance IF, which to instance DEC and so on...
If I want to instance a RV32IMAC core, 2 pipeline (IF-DEC/DEC-EXE), or 3 pipeline (IF/DEC/EXE), 
how to write my configuration?
I known there are many demos , but no detials about pipeline stages.

Thanks for your work.

Dolu1990, 2020-10-28:
So basicaly the memory/writeback stages aren't added by plugin, but by the VexRiscvConfig : 
https://github.com/SpinalHDL/VexRiscv/blob/dev/src/main/scala/vexriscv/demo/GenTwoThreeStage.scala#L12
see withMemoryStage,  withWriteBackStage

Then the fetch stages are added by the IBusSimplePlugin/IBusCachedPlugin,for instance here it specify no fetch stage : 
https://github.com/SpinalHDL/VexRiscv/blob/dev/src/main/scala/vexriscv/demo/GenTwoThreeStage.scala#L23

The main limitation is that the decode stage can't be merged in the execute stage currently, but else all can be merged

curliph, 2020-10-28:
Thanks a lot, Dolu~

fatalfeel, 2021-10-10:
help
https://github.com/SpinalHDL/VexRiscv/issues/209

Saqlinahamad, 2021-12-04:
Hi @Dolu1990 , Can you please look at #223,
I have deadlines to verify in the board

Saqlinahamad, 2021-12-07:
Hi @Dolu1990, #223

m-pp, 2022-04-04:
Hi @Dolu1990 , #241

ldlinh2k, 2022-07-17:
HI @Dolu1990

davidwilley, 2022-07-28:
Hi @Dolu1990 Can you please have a look at #246 ?

dockside-code, 2022-08-03:
Hi! @Dolu1990 would you please take a look at #259 ? Thanks!

hparpinel, 2022-12-22:
Hi, I am having issues running the demos

Light9162, 2023-01-05:
Hi, @Dolu1990 can you assist me on #293?

xie-1399, 2023-03-10:
help
#320

hszilard13, 2023-07-25:
Hi, can you help me with #357?

Shris7, 2023-09-30:
Hi @Dolu1990, can you help me with #369 ?

ic-hjx, 2024-05-14:
Hi @Dolu1990 , I have some problem with @ #406

karegoud, 2024-07-08:
Hi @Dolu1990 , Can you take a look at #414

Chaitanya-kumar-Y, 2024-07-30:
Hi @Dolu1990 , Can you help me with #418 ?

Chaitanya-kumar-Y, 2024-08-02:
Hi @Dolu1990, help me with #418

xavier-design, 2024-08-04:
help #419

karegoud, 2024-08-13:
@Dolu1990. Can you help with #419

xavier-design, 2024-08-21:
@Dolu1990 , Help me with #422

xavier-design, 2024-08-26:
@Dolu1990  Help with #422

karegoud, 2024-09-25:
@Dolu1990 can you have a look at #277

karegoud, 2025-01-15:
@Dolu1990 can you look at #439

xavier-design, 2025-02-12:
@Dolu1990  help with #443

xavier-design, 2025-02-16:
@Dolu1990  help #443

karegoud, 2025-06-10:
@Dolu1990 can you please look into #451