SpinalHDL/VexRiscv #116: Whether synchronous reset is supported ？

TianyangL, 2020-02-28:
Hi, this is  my frist SpinalHDL project，but i hope use SYNC reset，is it supported now ?

tomverbeure, 2020-02-28:
It is! Just set the reset strategy in the clock domain to SYNC.

tomverbeure, 2020-02-28:
Here's an example:
https://github.com/tomverbeure/panologic-g2/blob/3719743e07441301c1737872243cdab583e8486e/spinal/src/main/scala/pano/Pano.scala#L142

In SpinalHDL, the reset strategy is defined in the clock domain. You can change that strategy in one location, and all the generated RTL within that clock domain will automatically adapt.

TianyangL, 2020-02-28:
Yeah， I have seen the clock domin. But I wish modify VexRiscv. I do not find reset king config in VexRiscv. So I need to modify the source code is it ？

tomverbeure, 2020-02-28:
You will instantiate the VecRiscv in your design and somewhere in the levels above, there will be a ClockingArea that defines the clock domain used for that whole hierarchy.

The VexRiscv will simply inherit that clock domain and use that kind of reset method.

You don’t need to cha he anything in the VexRiscv code itself.

In the example that I gave you, if you create the Verilog, the VexRiscv will use SYNC reset. When you change that clock domain to be ASYNC, the RTL generates for the VexRiscv will become ASYNC reset.

Just try any of the many VexRiscv examples (or my project). Create the verilog. Then change the reset and observe what happens.

TianyangL, 2020-02-28:
Thank you!!!  ^_^

closed 2020-02-28 by TianyangL