SpinalHDL/VexRiscv #118: MuraxSim error in Ubuntu18.04

rayjyh, 2020-03-05:
Set up SpinalSim and Verilator according to https://spinalhdl.github.io/SpinalDoc/spinal/sim/introduction/#performance.

Got this error [sim_log.txt](https://github.com/SpinalHDL/VexRiscv/files/4293968/sim_log.txt) after sbt "test:runMain vexriscv.MuraxSim"

Dolu1990, 2020-03-05:
Hi, this docuementation is outdated, now it is on https://spinalhdl.github.io/SpinalDoc-RTD/

How did you found the link to the old version ?

rayjyh, 2020-03-06:
I didn't see any information about murax simulation on https://spinalhdl.github.io/SpinalDoc-RTD/. So I tried to follow the latest instruction  at https://github.com/SpinalHDL/VexRiscv#murax-soc. It failed with the 
[sim_log.txt](https://github.com/SpinalHDL/VexRiscv/files/4295853/sim_log.txt)

Dolu1990, 2020-03-09:
Ahhhhhhhhhhhh

You have to run that command in the root of the repo, so directly in ~/Desktop/VexRiscv :)

angeeelcm, 2021-02-10:
I have problems whit Murax SoC generation.
I follow the same steps that https://github.com/SpinalHDL/VexRiscv#murax-soc say and I got this: (after src/openocd -f tcl/interface/jtag_tcp.cfg -c "set MURAX_CPU0_YAML /home/spinalvm/Spinal/VexRiscv/cpu0.yaml" -f tcl/target/murax.cfg command)

Open On-Chip Debugger 0.10.0+dev-01230-g33f5b081-dirty (2021-02-10-13:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Error: BUG: command 'jtag_tcp' does not have the '.usage' field filled out
Info : only one transport option; autoselect 'jtag'
/home/spinalvm/Spinal/VexRiscv/cpu0.yaml
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter srst delay' not 'adapter_nsrst_delay'
Info : set servers polling period to 50ms
Error: cpuConfigFile /home/spinalvm/Spinal/VexRiscv/cpu0.yaml not found
Error: target 'fpga_spinal.cpu0' init failed

How can I fix it?