SpinalHDL/VexRiscv #119: VexRiscv native bus signals

PedroCarvalho64, 2020-03-20:
Hi,
I'm trying to implement the VexRiscv on an FPGA in order to build my custom search. I don't see any document that explains the signals from the native VexRiscv bus. Some signals seem to be clear to me, but some like **iBus_cmd_ready** and **iBus_rsp_valid** or **dBus_cmd_ready** and **dBus_rsp_ready** I can't differentiate the different between them.
I think that **iBus_rsp_valid** and **dBus_rsp_ready** are like acknowledge to the CPU but **iBus_cmd_ready** and **dBus_cmd_ready** I can't figure out their purpose.
I'm new to SpinalHDL so I've only generated the core in VHDL and I'm trying the core my peripherals in VHDL. 
If someone could clarify me those signals I would be very grateful.

Best regards. :)

tomverbeure, 2020-03-20:
The native bus of the VexRiscv is such that it can have multiple data request in flight. To make that possible, it splits request for data from return of data. Furthermore, it doesn't assume a fixed latency between a data request and data return.

When the VexRiscv needs data, it will assert iBus_cmd_valid with a valid address and it will keep valid asserted until the memory slave acknowledges the request with Ibus_cmd_ready.

The memory slave can then do whatever it wants to get the requested data. When the memory slave has the data, it will assert iBus_rsp_valid together with the data.

As said before: in theory, it's possible for the VexRiscv to issue multiple read requests before the data of the first request has returned. In practice, that depends on the configuration.

tomverbeure, 2020-03-20:
To give an example, [here](https://github.com/SpinalHDL/VexRiscv/blob/31667b18d8fb1acee198c141900f5dd874f1c8e3/src/main/scala/vexriscv/demo/MuraxUtiles.scala#L29-L30)'s an example where the iBus and the dBus are multiplexed into a single request bus to memory.

You can see here that dBus request are getting priority: it's cmd.ready gets asserted when the ready of the multiplexed bus is asserted. The iBus cmd.ready only gets asserted when dBus is not requesting any data.

PedroCarvalho64, 2020-03-23:
Ohh yes I see the difference between them now, but can the memory slave assert **dBus_cmd_ready** and **dBus_rsp_ready** at the same time or is there a minimum time between **dBus_cmd_ready** and **dBus_rsp_ready**?

tomverbeure, 2020-03-23:
If your memory slave can accept requests at all times, it's allowed to tie dBus_cmd_ready to 1.

So, yes, dBus_cmd_ready and dBus_rsp_ready can be 1 at the same time. 

As for the timing of dBus_rsp_ready, I *believe* that there should always be at least 1 clock cycle between the assertion of dBus_cmd_req and the *corresponding* dBus_rsp_ready. In other words: no asynchronous data return.

Tom

PedroCarvalho64, 2020-03-23:
Ok it's what I've tried which resulted into a halted cpu haha. 

Thank you very much.

Pedro

closed 2020-03-23 by PedroCarvalho64

Dolu1990, 2020-03-23:
@tomverbeure 

> As for the timing of dBus_rsp_ready, I believe that there should always be at least 1 clock cycle between the assertion of dBus_cmd_req and the corresponding dBus_rsp_ready. In other words: no asynchronous data return.

Right

@PedroCarvalho64 Then just to explaine the :
io.iBus.rsp.valid vs io.dBus.rsp.ready

Basicaly, in the case of the instruction stuff, the iBus is considered as a fork from the CPU pipeline, the instruction bus can be configured to have more than one pending request while keeping full CPU IPC.

For the dBus that's not the same, as it isn't as much performance critical than iBus, and as it has side effects, the dBus is considered as part of the CPU pipeline, it doesn't realy fork a memory transaction, as on the response of the dbus, it ask if the dBus is ready to move forward, a bit like AHB.