SpinalHDL/VexRiscv #120: Compile options for regression test programs

genesys2, 2020-04-01:
Hi,

i have another question regarding the regression testing. 

Short version: What compiler flags / linker file (?) did you use to compile the regression test programs (Coremark / Dhrystone)?

Long version: 
I need to modify the test programs (e.g. coremark) to cover the changes I made to the processor. Therefore, I recompiled coremark, converted it to binary and copied the new version to `/src/test/resources/bin/coremark_new.bin` and edited the main function of `/src/test/cpp/main.c` so that it also covers this file. 

````
WorkspaceRegression("coremark_new")).withRiscvRef()
        //->loadBin(string(REGRESSION_PATH) + "../../resources/bin/coremark_" + rv + ".bin", 0x80000000)
        ->loadBin(string(REGRESSION_PATH) + "../../resources/bin/coremark_new.bin", 0x80000000)
        ->bootAt(0x80000000)
        ->setIStall(withStall > 0)
        ->setDStall(withStall > 0)
        ->run(50e6);
````

Note: I don't necessarily need to compare the traces to a reference, thus I commented this section of Workspace `run` (riscvRefEnable = False didn't work for some reason):
````
if(riscvRefEnable && top->VexRiscv->lastStagePc != riscvRef.lastPc){
    cout << hex << " pc missmatch " << top->VexRiscv->lastStagePc << " should be " << riscvRef.lastPc << dec << endl;
    fail();
}
````

If I run this, I eventually get the error `??? periphWritesGolden` followed by a `periphWrite timout`. I assume, the coremark binary isn't correctly compiled and that’s what causes the issue but that is just a guess. Coremark is build with

````
riscv32-unknown-elf-gcc -march=rv32ima -mabi=ilp32 -O2 -mcmodel=medany -static -std=gnu99 -fno-common -nostdlib -nostartfiles -lm -lgcc -T ../riscv64-baremetal/link.ld -I../riscv64-baremetal -I. -DFLAGS_STR=\""-march=rv32ima -mabi=ilp32 -O2 -mcmodel=medany -static -std=gnu99 -fno-common -nostdlib -nostartfiles -lm -lgcc -T ../riscv64-baremetal/link.ld   -lm -lgcc"\" -DITERATIONS=0  core_list_join.c core_main.c core_matrix.c core_state.c core_util.c ../riscv64-baremetal/core_portme.c ../riscv64-baremetal/syscalls.c ../riscv64-baremetal/crt.S -o ./coremark.bare.riscv -lm -lgcc
````
Is my assumption correct, that this issue is related to my compile options or is my approach otherwise flawed?

Thank you! :-)

genesys2, 2020-04-08:
I figured a few things out, I found the UART.h which allows printing to the console when the external debugger is used. However, the timing instructions do not work as gdb gets stuck in the interrupt handler. Also, the UART write does not seem to work when the hex file is executed during regression. 

@Dolu1990 do you maybe know how to compile Coremark / Dhrystone so they work when loaded during regression?

Dolu1990, 2020-04-09:
@genesys2 Hi

Sorry for the delay.

So basicaly, the periphWritesGolden stuff is a cross check with the VexRiscv software model that the hardware CPU is properly reading/writing peripherals.

One thing, especialy about GDB / debug.

Since march there was a bug in the DebugPlugin of VexRiscv which was corrupting step by step debug.
It is fixed in dev and in master, else you can just cherry pick the related commit, as it is realy small.

>  do you maybe know how to compile Coremark / Dhrystone so they work when loaded during regression?

Yes, basicaly, you should remove the .withRiscvRef() from your workspace, basicaly, having it enable the software model <> hardware model cross check, but the issue is that the software model do not support JTAG.
So, removing withRiscvRef should fix your issue, let's me know else :)

Regards
Charles

genesys2, 2020-04-15:
Hi Charles,

thank you! It is now working :) 

Regards, 
Jan

closed 2020-04-15 by genesys2